openrazer/openrazer #1562: Blackwidow Chroma V2 Macro keys don't emit anything over USB?

Kusoneko, 2021-07-18:
### Distribution

Gentoo

### Support

- [X] Have you tried rebooting?
- [X] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Hello, I've been learning multiple languages for a while, and since I'm never really using the macro keys at all, I figured I'd look into using the macro keys to have FCITX + Mozc switch the layout based on which key is pressed. Essentially, having M1 be for English, M2 for French, M3 for Japanese, and finally M4 and M5 to switch between hiragana and katakana. I attempted to press the macro keys while changing keybindings in both FCITX and Mozc configuration utilities. It doesn't appear to recognize the keypresses at all. I'm aware that Japanese keyboards have special keys for switching between kana sets, such as the EISU key and all that, so I figured I'd start out by using wireshark with usbmon to detect keypresses and figure out just what key is detected when the macro keys are pressed, which would allow me to say tell Xorg to rebind these keys to the Japanese keyboard keys and/or the F13-F24 keys so they can be detected by FCITX and Mozc configuration utilities.

So, I ran `lsusb` and learned that my keyboard is on bus 2 device 3:

```
Bus 002 Device 003: ID 1532:0221 Razer USA, Ltd RZ03-0203 Gaming Keyboard [BlackWidow Chroma V2]
```

Unfortunately, After looking into wireshark, capturing the keyboard and pressing various keys and then all the macro keys, I noticed something:

On the device endpoint using the display filter `usb.addr == "2.3.0"` (note that the usb.addr filter takes it's parameter in the format `b.d.e` where b = bus, d = device and e = endpoint), the computer requests the descriptors for the device and configuration. Following that, one can learn that there are 3 endpoints on the device:

```
USB URB
    [Source: 2.3.0]
    <Source or Destination: 2.3.0>
    [Destination: host]
    <Source or Destination: host>
    URB id: 0xffff89bf5f4b1600
    URB type: URB_COMPLETE ('C')
    URB transfer type: URB_CONTROL (0x02)
    Endpoint: 0x80, Direction: IN
    Device: 3
    URB bus id: 2
    Device setup request: not relevant ('-')
    Data: present (0)
    URB sec: 1626631920
    URB usec: 802854
    URB status: Success (0)
    URB length [bytes]: 84
    Data length [bytes]: 84
    [Request in: 111]
    [Time from request: 0.000243000 seconds]
    Unused Setup Header
    Interval: 0
    Start frame: 0
    Copy of Transfer Flags: 0x0000200, Dir IN
    Number of ISO descriptors: 0
CONFIGURATION DESCRIPTOR
    bLength: 9
    bDescriptorType: 0x02 (CONFIGURATION)
    wTotalLength: 84
    bNumInterfaces: 3
    bConfigurationValue: 1
    iConfiguration: 0
    Configuration bmAttributes: 0xa0 NOT SELF-POWERED REMOTE-WAKEUP
    bMaxPower: 250 (500mA)
INTERFACE DESCRIPTOR (0.0): class HID
   bLength: 9
   bDescriptorType: 0x04 (INTERFACE)
   bInterfaceNumber: 0
   bAlternateSetting: 0
   bNumEndpoints: 1
   bInterfaceClass: HID (0x03)
   bInterfaceSubClass: Boot Interface (0x01)
   bInterfaceProtocol: Keyboard (0x01)
   iInterface: 0
HID DESCRIPTOR
   bLength: 9
   bDescriptorType: 0x21 (HID)
   bcdHID: 0x0111
   bCountryCode: Not Supported (0x00)
   bNumDescriptors: 1
   bDescriptorType: HID Report (0x22)
   wDescriptorLength: 61
ENDPOINT DESCRIPTOR
   bLength: 7
   bDescriptorType: 0x05 (ENDPOINT)
   bEndpointAddress: 0x81 IN Endpoint:1
   bmAttributes: 0x03
   wMaxPacketSize: 8
   bInterval: 1
INTERFACE DESCRIPTOR (1.0): class HID
   bLength: 9
   bDescriptorType: 0x04 (INTERFACE)
   bInterfaceNumber: 1
   bAlternateSetting: 0
   bNumEndpoints: 1
   bInterfaceClass: HID (0x03)
   bInterfaceSubClass: No Subclass (0x00)
   bInterfaceProtocol: Keyboard (0x01)
   iInterface: 0
HID DESCRIPTOR
   bLength: 9
   bDescriptorType: 0x21 (HID)
   bcdHID: 0x0111
   bCountryCode: Not Supported (0x00)
   bNumDescriptors: 1
   bDescriptorType: HID Report (0x22)
   wDescriptorLength: 159
ENDPOINT DESCRIPTOR
   bLength: 7
   bDescriptorType: 0x05 (ENDPOINT)
   bEndpointAddress: 0x81 IN Endpoint:2
   bmAttributes: 0x03
   wMaxPacketSize: 16
   bInterval: 1
INTERFACE DESCRIPTOR (2.0): class HID
   bLength: 9
   bDescriptorType: 0x04 (INTERFACE)
   bInterfaceNumber: 2
   bAlternateSetting: 0
   bNumEndpoints: 1
   bInterfaceClass: HID (0x03)
   bInterfaceSubClass: No Subclass (0x00)
   bInterfaceProtocol: Keyboard (0x01)
   iInterface: 0
HID DESCRIPTOR
   bLength: 9
   bDescriptorType: 0x21 (HID)
   bcdHID: 0x0111
   bCountryCode: Not Supported (0x00)
   bNumDescriptors: 1
   bDescriptorType: HID Report (0x22)
   wDescriptorLength: 94
ENDPOINT DESCRIPTOR
   bLength: 7
   bDescriptorType: 0x05 (ENDPOINT)
   bEndpointAddress: 0x81 IN Endpoint:3
   bmAttributes: 0x03
   wMaxPacketSize: 8
   bInterval: 1
```

Following that, I'm assuming 2.3.0 is the first interface endpoint and only contains the device and configuration descriptors, so it would make sense that 2.3.1 and 2.3.2 are the other 2 interfaces, indeed, 2.3.1 seems to register key presses, unfortunately for my attempt, nothing happens in 2.3.1 when the macro keys are pressed while every other keys are captured there, so I'm wondering how the keys are supposed to be recognized at all as being pressed if nothing is sent over the USB wire when the macro keys are pressed. Perhaps they're on a different endpoint specifically for them? So, I looked at what was captured on 2.3.2... and absolutely nothing was captured! Perhaps the last endpoint isn't on 2.3.2 but rather some following number? So I tried from 2.3.2 to 2.3.9 and none of them captured anything. So perhaps the last endpoint isn't for macro keys and instead is for something else that doesn't send anything over the wire when doing key presses. Perhaps it's for leds or for the Chroma stuff. Not really important to this here. So, considering all of this, how are the macro keys registered at all when doing key presses?

Since pressing these keys don't seem to do anything when pressed, how can someone either assign them macros and have the macros work or rebind them to different keys? One would assume there would be a way, since the polychromatic team appears to have planned to add macro functionality in the next version of their software.

I would appreciate some help in understanding this stuff so that I can perhaps manage to do what I'm trying to do.
Thank you,
Kusoneko.

lah7, 2021-07-18:
> Since pressing these keys don't seem to do anything when pressed, how can someone either assign them macros and have the macros work or rebind them to different keys? 

Have you tried testing the daemon's on-the-fly macro recording feature? Press FN+F9 [macro], press one of M1-M5 keys then type. I believe the keyboard needs to be in "driver mode" for them to function, so if the macro LED rapidly blinks, restart the daemon as it's probably no longer in that mode. For general use, they emit generic F13-F17 keys so they can be used by other programs.

For macros, I have noticed replaying typing uses the :us: layout, so I run this at login to "fix" that for my :gb: layout:

    setxkbmap -layout 'gb'

You could use `evtest` to check if any keycodes are emitted, for example, <kbd>M1</kbd> on my BlackWidow Chroma emits:
```
Event: time 1626640798.864174, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70068
Event: time 1626640798.864174, type 1 (EV_KEY), code 183 (KEY_F13), value 0
Event: time 1626640798.864174, -------------- SYN_REPORT ------------
```

(3 end points for input is normal, but only one of them emits key presses)

Some users have also had success using a key remapping program like [sezanzeb/key-mapper](https://github.com/sezanzeb/key-mapper).

---

> One would assume there would be a way, since the polychromatic team appears to have planned to add macro functionality in the next version of their software.

Not the next version, but [a version in the distant future](https://docs.polychromatic.app/roadmap/). It'll be more of a convenience feature since key rebinding isn't anything special in the Linux world.

The OpenRazer Python library does have a very basic API for configuring macros for M1-M5 too.

Kusoneko, 2021-07-20:
> Have you tried testing the daemon's on-the-fly macro recording feature? Press FN+F9 [macro], press one of M1-M5 keys then type. I believe the keyboard needs to be in "driver mode" for them to [...]

Oh, ty. That seemed to have been the issue, I never actually have the daemon running on boot for some reason, I usually have to start it manually via either polychromatic-controller or some other means. I'll probably have to add it to my .xinitrc file as something that's started along with my window manager so that the issue doesn't happen again. For the record, FN+F9 did make the Macro icon blink quickly, and all 3 event endpoints in evtest returned nothing when running without the daemon turned on, once the daemon is on, everything seems to work, although, recording macros via FN+F9 doesn't seem to work at all, I tried doing that, and the macro icon blinked slowly, I pressed M1 and then the icon stayed on, and then I tried typing `test[enter]` in the terminal out of curiousity, followed by pressing M1 again (hopefully that's how you stop the recording) and then pressing M1 didn't type test at all. It is however now recognized in evtest as F13, which is much better, and hopefully bindable/rebindable.

> 
> For macros, I have noticed replaying typing uses the us layout, so I run this at login to "fix" that for my gb layout:
> 
> ```
> setxkbmap -layout 'gb'
> ```

This wouldn't be an issue for me as the default layout I'm using is US, and when I need french I switch between Canadian Multilingual Standard and French (Canada) layouts depending on whether I need the french specific keys that are missing from either one layout or the other. The Japanese layout is the truly complicated one, since it's actually multiple layouts rolled into one, with the additional feature that whatever Latin letters you type on the keyboard gets converted to either of the kana types depending on which one you're on, and then you also have a drop-down menu that pops up with suggested kanji based on whatever you were typing.

> You could use `evtest` to check if any keycodes are emitted, for example, M1 on my BlackWidow Chroma emits:
> 
> ```
> Event: time 1626640798.864174, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70068
> Event: time 1626640798.864174, type 1 (EV_KEY), code 183 (KEY_F13), value 0
> Event: time 1626640798.864174, -------------- SYN_REPORT ------------
> ```
> 
> (3 end points for input is normal, but only one of them emits key presses)
> 
> Some users have also had success using a key remapping program like [sezanzeb/key-mapper](https://github.com/sezanzeb/key-mapper).

I'm not too keen on using external software when you can technically have xorg do it by itself by using the .Xmodmap file you can create in yoour home directory, which you can use to rebind keys to whatever other key you want. Hopefully that works fine with multiple layouts though. Otherwise, I can probably still do most of it using the FCITX or Mozc configuration utilities.

> Not the next version, but [a version in the distant future](https://docs.polychromatic.app/roadmap/). It'll be more of a convenience feature since key rebinding isn't anything special in the Linux [...]

By next version, I mean their eternal 0.6 that will become 1.x whenever it's complete.

lah7, 2021-07-20:
To stop a macro record, press FN+F9 again after typing the combination. I believe the macro is cancelled to avoid recursion if you try pressing the recording key (M1) again. It's slightly different to how Razer Synapse _officially_ does it.

Anyhoo, this confirms the macro keys require the device to be in "driver" mode. An alternate to starting the daemon could be to echo bytes to the sysfs interface, but I'm unsure exactly what the `echo` command would be.

You can find the path like so:

    find /sys -name "device_mode"
    /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6.2/1-6.2:1.2/0003:1532:0203.000C

So, now you just need to a way to execute a command to switch keyboard layouts and bind it to M1-M5 (F13-F17). Some desktop environments provide a built-in "Keyboard Shortcuts" or "Custom Shortcuts" facility which would work.

Kusoneko, 2021-07-21:
> So, now you just need to a way to execute a command to switch keyboard layouts and bind it to M1-M5 (F13-F17). Some desktop environments provide a built-in "Keyboard Shortcuts" or "Custom Shortcuts" facility which would work.

I'm not using a DE, but fcitx is not very helpful here. In fcitx settings I can seemingly bind any key but not the M1-M5 keys. When attempting to bind regular F-keys, it works, but when I press M1, instead of showing F13 it shows "Empty". I'm assuming this means I'm gonna have to go have a closer look at fcitx since they're the ones not accepting F13-F24 keys, which is odd since despite them not being present on keyboards very often, they *are* part of the specs for keys a keyboard is allowed to have in the USB HID specs.

closed 2022-06-19 by github-actions[bot]