openrazer/openrazer #1563: Support for Razer Naga X

Jonathanblanke, 2021-07-23:
### Device Info

* Name: <Naga X>
* Product Number: <rz01-03590100>
* Model Number: <rz01-0359>

### Device Information
```
Please post the output of
    lsusb -d 1532:0096
```

```
Please post the output of
    ls -lah /dev/input/by-id/
total 0
drwxr-xr-x 2 root root 300 Jul 22 18:14 .
drwxr-xr-x 4 root root 760 Jul 22 18:14 ..
lrwxrwxrwx 1 root root  10 Jul 22 18:14 usb-046d_HD_Pro_Webcam_C920_75F9C86F-event-if00 -> ../event17
lrwxrwxrwx 1 root root  10 Jul 22 18:14 usb-Performance_Designed_Products_PDP_Wired_Controller_for_Xbox_Series_X_-_White_Camo_0000DB27CA7D87BD-event-joystick -> ../event16
lrwxrwxrwx 1 root root   6 Jul 22 18:14 usb-Performance_Designed_Products_PDP_Wired_Controller_for_Xbox_Series_X_-_White_Camo_0000DB27CA7D87BD-joystick -> ../js0
lrwxrwxrwx 1 root root   9 Jul 22 18:14 usb-Razer_Razer_Huntsman_Mini_00000000001A-event-if03 -> ../event5
lrwxrwxrwx 1 root root   9 Jul 22 18:14 usb-Razer_Razer_Huntsman_Mini_00000000001A-event-kbd -> ../event2
lrwxrwxrwx 1 root root   9 Jul 22 18:14 usb-Razer_Razer_Huntsman_Mini_00000000001A-if01-event-kbd -> ../event3
lrwxrwxrwx 1 root root   9 Jul 22 18:14 usb-Razer_Razer_Huntsman_Mini_00000000001A-if02-event-mouse -> ../event4
lrwxrwxrwx 1 root root   9 Jul 22 18:14 usb-Razer_Razer_Huntsman_Mini_00000000001A-if02-mouse -> ../mouse0
lrwxrwxrwx 1 root root  10 Jul 22 18:14 usb-Razer_Razer_Naga_X-event-if01 -> ../event13
lrwxrwxrwx 1 root root   9 Jul 22 18:14 usb-Razer_Razer_Naga_X-event-mouse -> ../event9
lrwxrwxrwx 1 root root  10 Jul 22 18:14 usb-Razer_Razer_Naga_X-if01-event-kbd -> ../event10
lrwxrwxrwx 1 root root  10 Jul 22 18:14 usb-Razer_Razer_Naga_X-if02-event-kbd -> ../event14
lrwxrwxrwx 1 root root   9 Jul 22 18:14 usb-Razer_Razer_Naga_X-mouse -> ../mouse1

```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>

Noah-Leick, 2021-07-23:
Duplicate of #1463

z3ntu, 2021-07-25:
As per previous comment

closed 2021-07-25 by z3ntu