openrazer/openrazer #1564: Support for Razer Kraken BT Kitty Edition (Bluetooth)

melissaboiko, 2021-07-23:
This is the Bluetooth version of #949 .

### Device Info

* Name: Razer Kraken BT Kitty Edition
* Product Number: RZ04-03520100-R3M1

### Device Information
```
[76326.701393] input: Razer Kraken BT Kitty Edition (AVRCP) as /devices/virtual/input/input49
```
```
▸ hcitool info 44:5E:CD:11:C4:66
Requesting information ...
        BD Address:  44:5E:CD:11:C4:66
        OUI Company: Razer Inc (44-5E-CD)
        Device Name: Razer Kraken BT Kitty Edition
        LMP Version: 5.0 (0x9) LMP Subversion: 0x2
        Manufacturer: PixArt Imaging Inc. (1628)
        Features page 0: 0xbf 0xee 0x8d 0xfa 0xdb 0xbf 0x5b 0x87
                <3-slot packets> <5-slot packets> <encryption> <slot offset>
                <timing accuracy> <role switch> <sniff mode> <RSSI>
                <channel quality> <SCO link> <HV3 packets> <u-law log>
                <A-law log> <CVSD> <power control> <transparent SCO>
                <broadcast encrypt> <EDR ACL 2 Mbps> <enhanced iscan>
                <interlaced iscan> <interlaced pscan> <inquiry with RSSI>
                <extended SCO> <EV4 packets> <EV5 packets> <AFH cap. slave>
                <AFH class. slave> <LE support> <3-slot EDR ACL>
                <5-slot EDR ACL> <sniff subrating> <pause encryption>
                <AFH cap. master> <AFH class. master> <EDR eSCO 2 Mbps>
                <3-slot EDR eSCO> <extended inquiry> <LE and BR/EDR>
                <simple pairing> <encapsulated PDU> <non-flush flag> <LSTO>
                <inquiry TX power> <EPC> <extended features>
        Features page 1: 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00
```

### Packet Captures

I don't have access to Synapse but I can post Android bluetooth captures from the Razer Audio and Chroma apps, is that desirable?

I'm willing to help with this in any way including programming etc.

### Android Chroma app screenshots
[image]
It has on/off toggles for the "gaming mode" (lower latency) and Chroma LEDs.  Latency can be toggled on the hardware button too, but not Chroma.

[image]
Some settings that it seems to toggle on hardware: minutes to automatic turn off, interface language, firmware (can be updated from here).

[image]
Languages supported: German, English, Spanish, French, Japanese, Mandarin, Korean.  Unsure if it loads actual audio files, it takes a while to update over BT.

[image]
A few equalizer modes, including "custom".  Unsure if this has anything to do with the hardware or is just an Android equalizer.

[image]
LEDs are setable on the separate Chroma app. This the static single colour mode (not shown: all Chroma modes have a separate control for brightness).

[image]
Breathing mode with 2 colors.

[image]
Rainbow cycle can't set custom colours.

[image]
Audiometer mode, pulses with audio stack amplitude.

The Chroma app also has functions to flash lights on notifications (colours customizable per app), alarm, SMS, and phone calls.

z3ntu, 2021-07-25:
Does the device show up in `lsusb` on Linux?

melissaboiko, 2021-07-26:
no. there is an USB port but it seems to be only for charging, you can't listen to music over USB on this.  both music and Chroma control appear to be Bluetooth-only.  Plugging in the charger doesn't even show anything on dmesg.

melissaboiko, 2021-07-26:
Some more info I should've posted before.  The device produces 2 Bluetooth addresses, one labelled `Razer Kraken BT Kitty Edition` , the other `Razer Stereo`.  This is the logs I get when I connect to the first:

```syslog
kernel: [104360.454990] input: Razer Kraken BT Kitty Edition (AVRCP) as /devices/virtual/input/input58
/usr/libexec/gdm-x-session[1647]: (II) config/udev: Adding input device Razer Kraken BT Kitty Edition (AVRCP) (/dev/input/event27)
/usr/libexec/gdm-x-session[1647]: (**) Razer Kraken BT Kitty Edition (AVRCP): Applying InputClass "libinput keyboard catchall"
/usr/libexec/gdm-x-session[1647]: (II) Using input driver 'libinput' for 'Razer Kraken BT Kitty Edition (AVRCP)'
/usr/libexec/gdm-x-session[1647]: (II) systemd-logind: got fd for /dev/input/event27 13:91 fd 117 paused 0
/usr/libexec/gdm-x-session[1647]: (**) Razer Kraken BT Kitty Edition (AVRCP): always reports core events
/usr/libexec/gdm-x-session[1647]: (**) Option "Device" "/dev/input/event27"
/usr/libexec/gdm-x-session[1647]: (**) Option "_source" "server/udev"
/usr/libexec/gdm-x-session[1647]: (II) event27 - Razer Kraken BT Kitty Edition (AVRCP): is tagged by udev as: Keyboard
/usr/libexec/gdm-x-session[1647]: (II) event27 - Razer Kraken BT Kitty Edition (AVRCP): device is a keyboard
/usr/libexec/gdm-x-session[1647]: (II) event27 - Razer Kraken BT Kitty Edition (AVRCP): device removed
/usr/libexec/gdm-x-session[1647]: (**) Option "config_info" "udev:/sys/devices/virtual/input/input58/event27"
/usr/libexec/gdm-x-session[1647]: (II) XINPUT: Adding extended input device "Razer Kraken BT Kitty Edition (AVRCP)" (type: KEYBOARD, id 28)
/usr/libexec/gdm-x-session[1647]: (**) Option "xkb_model" "pc105"
/usr/libexec/gdm-x-session[1647]: (**) Option "xkb_layout" "us,de"
/usr/libexec/gdm-x-session[1647]: (**) Option "xkb_variant" ","
/usr/libexec/gdm-x-session[1647]: (**) Option "xkb_options" "grp:alt_shift_toggle,compose:rwin,terminate:ctrl_alt_bksp,grp_led:scroll,ctrl:nocaps"
/usr/libexec/gdm-x-session[1647]: (II) event27 - Razer Kraken BT Kitty Edition (AVRCP): is tagged by udev as: Keyboard
/usr/libexec/gdm-x-session[1647]: (II) event27 - Razer Kraken BT Kitty Edition (AVRCP): device is a keyboard
/usr/libexec/gdm-x-session[258577]: The XKEYBOARD keymap compiler (xkbcomp) reports:
/usr/libexec/gdm-x-session[258577]: > Error:            Meta_L added to symbol map for multiple modifiers
/usr/libexec/gdm-x-session[258577]: >                   Using Mod4, ignoring Mod1.
/usr/libexec/gdm-x-session[258577]: > Error:            Key <META> added to map for multiple modifiers
/usr/libexec/gdm-x-session[258577]: >                   Using Mod4, ignoring Mod1.
/usr/libexec/gdm-x-session[258577]: Errors from xkbcomp are not fatal to the X server
```

I don't know if treating it as a 'keyboard' is normal to have access to the hardware keys, but AVRCP is the Bluetooth audio-video remote control profile, and it does show up as a headset for Gnome and I'm able to listen to audio on it.

If I try to connect to `Razer Stereo`, this  is what happens:

```
gnome-control-c[258986]: Setting up /org/bluez/hci0/dev_44_5E_CD_12_C4_66 failed: GDBus.Error:org.bluez.Error.ConnectionAttemptFailed: Page Timeout
```

I can't get information on this address at all; `hcitool info` gives an I/O error, and `bt-device` straight out segfaults :) But I can't connect to this address on Android either, and as far as I can tell, the Chroma app works over Bluetooth over the first address only.

(got these on kernel 5.13.4, Debian bullseye, bluez 5.55-3.1 ).

z3ntu, 2021-08-01:
Possible that the device is using a completely custom protocol over Bluetooth instead of USB.

You can sniff the bluetooth traffic in the Android developer options, please do that and upload it here. It is unlikely though that the device is easily integrated into openrazer due to the vast difference to all other devices.

librick, 2023-07-12:
I would also like support for this device. I sniffed the bluetooth traffic via Android and found the following info:

My headset's MAC address is of the form `00:14:41:XX:XX:XX`
The device has a primary "unknown" service with the following UUID:
`416d656c-2d52-617a-6572-424c4501f40a`.
Within that service is an "unknown" characteristic:
`416d0000-2d52-617a-6572-424c4501f40a`
and that characteristic's flags indicate that it's `WRITE NO RESPONSE`.

Commands are sent to the headset via the Attribute Protocol (ATT).  
Write commands (ATT opcode 0x52) are sent to the device for that characteristic (`416d0000-2d52-617a-6572-424c4501f40a`).

I've documented the following ATT commands/values.
In all cases, the first two octets appear to be an operation to perform, and the third octet appears to be the length of the remaining data in bytes.

#### LEDs On/Off and Spectrum Mode
`0xc0 00 01 00` turns the LEDs off
`0xc0 00 01 ff` turns the LEDs on
`0xc0 00 01 01` turns on "SPECTRUM" mode (cycles through rainbow)

#### Breathing Mode
`0xc2 00 06 RR GG BB RR GG BB` turns on "BREATHING" mode,
where the LEDs fade in and out between two colors, each provided as 3-octet hex values (RR GG BB), e.g.:  
`0xc2 00 06 ff 00 00 00 ff 00` // red and green fading  
`0xc2 00 06 ff 00 00 00 00 ff` // red and blue fading  
`0xc2 00 06 00 00 ff 00 00 ff` // blue and blue fading  

#### Static Mode
`0xc3 00 03 RR GG BB` turns on "STATIC" mode,
where the LEDs are set to a single color, e.g.:   
`0xc3 00 03 ff 00 00` // red  
`0xc3 00 03 00 ff 00` // green  
`0xc3 00 03 00 00 ff` // blue  

#### Audio Meter Mode
`0xc5 00 03 RR GG BB` turns on "AUDIO METER" mode,
where a single color is set and the brightness varies with headset audio levels, e.g.:  
`0xc5 00 03 ff 00 00` // music reactive, red  
`0xc5 00 03 00 ff 00` // music reactive, green  
`0xc5 00 03 00 00 ff` // music reactive, blue  

####  Auto Standby
Values are of the form `0xa7 00 01 XX`, where XX maps to some amount of time.
I've established the following mappings:
`0xa7 00 01 00` // "Never"
`0xa7 00 01 01` // "After 5 minutes"
`0xa7 00 01 02` // "After 15 minutes"
`0xa7 00 01 03` // "After 30 minutes"
`0xa7 00 01 04` // "After 45 minutes"
`0xa7 00 01 05` // "After 1 hour"

## Miscellaneous
In addition to the options above, the Android app provides the ability to change the prompt language, although having done it through Razer's app, it seems like a more complicated process than just writing to a characteristic (there's a whole loading screen and I think the headset disconnects and reconnects). A complete implementation would address this, but I don't really care for my personal use.

As far as the app's `Equalizer` and `Gaming Mode` settings, *I think* they just change the bluetooth audio profile; I didn't bother look into those. Nor have I looked into how the app retrieves the headset's current settings. If there's interest I can do more research.

It would be great to have support for this device. Great project btw. Thanks!

grbaugher, 2023-07-12:
I also would be happy to see this added and would help if possible to identify the functions it uses.

librick, 2023-07-14:
Small update, I found the FCC ID and filing for this model.
FCC ID RWO-RZ040352
https://fccid.io/RWO-RZ040352
https://fccid.io/RWO-RZ040352/Internal-Photos/Internal-Photos-1-4918477
https://fccid.io/RWO-RZ040352/Internal-Photos/Internal-Photos-2-4918478
From looking at the pictures, the model # of the main IC is "PAU1602FB"
Searching for that led me to https://device.report/e-filliate-inc/pau1602fb-s1r1

librick, 2023-07-16:
I'm currently looking into using [Bleak](https://github.com/hbldh/bleak), a Python BLE library, to integrate more BLE devices into OpenRazer, including this device. Any thoughts from the maintainers on this? Is there existing tooling within OpenRazer for working with BLE-based devices?

z3ntu, 2023-07-16:
If I were in your shoes, I'd definitely first try making a standalone Python script that changes these settings, and only afterwards look into how to integrate it in openrazer.

From what you've showed so far it seems to somewhat fit into the effect model we currently have in openrazer, but generally not a fan of putting a completely different protocol into openrazer. The scope as-is is big enough :wink: