openrazer/openrazer #1565: Fedora 34 - No lighting at all

shawnmuzick, 2021-07-24:
### Distribution

### Support

- [ ] Have you tried rebooting?
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

After a recent update to Fedora 34, I've lost all lighting on my Razer Mamba and Blackwidow.

shawnmuzick, 2021-07-25:
lsusb | grep 1532
Bus 001 Device 002: ID 1532:0228 Razer USA, Ltd BlackWidow Elite
Bus 001 Device 004: ID 1532:006c Razer USA, Ltd Mamba Elite (Wired)

shawnmuzick, 2021-07-25:
sudo journalctl -b | grep -i razer_mount

Jul 25 00:19:04 glados razer_mount[717]: Driver razerkbd
Jul 25 00:19:04 glados razer_mount[714]: Driver razermouse
Jul 25 00:19:04 glados razer_mount[718]: Device_ID 1-9
Jul 25 00:19:04 glados razer_mount[719]: Device_ID 1-11
Jul 25 00:19:04 glados razer_mount[720]: Modprobing razerkbd
Jul 25 00:19:04 glados razer_mount[721]: Modprobing razermouse
Jul 25 00:19:05 glados razer_mount[793]: Modprobed razermouse
Jul 25 00:19:05 glados razer_mount[798]: Driver razermouse
Jul 25 00:19:05 glados razer_mount[800]: Driver razermouse
Jul 25 00:19:05 glados razer_mount[799]: Driver razermouse
Jul 25 00:19:05 glados razer_mount[803]: Device_ID 1-11:1.2
Jul 25 00:19:05 glados razer_mount[802]: Device_ID 1-11:1.0
Jul 25 00:19:05 glados razer_mount[801]: Device_ID 1-11:1.1
Jul 25 00:19:05 glados razer_mount[807]: Driver razermouse
Jul 25 00:19:05 glados razer_mount[808]: Driver razermouse
Jul 25 00:19:05 glados razer_mount[809]: Driver razermouse
Jul 25 00:19:05 glados razer_mount[810]: Device_ID 0003:1532:006C.0007
Jul 25 00:19:05 glados razer_mount[811]: Device_ID 0003:1532:006C.0006
Jul 25 00:19:05 glados razer_mount[812]: Device_ID 0003:1532:006C.0005
Jul 25 00:19:05 glados razer_mount[813]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:006C.0007/
Jul 25 00:19:05 glados razer_mount[814]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:006C.0006/
Jul 25 00:19:05 glados razer_mount[815]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:006C.0005/
Jul 25 00:19:05 glados razer_mount[819]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:006C.0006/
Jul 25 00:19:05 glados razer_mount[821]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:006C.0007/
Jul 25 00:19:05 glados razer_mount[820]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:006C.0005/
Jul 25 00:19:05 glados razer_mount[826]: Modprobed razerkbd
Jul 25 00:19:05 glados razer_mount[832]: Driver razerkbd
Jul 25 00:19:05 glados razer_mount[831]: Driver razerkbd
Jul 25 00:19:05 glados razer_mount[833]: Driver razerkbd
Jul 25 00:19:05 glados razer_mount[834]: Device_ID 1-9:1.1
Jul 25 00:19:05 glados razer_mount[836]: Device_ID 1-9:1.2
Jul 25 00:19:05 glados razer_mount[835]: Device_ID 1-9:1.0
Jul 25 00:19:05 glados razer_mount[841]: Driver razerkbd
Jul 25 00:19:05 glados razer_mount[842]: Driver razerkbd
Jul 25 00:19:05 glados razer_mount[843]: Driver razerkbd
Jul 25 00:19:05 glados razer_mount[844]: Device_ID 0003:1532:0228.0002
Jul 25 00:19:05 glados razer_mount[845]: Device_ID 0003:1532:0228.0001
Jul 25 00:19:05 glados razer_mount[846]: Device_ID 0003:1532:0228.0003
Jul 25 00:19:05 glados razer_mount[847]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:0228.0002/
Jul 25 00:19:05 glados razer_mount[848]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:0228.0001/
Jul 25 00:19:05 glados razer_mount[849]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:0228.0003/
Jul 25 00:19:06 glados razer_mount[852]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:0228.0002/
Jul 25 00:19:06 glados razer_mount[854]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:0228.0001/
Jul 25 00:19:06 glados razer_mount[855]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:0228.0003/

shawnmuzick, 2021-07-25:
cat ~/.local/share/openrazer/logs/razer.log

2021-07-24 18:28:08 | razer                          | INFO     | Serving DBus
2021-07-25 00:20:20 | razer                          | INFO     | Initialising Daemon (v3.0.1). Pid: 2110
2021-07-25 00:20:20 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2021-07-25 00:20:20 | razer                          | INFO     | Found device.0: 0003:1532:006C.0005
2021-07-25 00:20:20 | razer.device0                  | INFO     | Initialising device.0 RazerMambaElite
2021-07-25 00:20:20 | razer                          | INFO     | Found device.1: 0003:1532:0228.0003
2021-07-25 00:20:20 | razer.device1                  | INFO     | Initialising device.1 RazerBlackWidowElite
2021-07-25 00:20:20 | razer.device1                  | INFO     | Putting device into driver mode. Daemon will handle special functionality
2021-07-25 00:20:20 | razer                          | INFO     | Serving DBus

lah7, 2021-07-25:
Looks like the driver and daemon are working as expected.

* Do you use a frontend? What happens if you try changing the lighting / brightness?
* Check `~/. config/openrazer/razer.conf` and turn off anything that may manipulate lighting (screensaver, persistence), then restart the daemon.

shawnmuzick, 2021-07-25:
~/.config/openrazer/razer.conf 

1 [General]
  2 # Verbose logging (logs debug messages - lotsa spam)
  3 verbose_logging = False
  4 
  5 
  6 [Startup]
  7 # Set the sync effects flag to true so any assignment of effects will work across devices
  8 sync_effects_enabled = False
  9 
 10 # Turn off the devices when the systems screensaver kicks in
 11 devices_off_on_screensaver = False
 12 
 13 # Mouse battery notifier
 14 mouse_battery_notifier = False
 15 
 16 # Mouse battery notification frequency [s] (0 to disable)
 17 mouse_battery_notifier_freq = 600
 18 
 19 # Apply effects saved to disk when daemon starts
 20 restore_persistence = False
 21 
 22 [Statistics]
 23 # Collects number of keypresses per hour per key used to generate a heatmap
 24 key_statistics = True

shawnmuzick, 2021-07-25:
Additionally

Using RazerGenie as a frontend, any changes are completely ignored, issue persists.
I've tested the hardware on a windows device, lighting/etc works, so there shouldn't be an issue with the physical units,

muellphi, 2021-07-26:
Same for me on all devices (also on Fedora 34):

```
Bus 005 Device 005: ID 1532:0226 Razer USA, Ltd Huntsman Elite
Bus 005 Device 003: ID 1532:022b Razer USA, Ltd Tartarus V2
Bus 005 Device 006: ID 1532:007e Razer USA, Ltd RC30-030502 Mouse Dock
Bus 005 Device 004: ID 1532:007b Razer USA, Ltd RC30-0305 Gaming Mouse Dongle [Viper Ultimate (Wireless)]
```

I use Polychromatic as Frontend and noticed that brightness for all devices was set to 0%.
All other settings persisted.

After resetting the brightness back to 100% everything works as before.

shawnmuzick, 2021-07-26:
That was it. I feel like an idiot. The brightness just got reset to 0 somehow.

closed 2021-07-26 by shawnmuzick