openrazer/openrazer #1566: My Deathadder Essential has different ID. 1532:0098

AchalS-iglushrek, 2021-07-26:
### Device Info

* Name: Deathadder Essential
* Product Number: RZ01-03850100
* Model Number: RZ01-0385

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
    
Bus 001 Device 002: ID 1532:0098 Razer USA, Ltd Razer DeathAdder Essential
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0098 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer DeathAdder Essential
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1

```

```
Please post the output of
    ls -lah /dev/input/by-id/
    
total 0
drwxr-xr-x 2 root root 220 Jul 26 16:20 .
drwxr-xr-x 4 root root 620 Jul 26 16:20 ..
lrwxrwxrwx 1 root root   9 Jul 26 16:20 usb-C-Media_Electronics_Inc._USB_PnP_Sound_Device-event-if02 -> ../event9
lrwxrwxrwx 1 root root  10 Jul 26 16:20 usb-E-Signal_GIGABYTE_Keyboard-event-if01 -> ../event12
lrwxrwxrwx 1 root root  10 Jul 26 16:20 usb-E-Signal_GIGABYTE_Keyboard-event-kbd -> ../event10
lrwxrwxrwx 1 root root  10 Jul 26 16:20 usb-E-Signal_GIGABYTE_Keyboard-if02-event-kbd -> ../event13
lrwxrwxrwx 1 root root   9 Jul 26 16:20 usb-Razer_Razer_DeathAdder_Essential-event-if01 -> ../event7
lrwxrwxrwx 1 root root   9 Jul 26 16:20 usb-Razer_Razer_DeathAdder_Essential-event-mouse -> ../event3
lrwxrwxrwx 1 root root   9 Jul 26 16:20 usb-Razer_Razer_DeathAdder_Essential-if01-event-kbd -> ../event4
lrwxrwxrwx 1 root root   9 Jul 26 16:20 usb-Razer_Razer_DeathAdder_Essential-if02-event-kbd -> ../event8
lrwxrwxrwx 1 root root   9 Jul 26 16:20 usb-Razer_Razer_DeathAdder_Essential-mouse -> ../mouse0
    
```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>
Same as a normal deathadder

z3ntu, 2021-08-01:
As a test you can try replacing all occurances of 006E in the code with 0098 and see if that works with your device.

pulsar17, 2021-09-09:
I am facing the same issue. I tried replacing `006E` with `0098`, then built the driver again using instructions in the wiki. 
Here's what Polychromatic shows for the custom built driver

[image]

The latter error in the image also appears when using openrazer from the PPA. The former error (about udev rules) is the new one)

OS Info : Pop!_OS 21.04 x86_64

pulsar17, 2021-09-09:
The logs show some kind of permission error:
```
2021-09-10 00:36:14 | razer                          | INFO     | Serving DBus
2021-09-10 00:38:29 | razer                          | INFO     | Stopping daemon on signal 15
2021-09-10 00:38:32 | razer                          | INFO     | Initialising Daemon (v3.1.0). Pid: 9620
2021-09-10 00:38:32 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2021-09-10 00:38:32 | razer                          | INFO     | Found device.0: 0003:1532:0098.0008
2021-09-10 00:38:32 | razer                          | CRITICAL | Could not access /sys/devices/pci0000:00/0000:00:01.3/0000:03:00.0/usb1/1-5/1-5:1.0/0003:1532:0098.0008/device_type, file is not owned by plugdev
```
I changed ownership of this particular file. This fixed the 'Check device compatibility' issue but then there are many more files which cause permission denied error when I run `openrazer-daemon -Fv`. 

Here is a long listing of the parent directory of the file in logs:
```
-r--r--r-- 1 root     root    4096 Sep 10 00:43 country
-rw-rw---- 1 root     root    4096 Sep 10 00:39 device_mode
-r--r----- 1 root     root    4096 Sep 10 00:27 device_serial
-r--r----- 1 pulsar17 plugdev 4096 Sep 10 00:40 device_type
-rw-rw---- 1 root     root    4096 Sep 10 00:27 dpi
lrwxrwxrwx 1 root     root       0 Sep 10 00:27 driver -> ../../../../../../../../bus/hid/drivers/razermouse
-r--r----- 1 root     root    4096 Sep 10 00:43 firmware_version
drwxr-xr-x 3 root     root       0 Sep 10 00:27 hidraw
drwxr-xr-x 3 root     root       0 Sep 10 00:27 input
-rw-rw---- 1 root     root    4096 Sep 10 00:43 logo_led_brightness
--w--w---- 1 root     root    4096 Sep 10 00:43 logo_matrix_effect_breath
--w--w---- 1 root     root    4096 Sep 10 00:43 logo_matrix_effect_none
--w--w---- 1 root     root    4096 Sep 10 00:43 logo_matrix_effect_static
-r--r--r-- 1 root     root    4096 Sep 10 00:43 modalias
-rw-rw---- 1 root     root    4096 Sep 10 00:43 poll_rate
drwxr-xr-x 2 root     root       0 Sep 10 00:43 power
-r--r--r-- 1 root     root    4096 Sep 10 00:27 report_descriptor
-rw-rw---- 1 root     root    4096 Sep 10 00:43 scroll_led_brightness
--w--w---- 1 root     root    4096 Sep 10 00:43 scroll_matrix_effect_breath
--w--w---- 1 root     root    4096 Sep 10 00:43 scroll_matrix_effect_none
--w--w---- 1 root     root    4096 Sep 10 00:43 scroll_matrix_effect_static
lrwxrwxrwx 1 root     root       0 Sep 10 00:27 subsystem -> ../../../../../../../../bus/hid
--w--w---- 1 root     root    4096 Sep 10 00:43 test
-rw-r--r-- 1 root     root    4096 Sep 10 00:27 uevent
-r--r----- 1 root     root    4096 Sep 10 00:43 version

```
I somehow messed up the ownership. Any further help will be appreciated. Thanks!

lah7, 2021-09-09:
The ownership of those files should be `root:plugdev` [owner:group]. Either try replugging in the device, or manually "fix" the permission in that folder:

    sudo chgrp plugdev *

The udev rules are supposed to do this automatically when the device is connected, and at start-up.

pulsar17, 2021-09-09:
@lah7 manually changing group permissions fixed the issue. Thanks a lot!

ascot-telmar, 2021-10-06:
I have myself experienced this issue today with the same product - with manually changing group permissions resolving the issue.

jerdie, 2021-10-14:
Same issue.

Tried building rebuilding from stable branch using guide from https://github.com/openrazer/openrazer/wiki/Building-a-package.
 
```
 lrwxrwxrwx 1 root root   9 Oct 14 18:45 usb-3044_MIIIW_USB_Recriver-event-if01 -> ../event8
lrwxrwxrwx 1 root root   9 Oct 14 18:45 usb-3044_MIIIW_USB_Recriver-event-kbd -> ../event7
lrwxrwxrwx 1 root root   9 Oct 14 18:45 usb-3044_MIIIW_USB_Recriver-if01-event-mouse -> ../event9
lrwxrwxrwx 1 root root   9 Oct 14 18:45 usb-3044_MIIIW_USB_Recriver-if01-mouse -> ../mouse1
lrwxrwxrwx 1 root root  10 Oct 14 18:45 usb-CN02FP438LG00084ARMJA02_Integrated_Webcam_HD_200901010001-event-if00 -> ../event22
lrwxrwxrwx 1 root root  10 Oct 14 18:45 usb-CN02FP438LG00084ARMJA02_Integrated_Webcam_HD_200901010001-event-if02 -> ../event23
lrwxrwxrwx 1 root root  10 Oct 14 18:45 usb-Razer_Razer_DeathAdder_Essential-event-if01 -> ../event14
lrwxrwxrwx 1 root root  10 Oct 14 18:45 usb-Razer_Razer_DeathAdder_Essential-event-mouse -> ../event12
lrwxrwxrwx 1 root root  10 Oct 14 18:45 usb-Razer_Razer_DeathAdder_Essential-if01-event-kbd -> ../event13
lrwxrwxrwx 1 root root  10 Oct 14 18:45 usb-Razer_Razer_DeathAdder_Essential-if02-event-kbd -> ../event15
lrwxrwxrwx 1 root root   9 Oct 14 18:45 usb-Razer_Razer_DeathAdder_Essential-mouse -> ../mouse2
```

```
Bus 001 Device 006: ID 1532:0098 Razer USA, Ltd Razer DeathAdder Essential
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0098 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer DeathAdder Essential
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
   Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
[... 208 lines cut ...]
2021-10-14 19:40:03 | razer                          | DEBUG    | Loaded device specification: RazerOuroboros -------------------------- (1532:0032)
2021-10-14 19:40:03 | razer                          | DEBUG    | Loaded device specification: RazerTaipan ----------------------------- (1532:0034)
2021-10-14 19:40:03 | razer                          | DEBUG    | Loaded device specification: RazerTartarus --------------------------- (1532:0201)
2021-10-14 19:40:03 | razer                          | DEBUG    | Loaded device specification: RazerTartarusChroma --------------------- (1532:0208)
2021-10-14 19:40:03 | razer                          | DEBUG    | Loaded device specification: RazerTartarusV2 ------------------------- (1532:022B)
2021-10-14 19:40:03 | razer                          | DEBUG    | Loaded device specification: RazerViper ------------------------------ (1532:0078)
2021-10-14 19:40:03 | razer                          | DEBUG    | Loaded device specification: RazerViper ------------------------------ (1532:0078)
2021-10-14 19:40:03 | razer                          | DEBUG    | Loaded device specification: RazerViperMini -------------------------- (1532:008A)
2021-10-14 19:40:03 | razer                          | DEBUG    | Loaded device specification: RazerViperUltimateWired ----------------- (1532:007A)
2021-10-14 19:40:03 | razer                          | DEBUG    | Loaded device specification: RazerViperUltimateWireless -------------- (1532:007B)
2021-10-14 19:40:03 | razer                          | DEBUG    | Adding razer.devices.getOffOnScreensaver method to DBus
2021-10-14 19:40:03 | razer                          | DEBUG    | Adding razer.daemon.version method to DBus
2021-10-14 19:40:03 | razer                          | DEBUG    | Adding razer.daemon.stop method to DBus
2021-10-14 19:40:03 | razer                          | DEBUG    | Adding razer.devices.supportedDevices method to DBus
2021-10-14 19:40:03 | razer                          | DEBUG    | Adding razer.devices.getDevices method to DBus
2021-10-14 19:40:03 | razer                          | DEBUG    | Adding razer.devices.getSyncEffects method to DBus
2021-10-14 19:40:03 | razer                          | DEBUG    | Adding razer.devices.syncEffects method to DBus
2021-10-14 19:40:03 | razer                          | DEBUG    | Adding razer.devices.enableTurnOffOnScreensaver method to DBus
2021-10-14 19:40:03 | razer                          | INFO     | Serving DBus

```

OS:
NAME="openSUSE Tumbleweed"
VERSION="20211008"

CaptAngryEyes, 2021-11-25:
I too have this issue. The razer id does not match with the one on the website and thus is not compatible with the polychromatic software.

PR #1642 (2021-12-04): Add support for Razer DeathAdder Essential (2021)
Adds support for Razer DeathAdder Essential with a new PID of 0098.
Fix #1566
closed 2021-12-04 by z3ntu