openrazer/openrazer #1568: Support for Blackwidow TE Chroma V2 Tenkeyless

adityabisoi, 2021-07-28:
### Device Info

* Name: Blackwidow TE Chroma V2 Tenkeyless
* Product Number: RZ03-02190700-R3M1
* Model Number: RZ03-0219

### Device Information
**Please post the output of
    lsusb -d 1532: -v**
```
Bus 001 Device 009: ID 1532:0209 Razer USA, Ltd BlackWidow Tournament Edition Chroma
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0209 BlackWidow Tournament Edition Chroma
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer BlackWidow Tournament Edition Chroma
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
```
**Please post the output of
    ls -lah /dev/input/by-id/**
```
total 0
drwxr-xr-x 2 root root 200 Jul 28 18:54 .
drwxr-xr-x 4 root root 640 Jul 28 18:54 ..
lrwxrwxrwx 1 root root   9 Jul 28 18:05 usb-04081-0009480017381010301_USB2.0_HD_UVC_WebCam_200901010001-event-if00 -> ../event7
lrwxrwxrwx 1 root root  10 Jul 28 18:05 usb-ITE_Tech._Inc._ITE_Device_8910_-event-kbd -> ../event13
lrwxrwxrwx 1 root root  10 Jul 28 18:05 usb-Logitech_USB_Receiver-if01-event-mouse -> ../event14
lrwxrwxrwx 1 root root   9 Jul 28 18:05 usb-Logitech_USB_Receiver-if01-mouse -> ../mouse3
lrwxrwxrwx 1 root root   9 Jul 28 18:54 usb-Razer_Razer_BlackWidow_Tournament_Edition_Chroma-event-kbd -> ../event8
lrwxrwxrwx 1 root root   9 Jul 28 18:54 usb-Razer_Razer_BlackWidow_Tournament_Edition_Chroma-if01-event-kbd -> ../event9
lrwxrwxrwx 1 root root  10 Jul 28 18:54 usb-Razer_Razer_BlackWidow_Tournament_Edition_Chroma-if02-event-mouse -> ../event17
lrwxrwxrwx 1 root root   9 Jul 28 18:54 usb-Razer_Razer_BlackWidow_Tournament_Edition_Chroma-if02-mouse -> ../mouse2
```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>

z3ntu, 2021-08-01:
The device is already supported:

> Razer BlackWidow Tournament Edition Chroma | 1532:0209

Does it not work for you?

closed 2021-08-01 by z3ntu

Hydranoyd, 2025-02-07:
> The device is already supported:
> 
> > Razer BlackWidow Tournament Edition Chroma | 1532:0209
> 
> Does it not work for you?

That is not the same device. The device you quoted is a BlackWidow TE V1, the request was for a BlackWidow TE V2.