openrazer/openrazer #1569: Support for Razer Hammerhead

cbeaujoin, 2021-07-29:
### Device Info

* Name: Razer Hammerhead
* Product Number:  AC31702
* Model Number:  RZ12-02970100

### Device Information

$ bluetoothctl
Agent registered
# connect 64:31:39:64:08:18
Attempting to connect to 64:31:39:64:08:18
Connection successful
[Razer Hammerhead True Wireless Earbuds]# info
Device 64:31:39:64:08:18 (public)
	Name: Razer Hammerhead True Wireless Earbuds
	Alias: Razer Hammerhead True Wireless Earbuds
	Class: 0x00240418
	Icon: audio-card
	Paired: yes
	Trusted: yes
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Serial Port               (00001101-0000-1000-8000-00805f9b34fb)
	UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: Advanced Audio Distribu.. (0000110d-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)

### Packet Captures

N/A

### Synapse Screenshots
N/A

# Issue
Input issue with A2DP Sink  configuration.
Variable behavior with HSP/HFP configuration

z3ntu, 2021-08-01:
What functionality does this device have? Any colors you can control? Also does the device show up as USB device in lsusb?

cbeaujoin, 2021-08-02:
Hi @z3ntu , this is earbuds, no color, doesn't show up as an USB device with lsusb.

z3ntu, 2021-08-02:
What part of the device do you want to control with openrazer then?

limacohen, 2021-10-30:
Just chiming in to say there are several features that can be controlled through the "Razer Audio" app, including lighting, and what I'm more interested in: "Quick Connect" and "Gaming Mode". Specially the first one, that allows fast switching between Bluetooth devices without having to disconnect prior. These are both listed under "Smart Link".
Currently you can toggle Gaming Mode using the touch controls, but Quick Connect remains locked inside the app, so it would be awesome to be able to automate this.

I don't have a pair of the original ones (v1) but I do have access to the X variant.

farion, 2024-10-02:
I'm using also Razer Hammerhead Hyperspeed. Please note, that they can be connected via BT and 2.4Ghz USB-C Dongle. I usally connect them via USB-C-Dongle. My earbuds do have the following features (in the app): Color-Settings (there is one LED per bud), Gaming-Mode, Quick-Connect, Battery-Level.

lsusb
```
Bus 001 Device 081: ID 1532:100e Razer USA, Ltd Razer Hammerhead HyperSpeed 2.4
```

headsetcontrol --dev -- --list --device 0x1532:0x100e
```
Device Found
  VendorID: 0x1532
 ProductID: 0x100e
  path: /dev/hidraw3
  serial_number: R002000003A8
  Manufacturer: (null)
  Product:      Razer Hammerhead HyperSpeed 2.4
  Interface:    0
  Usage-Page: 0xc Usageid: 0x1

Device Found
  VendorID: 0x1532
 ProductID: 0x100e
  path: /dev/hidraw3
  serial_number: R002000003A8
  Manufacturer: (null)
  Product:      Razer Hammerhead HyperSpeed 2.4
  Interface:    0
  Usage-Page: 0xff00 Usageid: 0x1
```

How can I help?

z3ntu, 2024-10-02:
@farion I think your device is #2302 even though the USB PID is different for some reason (0a3a vs 100e)

farion, 2024-10-14:
Probably you are right. The different PID is probably caused by a new revision. The original device in #2302 is RZ12-03820200, mine is RZ12-03820300 (is printed on the battery).