openrazer/openrazer #1570: Add support for  Razer Raiju Tournament Edition

khalilgharbaoui, 2021-07-31:
### Device Info

* Name: Razer Raiju Tournament Edition
* Product Number: Bus 001 Device 008: ID 1532:1007 Razer USA, Ltd
* Model Number: RZ06-0261

### Device Information
```

Bus 001 Device 008: ID 1532:1007 Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x1007 
  bcdDevice            1.01
  iManufacturer           1 Razer
  iProduct                2 Razer Raiju Tournament Edition Wired
  iSerial                 3 PM1921C04103639
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          225
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              0 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength           71
        bInCollection           2
        baInterfaceNr( 0)       1
        baInterfaceNr( 1)       2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0101 USB Streaming
        bAssocTerminal          6
        bNrChannels             2
        wChannelConfig     0x0003
          Left Front (L)
          Right Front (R)
        iChannelNames           0 
        iTerminal               0 
      AudioControl Interface Descriptor:
        bLength                10
[... 195 lines cut ...]
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               5
Device Status:     0x0000
  (Bus Powered)

```

```
Please post the output of
    lrwxrwxrwx 1 root root  11 jul 31 15:35 usb-Razer_Razer_Raiju_Tournament_Edition_Wired_PM1921C04103639-if03-event-joystick -> ../event257
lrwxrwxrwx 1 root root   6 jul 31 15:35 usb-Razer_Razer_Raiju_Tournament_Edition_Wired_PM1921C04103639-if03-joystick -> ../js2

```

### Packet Captures
... to be continued
### Synapse Screenshots
... to be continued

pablo3x6, 2021-11-18:
Any news on this device?

z3ntu, 2022-01-15:
Does this device have any lighting or anything else to configure?

pablo3x6, 2022-01-30:
> Does this device have any lighting or anything else to configure?

Just the main lights, no chroma.

khalilgharbaoui, 2022-01-31:
Notthing to add then..  closing this issue!

closed 2022-01-31 by khalilgharbaoui