openrazer/openrazer #1571: Deathadder V2 Pro does not work on reboot

Michael-Gallo, 2021-08-01:
My Razer Deathadder V2 Pro does not work wirelessly upon reboot and I need to unplug and replug the adapter every time.

I am running Artix Linux with Runit so if this can be explained by me not using the systemd service let me know (will probably give me an excuse to learn how to write a runit service).

Attached are the logs I get from "openrazer-daemon -Fv" , razer.txt is the logs upon first reboot and razer2.txt is the logs I get after replugging the adapter
[razer.txt](https://github.com/openrazer/openrazer/files/6912826/razer.txt)
[razer2.txt](https://github.com/openrazer/openrazer/files/6912827/razer2.txt)
the openrazer packages I have installed are
`local/openrazer-daemon-git 3.0.1.r20.g5ddf10eb-1
    Userspace daemon that abstracts access to the kernel driver. Provides a DBus service for applications to use.
local/openrazer-driver-dkms-git 3.0.1.r20.g5ddf10eb-1
    Kernel driver for Razer devices (DKMS-variant)
local/openrazer-meta-git 3.0.1.r20.g5ddf10eb-1
    Meta package for installing all required openrazer packages.
local/python-openrazer-git 3.0.1.r20.g5ddf10eb-1
    Python library for accessing the Razer daemon from Python.
`

I am in the plugdev group
Module openrazer-driver/3.0.1 is installed on the correct kernel
lsusb always shows the deathadder

Mouse works on reboot in BIOS so I know this is an OS/driver issue

z3ntu, 2021-08-11:
Does the mouse work with the openrazer-daemon not running?

Michael-Gallo, 2021-08-11:
Yes, provided I replug the mouse it'll work whether openrazer-daemon is running or not

z3ntu, 2021-08-12:
That's weird. Did you try uninstalling openrazer completely and see if the situation changes?

Michael-Gallo, 2021-08-15:
I have not but somehow this problem went away on its own. Don't remember if I updated before the last reboot before it started working again.

closed 2021-08-15 by Michael-Gallo