openrazer/openrazer #1573: When will Openrazer support Naga X mouse??  (sigh)

motoryzen, 2021-08-04:
When I bought this Razer Naga X Mouse, I didn't know it wasn't supported.  I just want a way to do the following things in Linux Mint 20.2 Cinnamon:

1.  Set my preferred color
2.   Set the 1-12 side buttons to do what I want ( 11 as backspace ...1 as Home and 3 as End).   *sigh*

z3ntu, 2021-08-11:
Please provide the output of `lsusb | grep 1532`

motoryzen, 2021-08-12:
Sorry..forgot to include inxi -Fxxxrz report
`System:    Kernel: 5.11.0-25-generic x86_64 bits: 64 compiler: N/A Desktop: Cinnamon 5.0.5 wm: muffin 5.0.1 
           dm: LightDM 1.30.0 Distro: Linux Mint 20.2 Uma base: Ubuntu 20.04 focal 
Machine:   Type: Desktop Mobo: ASRock model: X570 Taichi serial: <filter> BIOS: American Megatrends v: P4.40 
           date: 06/23/2021 
CPU:       Topology: 12-Core model: AMD Ryzen 9 5900X bits: 64 type: MT MCP arch: Zen 3 L2 cache: 6144 KiB 
           flags: avx avx2 lm nx pae sse sse2 sse3 sse4_1 sse4_2 sse4a ssse3 svm bogomips: 177600 
           Speed: 3700 MHz min/max: N/A Core speeds (MHz): 1: 3700 2: 3700 3: 3700 4: 3700 5: 3700 6: 3700 7: 3610 8: 3700 
           9: 3700 10: 3700 11: 3700 12: 3700 13: 3700 14: 3700 15: 3700 16: 3700 17: 3700 18: 3700 19: 3700 20: 3700 
           21: 3700 22: 3700 23: 3700 24: 3700 
Graphics:  Device-1: Advanced Micro Devices [AMD/ATI] Ellesmere [Radeon RX 470/480/570/570X/580/580X/590] 
           vendor: Sapphire Limited Nitro+ driver: amdgpu v: kernel bus ID: 0b:00.0 chip ID: 1002:67df 
           Display: x11 server: X.Org 1.20.11 driver: amdgpu,ati unloaded: fbdev,modesetting,vesa tty: N/A 
           OpenGL: renderer: Radeon RX 580 Series (POLARIS10 DRM 3.40.0 5.11.0-25-generic LLVM 12.0.1) 
           v: 4.6 Mesa 21.2.0 - kisak-mesa PPA direct render: Yes 
Audio:     Device-1: AMD Ellesmere HDMI Audio [Radeon RX 470/480 / 570/580/590] vendor: Sapphire Limited 
           driver: snd_hda_intel v: kernel bus ID: 0b:00.1 chip ID: 1002:aaf0 
           Device-2: Advanced Micro Devices [AMD] Starship/Matisse HD Audio vendor: ASRock driver: snd_hda_intel v: kernel 
           bus ID: 0e:00.4 chip ID: 1022:1487 
           Sound Server: ALSA v: k5.11.0-25-generic 
Network:   Device-1: Intel I211 Gigabit Network vendor: ASRock driver: igb v: kernel port: f000 bus ID: 06:00.0 
           chip ID: 8086:1539 
           IF: enp6s0 state: up speed: 1000 Mbps duplex: full mac: <filter> 
Drives:    Local Storage: total: 23.47 TiB used: 16.11 TiB (68.6%) 
           ID-1: /dev/nvme0n1 vendor: Intel model: SSDPEDMW800G4 size: 745.21 GiB speed: 31.6 Gb/s lanes: 4 
           serial: <filter> rev: 8EV10174 scheme: MBR 
           ID-2: /dev/sda vendor: Toshiba model: HDWR11A size: 9.10 TiB speed: 6.0 Gb/s rotation: 7200 rpm 
           serial: <filter> rev: 0603 
           ID-3: /dev/sdb vendor: Samsung model: SSD 860 EVO 4TB size: 3.64 TiB speed: 6.0 Gb/s serial: <filter> rev: 4B6Q 
           scheme: GPT 
           ID-4: /dev/sdc vendor: Samsung model: SSD 860 EVO 1TB size: 931.51 GiB speed: 6.0 Gb/s serial: <filter> 
           rev: 3B6Q scheme: MBR 
           ID-5: /dev/sdd type: USB vendor: Toshiba model: HDWR11A size: 9.10 TiB rotation: 7200 rpm serial: <filter> 
           rev: 0502 scheme: GPT 
Partition: ID-1: / size: 732.02 GiB used: 59.75 GiB (8.2%) fs: ext4 dev: /dev/nvme0n1p5 
Sensors:   System Temperatures: cpu: 45.5 C mobo: N/A gpu: amdgpu temp: 49 C 
           Fan Speeds (RPM): N/A gpu: amdgpu fan: 838 
Repos:     No active apt repos in: /etc/apt/sources.list 
           Active apt repos in: /etc/apt/sources.list.d/additional-repositories.list 
           1: deb https://dl.winehq.org/wine-builds/ubuntu/ focal main
           Active apt repos in: /etc/apt/sources.list.d/caldas-lopes-ppa-focal.list 
           1: deb http://ppa.launchpad.net/caldas-lopes/ppa/ubuntu focal main
           Active apt repos in: /etc/apt/sources.list.d/cappelikan-ppa-focal.list 
           1: deb http://ppa.launchpad.net/cappelikan/ppa/ubuntu focal main
           2: deb-src http://ppa.launchpad.net/cappelikan/ppa/ubuntu focal main
           Active apt repos in: /etc/apt/sources.list.d/google-chrome.list 
           1: deb [arch=amd64] http://dl.google.com/linux/chrome/deb/ stable main
           Active apt repos in: /etc/apt/sources.list.d/kisak-kisak-mesa-focal.list 
           1: deb http://ppa.launchpad.net/kisak/kisak-mesa/ubuntu focal main
           No active apt repos in: /etc/apt/sources.list.d/lunarg-vulkan-1.2.162-focal.list 
           Active apt repos in: /etc/apt/sources.list.d/lutris-team-lutris-focal.list 
           1: deb http://ppa.launchpad.net/lutris-team/lutris/ubuntu focal main
           Active apt repos in: /etc/apt/sources.list.d/official-package-repositories.list 
           1: deb http://packages.linuxmint.com uma main upstream import backport
           2: deb http://archive.ubuntu.com/ubuntu focal main restricted universe multiverse
           3: deb http://archive.ubuntu.com/ubuntu focal-updates main restricted universe multiverse
           4: deb http://archive.ubuntu.com/ubuntu focal-backports main restricted universe multiverse
           5: deb http://security.ubuntu.com/ubuntu/ focal-security main restricted universe multiverse
           6: deb http://archive.canonical.com/ubuntu/ focal partner
           Active apt repos in: /etc/apt/sources.list.d/official-source-repositories.list 
           1: deb-src http://packages.linuxmint.com uma main upstream import backport
           2: deb-src http://archive.ubuntu.com/ubuntu focal main restricted universe multiverse
           3: deb-src http://archive.ubuntu.com/ubuntu focal-updates main restricted universe multiverse
           4: deb-src http://archive.ubuntu.com/ubuntu focal-backports main restricted universe multiverse
           5: deb-src http://security.ubuntu.com/ubuntu/ focal-security main restricted universe multiverse
[... cut ...]
`Failed to fetch http://ppa.launchpad.net/openshot.developers/ppa/ubuntu/dists/focal/InRelease  Could not connect to ppa.launchpad.net:80 (91.189.95.85), connection timed out [IP: 91.189.95.85 80]Failed to fetch http://ppa.launchpad.net/polychromatic/stable/ubuntu/dists/focal/InRelease  Unable to connect to ppa.launchpad.net:http: [IP: 91.189.95.85 80]Failed to fetch http://ppa.launchpad.net/rael-gc/ubuntu-xboxdrv/ubuntu/dists/focal/InRelease  Unable to connect to ppa.launchpad.net:http: [IP: 91.189.95.85 80]Some index files failed to download. They have been ignored, or old ones used instead.`

Internet isn't the problem.  ( sigh).  Rabbit hole time xD

z3ntu, 2021-08-12:
Duplicate of #1463

The "apt update" error seems to be caused by your internet connection, and the ppa infra is also out of my control.

closed 2021-08-12 by z3ntu

motoryzen, 2021-08-12:
*facepalm*

MY internet connection is NOT the problem as I have already stated.

lah7, 2021-08-12:
> `connection timed out`
> `Unable to connect to ppa.launchpad.net:http: [IP: 91.189.95.85 80]Some index files failed to download. They have been ignored, or old ones used instead.`

It's possible that Launchpad was unavailable at the time. Try running `sudo apt update` in the Terminal now and see if it connects.

If you still get a time out, then it is likely a network/routing issue. Try running `ping ppa.launchpad.net`. It works on our end. Could something be blocking/filtering on your side?

Also, per above (#1463), even when you get the software up and running, the Naga X mouse isn't supported yet.

motoryzen, 2021-08-12:
Results of ping ppa.launchpad.net
`PING ppa.launchpad.net (91.189.95.85) 56(84) bytes of data.
64 bytes from ganondorf.canonical.com (91.189.95.85): icmp_seq=1 ttl=32 time=135 ms
64 bytes from ganondorf.canonical.com (91.189.95.85): icmp_seq=2 ttl=32 time=134 ms
64 bytes from ganondorf.canonical.com (91.189.95.85): icmp_seq=4 ttl=32 time=135 ms
64 bytes from ganondorf.canonical.com (91.189.95.85): icmp_seq=5 ttl=32 time=136 ms
64 bytes from ganondorf.canonical.com (91.189.95.85): icmp_seq=6 ttl=32 time=134 ms
64 bytes from ganondorf.canonical.com (91.189.95.85): icmp_seq=8 ttl=32 time=135 ms
64 bytes from ganondorf.canonical.com (91.189.95.85): icmp_seq=9 ttl=32 time=132 ms
64 bytes from ganondorf.canonical.com (91.189.95.85): icmp_seq=10 ttl=32 time=130 ms
^Z
[1]+  Stopped                 ping ppa.launchpad.net`

And yes I've done sudo apt update before doing those commands, when instructed and even after as well.  And I have no special blocks or filtering going on in my router and don't have a VPN either. NO

motoryzen, 2021-08-12:
Yeah I'm fine for now with the naga x mouse not yet being supported.  I just would like to get the keyboard part going though

motoryzen, 2021-08-12:
Running sudo apt update yields:
`Ign:1 http://packages.linuxmint.com uma InRelease
Hit:2 http://dl.google.com/linux/chrome/deb stable InRelease                                                                 
Hit:3 http://packages.linuxmint.com uma Release                                                                              
Hit:5 https://dl.winehq.org/wine-builds/ubuntu focal InRelease                                                               
Get:6 http://security.ubuntu.com/ubuntu focal-security InRelease [114 kB]                                                    
Hit:7 http://ppa.launchpad.net/caldas-lopes/ppa/ubuntu focal InRelease                          
Hit:8 http://archive.ubuntu.com/ubuntu focal InRelease                                                                       
Hit:9 http://archive.canonical.com/ubuntu focal InRelease                                                                    
Get:10 http://archive.ubuntu.com/ubuntu focal-updates InRelease [114 kB]                                  
Hit:11 http://ppa.launchpad.net/cappelikan/ppa/ubuntu focal InRelease                      
Hit:12 http://ppa.launchpad.net/kisak/kisak-mesa/ubuntu focal InRelease                                               
Get:13 http://archive.ubuntu.com/ubuntu focal-backports InRelease [101 kB]                                            
Hit:14 http://ppa.launchpad.net/lutris-team/lutris/ubuntu focal InRelease                                 
Get:15 http://archive.ubuntu.com/ubuntu focal-updates/universe Sources [182 kB]                           
Get:16 http://archive.ubuntu.com/ubuntu focal-updates/main amd64 c-n-f Metadata [13.9 kB]                         
Get:17 http://archive.ubuntu.com/ubuntu focal-updates/universe amd64 Packages [844 kB]                       
Get:18 http://security.ubuntu.com/ubuntu focal-security/universe Sources [68.2 kB]                 
Get:19 http://ppa.launchpad.net/openrazer/daily/ubuntu focal InRelease [18.0 kB]                                      
Get:20 http://security.ubuntu.com/ubuntu focal-security/universe amd64 Packages [636 kB]              
Get:21 http://archive.ubuntu.com/ubuntu focal-updates/universe i386 Packages [626 kB]                
Err:19 http://ppa.launchpad.net/openrazer/daily/ubuntu focal InRelease                                                       
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 073E051D7B2AEE37
Hit:22 http://ppa.launchpad.net/openshot.developers/ppa/ubuntu focal InRelease                             
Get:23 http://security.ubuntu.com/ubuntu focal-security/universe i386 Packages [506 kB]
Hit:24 http://ppa.launchpad.net/rael-gc/ubuntu-xboxdrv/ubuntu focal InRelease   
Get:25 http://security.ubuntu.com/ubuntu focal-security/universe amd64 c-n-f Metadata [12.1 kB]
Get:26 http://ppa.launchpad.net/openrazer/stable/ubuntu focal InRelease [18.1 kB]    
Err:26 http://ppa.launchpad.net/openrazer/stable/ubuntu focal InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 073E051D7B2AEE37
Get:27 http://ppa.launchpad.net/polychromatic/stable/ubuntu focal InRelease [18.0 kB]
Err:27 http://ppa.launchpad.net/polychromatic/stable/ubuntu focal InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 96B9CD7C22E2C8C5
Reading package lists... Done
W: GPG error: http://ppa.launchpad.net/openrazer/daily/ubuntu focal InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 073E051D7B2AEE37
E: The repository 'http://ppa.launchpad.net/openrazer/daily/ubuntu focal InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
W: GPG error: http://ppa.launchpad.net/openrazer/stable/ubuntu focal InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 073E051D7B2AEE37
E: The repository 'http://ppa.launchpad.net/openrazer/stable/ubuntu focal InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
W: GPG error: http://ppa.launchpad.net/polychromatic/stable/ubuntu focal InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 96B9CD7C22E2C8C5
E: The repository 'http://ppa.launchpad.net/polychromatic/stable/ubuntu focal InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.`

z3ntu, 2021-08-13:
Try running these two commands
```
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 073E051D7B2AEE37
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 96B9CD7C22E2C8C5
```
As you can see your computer can reach launchpad.net now and it's a different error now.

motoryzen, 2021-08-13:
results of 1st command:
`Executing: /tmp/apt-key-gpghome.TvVYtIvxP4/gpg.1.sh --keyserver keyserver.ubuntu.com --recv-keys 073E051D7B2AEE37
gpg: key 073E051D7B2AEE37: public key "Launchpad PPA for OpenRazer" imported
gpg: Total number processed: 1
gpg:               imported: 1
gpg: invalid key resource URL '/etc/apt/trusted.gpg.d/home:erigas:cpupower-gui.gpg'
gpg: keyblock resource '(null)': General error
gpg: key 76F1A20FF987672F: 1 signature not checked due to a missing key
gpg: key 7721F63BD38B4796: 2 signatures not checked due to missing keys
gpg: key C5BDBD015C83816E: 1 signature not checked due to a missing key
gpg: key A6616109451BBBF2: 12 signatures not checked due to missing keys
gpg: key 3B4FE6ACC0B21F32: 3 signatures not checked due to missing keys
gpg: key D94AA3F0EFE21092: 3 signatures not checked due to missing keys
gpg: key C8CAB6595FDFF622: 2 signatures not checked due to missing keys
gpg: key 871920D1991BC93C: 1 signature not checked due to a missing key
gpg: Total number processed: 16
gpg:       skipped new keys: 16`
---- results of 2nd command ==-=
`Executing: /tmp/apt-key-gpghome.lGGavghYGx/gpg.1.sh --keyserver keyserver.ubuntu.com --recv-keys 96B9CD7C22E2C8C5
gpg: key 96B9CD7C22E2C8C5: public key "Launchpad PPA for Polychromatic" imported
gpg: Total number processed: 1
gpg:               imported: 1
gpg: invalid key resource URL '/etc/apt/trusted.gpg.d/home:erigas:cpupower-gui.gpg'
gpg: keyblock resource '(null)': General error
gpg: key 76F1A20FF987672F: 1 signature not checked due to a missing key
gpg: key 7721F63BD38B4796: 2 signatures not checked due to missing keys
gpg: key C5BDBD015C83816E: 1 signature not checked due to a missing key
gpg: key A6616109451BBBF2: 12 signatures not checked due to missing keys
gpg: key 3B4FE6ACC0B21F32: 3 signatures not checked due to missing keys
gpg: key D94AA3F0EFE21092: 3 signatures not checked due to missing keys
gpg: key C8CAB6595FDFF622: 2 signatures not checked due to missing keys
gpg: key 871920D1991BC93C: 1 signature not checked due to a missing key
gpg: Total number processed: 16
gpg:       skipped new keys: 16`

So I tried going through all the instructed steps to install polychromatic again per that site and the same exact results are the case:

Sudo add-repostiory..blah blah
`sudo add-apt-repository ppa:polychromatic/stable
You are about to add the following PPA:
 
Polychromatic, an open source RGB lighting management front-end application
to customise compatible peripherals.

This PPA contains tried & tested builds of the application.

 More info: https://launchpad.net/~polychromatic/+archive/ubuntu/stable
Press Enter to continue or Ctrl+C to cancel

Executing: /tmp/apt-key-gpghome.JPJgf8iI1W/gpg.1.sh --keyserver hkps://keyserver.ubuntu.com:443 --recv-keys C0D54C34D00160459588000E96B9CD7C22E2C8C5
gpg: key 96B9CD7C22E2C8C5: public key "Launchpad PPA for Polychromatic" imported
gpg: Total number processed: 1
gpg:               imported: 1
gpg: invalid key resource URL '/etc/apt/trusted.gpg.d/home:erigas:cpupower-gui.gpg'
gpg: keyblock resource '(null)': General error
gpg: key 76F1A20FF987672F: 1 signature not checked due to a missing key
gpg: key 7721F63BD38B4796: 2 signatures not checked due to missing keys
gpg: key C5BDBD015C83816E: 1 signature not checked due to a missing key
gpg: key A6616109451BBBF2: 12 signatures not checked due to missing keys
gpg: key 3B4FE6ACC0B21F32: 3 signatures not checked due to missing keys
gpg: key D94AA3F0EFE21092: 3 signatures not checked due to missing keys
gpg: key C8CAB6595FDFF622: 2 signatures not checked due to missing keys
gpg: key 871920D1991BC93C: 1 signature not checked due to a missing key
gpg: Total number processed: 16
gpg:       skipped new keys: 16`

sudo apt update

`Ign:1 http://packages.linuxmint.com uma InRelease
Get:2 http://security.ubuntu.com/ubuntu focal-security InRelease [114 kB]                                                    
[... 21 lines cut ...]
Err:18 http://ppa.launchpad.net/polychromatic/stable/ubuntu focal InRelease
  Unable to connect to ppa.launchpad.net:http:
Err:19 http://ppa.launchpad.net/rael-gc/ubuntu-xboxdrv/ubuntu focal InRelease
  Unable to connect to ppa.launchpad.net:http:
Fetched 114 kB in 30s (3,751 B/s)
Reading package lists... Done
Building dependency tree       
Reading state information... Done
83 packages can be upgraded. Run 'apt list --upgradable' to see them.
W: Failed to fetch http://ppa.launchpad.net/caldas-lopes/ppa/ubuntu/dists/focal/InRelease  Could not connect to ppa.launchpad.net:80 (91.189.95.85), connection timed out
W: Failed to fetch http://ppa.launchpad.net/cappelikan/ppa/ubuntu/dists/focal/InRelease  Unable to connect to ppa.launchpad.net:http:
W: Failed to fetch http://ppa.launchpad.net/kisak/kisak-mesa/ubuntu/dists/focal/InRelease  Unable to connect to ppa.launchpad.net:http:
W: Failed to fetch http://ppa.launchpad.net/lutris-team/lutris/ubuntu/dists/focal/InRelease  Unable to connect to ppa.launchpad.net:http:
W: Failed to fetch http://ppa.launchpad.net/openrazer/daily/ubuntu/dists/focal/InRelease  Unable to connect to ppa.launchpad.net:http:
W: Failed to fetch http://ppa.launchpad.net/openrazer/stable/ubuntu/dists/focal/InRelease  Unable to connect to ppa.launchpad.net:http:
W: Failed to fetch http://ppa.launchpad.net/openshot.developers/ppa/ubuntu/dists/focal/InRelease  Unable to connect to ppa.launchpad.net:http:
W: Failed to fetch http://ppa.launchpad.net/polychromatic/stable/ubuntu/dists/focal/InRelease  Unable to connect to ppa.launchpad.net:http:
W: Failed to fetch http://ppa.launchpad.net/rael-gc/ubuntu-xboxdrv/ubuntu/dists/focal/InRelease  Unable to connect to ppa.launchpad.net:http:
W: Some index files failed to download. They have been ignored, or old ones used instead.`

Then  sudo apt install polychromatic
`Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package polychromatic`

motoryzen, 2021-08-13:
Update 6:30 pm central time 8-13-21

By some odd random miracle...I got at least Step 1 of my goal..working. 

Upon fresh clean install of LM 20.2 Cinnamon. on my backup pc ( 4790k unit)... I did all updates except Bluez, rebooted,  updated to absolutely latest available Kernel version in UPdate manager and all Kernel-coherent updates.  

Then I followed instructions of https://openrazer.github.io/#ubuntu to try installing openrazer-meta    .   ....  Next....  Try to install polychromatic via that link's instructions from here -->>>  https://polychromatic.app/ clicking on Download and following instructions for all three varieties. stable,  testing and edge ( received almost similar errors but not quite as many..oddly). said "unable to locate Polychromatic package" just like with my main PC i'm typing this message from

Finally I click on the Razergenie link on openrazer.github.io/#download   which took me to  http://github.com/z3ntu/RazerGenie    ..... I scrolled down to the -- Installation -- section and selected "Flathub"  straight to the rightvia the Flatpack option in white font......which took me to    http://flathub.org/apps/details/xyz.z3ntu.razergenie        .    obvious  install button right up top. 

And All color options..just work ( of course I haven't tested the Naga X mouse yet.... I'll report back when  after I reconnect the keyboard and mouse to my main rig)

motoryzen, 2021-08-13:
Now if only I could make it to where the Razer genie ( be it the flathub version or the regular version...BOTH.work btw  :D   ) would default to Static instead of Breath-single   *shrugs*.  I must manually always change that as of for the time being.   But  main mission overall...solved.

Goal 2.  Worst case, if it still doesn't recognize my Naga X mouse = I can boot up my windows drive connect the mouse to that PC and edit the side button's behavior ( keyboard actions mainly..like backspace..home..etc ) and it saves that onboard the mouse.    I'm good there.

lah7, 2021-08-14:
I noticed in the output above there's an error likely why repositories are not working on your system:

> gpg: invalid key resource URL '/etc/apt/trusted.gpg.d/home:erigas:cpupower-gui.gpg'
> gpg: keyblock resource '(null)': General error

^^ It's likely to be fixable without reinstalling the OS by looking up the error.

Just a note - RazerGenie worked because you've installed a Flatpak, which isn't using the same "Debian package/repository" system.

By the way, ideally only be one of the repositories should be enabled (`stable/daily` for OpenRazer, and `stable/testing/edge` for Polychromatic). It is possible to download the `.deb` files from Launchpad or GitHub release notes (for some projects) but a repository would be better so they stay up-to-date. I believe Flatpak will update automatically.

---
Regarding the device "remembering" the effect across computers, that's expected as OpenRazer uses hardware effects. Razer Synapse 3 performs its operations in software which usually causes the mouse to reset on next boot.

>  razer genie on this main rig doesn't recognize my keyboard nor the naga x mouse

Did you run this command? If not, run and logout/reboot - the keyboard at least should appear afterwards:

    sudo gpasswd -a $USER plugdev