openrazer/openrazer #1574: Support for Breathing Logo on newer Razer Blades

j-maynard, 2021-08-05:
The newer Razer Blades support a breathing logo effect.  Currently OpenRazer only supports On/Off for the logo.  Raising this issue to track support for this feature.  Below is a screen shot from the lighting options for the Blade Advamce Mid 2021 4k Model.

[image]

z3ntu, 2021-08-11:
Please provide USB packet captures. If possible turn off all rgb lighting except for the logo one when doing that.

j-maynard, 2021-10-11:
[blade-advanced-mid-2021-breathing.zip](https://github.com/openrazer/openrazer/files/7320372/blade-advanced-mid-2021-breathing.zip)
Attached is a pcap for the breathing logo on the Mid-2021 Razer Blade Advanced.  This pcap follows the following actions:
* Off -> Static
* Static -> Off
* Off -> Breathing
* Breathing -> Off

I've included the off to static and back again as reference points.  This should help with identifying what's going on with the breathing logo.  From my own observations so far its not done in software but actually done with a single request rather than keep updating the brightness of the logo in software.

brian-mcnamara, 2022-04-09:
Just to chime in here, it looks like this was prevented in #423 as an undocumented feature. Based on the comments, its simply a tri-state (0,1,2) value.

closed 2022-09-08 by github-actions[bot]