openrazer/openrazer #1575: pylib: Pulsate/blinking missing

lah7, 2021-08-06:
Devices like BlackWidow Ultimate 2013, Orbweaver and Blackwidow Stealth have a **Pulsate** effect. Under the Python library (or under fake driver at least), the pulsate function doesn't exist, even though it's supposed to be there.

```
AttributeError: 'RazerFX' object has no attribute 'pulsate'
```

Oddly enough, it **is** there:

https://github.com/openrazer/openrazer/blob/5ddf10eb85a803c83fa31465c1df4bdf0e860a17/pylib/openrazer/client/fx.py#L770-L786

Likewise,  `blinking` might also be affected as a `grep` search doesn't reveal the capable mice in the fake driver configs.

z3ntu, 2021-08-11:
These two functions don't seem to be used by the pylib
```
@endpoint('razer.device.lighting.bw2013', 'setPulsate')
@endpoint('razer.device.lighting.bw2013', 'setStatic')
```
PRs welcome ;) But a rewrite of everything is really overdue :/

Not sure what the existing `pulsate` function in the pylib is supposed to call...

lah7, 2021-08-21:
I was having a poke as multiple users are reporting this issue, but no luck. What I can dig out:

* The functions are in fact missing in `RazerFX` class (the remains above are in `SingleLed`)
* Daemon determines the capability as `bw_set_pulsate` and `bw_set_static` - although only `lighting_pulsate` is being set.
* These are on D-Bus endpoint `razer.device.lighting.bw2013` instead of `razer.device.lighting.chroma`

I think I'll workaround in the app for now by echoing to the driver file directly, as `_available_features` lists `razer.device.lighting.bw2013` to know.