openrazer/openrazer #1579: Analog key input support for Huntsman V2 Analog

z3ntu, 2021-08-15:
Switch the device to driver mode and handle the custom events there.

We get a value between 0x00 and 0xFF for each key depending on how far it is pressed. This needs to be either converted to normal keypresses on an either predefined value, or user configurable through sysfs.

Not sure if it is wanted to pass the values as-is to user space for it to be handled there - analog keyboards are probably too niche for that at the moment.

meeuw, 2022-07-06:
How about passing the analog values as-is to user space and translate the custom events to [injected input](https://python-evdev.readthedocs.io/en/latest/tutorial.html#injecting-input) through evdev?

Would you accept a PR which adds this capability to the openrazer daemon?

I'm thinking of buying an analog keyboard to, next to normal keyboard usage, control the mouse using hjkl and use the key pressure to control the speed.

meeuw, 2022-07-09:
Sound reasonable yes. I found this project: https://github.com/kempniu/evmapy which does most of my suggestion already.

When I hack RazerHuntsmanV2Analog into `_RippleKeyboard` would it already send the analog key presses as events? Or does this still require some work in the keyboard kernel module? (I haven't bought one yet so I cannot test).

Btw, I'm considering https://www.razer.com/gaming-keyboards/razer-huntsman-mini-analog . This keyboard isn't supported at all yet by openrazer right?

meeuw, 2022-07-16:
Haven't bought it yet but I figured evmapy wouldn't work because it cannot do "hold key" (repeat key when it's hold).

I'm fiddling a bit with other analog input devices I have (8bitdo gamepad) and that has some interesting feature I haven't thought of: the "trigger button" is analog but when you fully trigger this button it registers a "key down" event.

In your description I read that "or" as an exclusive or but I think we should do both: send analog events AND send key up/down events.

Trigger button:
[image]

z3ntu, 2022-07-23:
@meeuw Switching the keyboard to "driver mode" (which is required for the kernel to get any analog events and not just standard key presses as happens in the current "device mode") would require lots of work in the kernel still. I believe I had some code somewhere that would at least print the 0x00 -> 0xFF value for each key but not much more than that.

meeuw, 2022-07-24:
> @meeuw Switching the keyboard to "driver mode" (which is required for the kernel to get any analog events and not just standard key presses as happens in the current "device mode") would require [...]

Would be awesome if you could share that, in the meantime I'll try to find to do it myself.

meeuw, 2022-07-25:
I'm able to receive the analog events from razer_raw_event, I'm struggling a bit to translate these to ABS events, Linux doesn't seem to like a keyboard which behaves like a gamepad... Key events are fine so I guess I'll start with that.

I still wonder if I can use razer_event, all keys are mapped to scancode 0 but ARE registered...

meeuw, 2022-07-25:
:partying_face: still work in progress but I the keyboard is working in "driver mode" !

https://github.com/meeuw/openrazer/commits/huntsman-mini-analog

meeuw, 2022-07-30:
FYI: Implemented "user configurable through sysfs", also added absolute events for (almost) all keys!

Grundik, 2022-11-26:
Maybe its possible to implement virtual gamepad, and allow user to map keys to its buttons/axises?..

meeuw, 2022-11-29:
> Maybe its possible to implement virtual gamepad, and allow user to map keys to its buttons/axises?..

Definitely possible yes but we would need to define an API for mapping razer custom key codes to absolute axes and need a user-space tool to initialize this.

I'm afraid this would be too much for the openrazer project.

pkreuzt, 2024-07-26:
Sorry for bumping this old issue, but I think there's something that hasn't been said yet.

Most people wouldn't probably be interested in the analog feature for the use case of progressive triggers and the like, but this feature is also used to regulate the "sensitivity" of the keyboard. I mean the general key travel distance required for event activation. I don't think this is not a niche feature at all and should be easier to implement. Any chance of having this done in the future?

meeuw, 2024-07-26:
Hi @pkreuzt, it's all or nothing with the analog feature. The keyboard has two modes, analog and digital. In digital mode the sensitivity isn't configurable. In analog mode no normal key reports are sent (only razar custom reports containing how far the key is pressed). These need to be translated to normal key reports and can consider a specific sensitivity.

For razer-analog I implemented this here: https://github.com/meeuw/razer-analog/blob/47ded849c22886e6371a62746ebf49a2fcae0eaa/razer_analog/driver.py#L257

I don't think this is easier to implement; it still requires full analog support.

meeuw, 2024-07-29:
Hi @Grundik, don't put your hopes up to high, for what I could reverse engineer the profiles are just stupid UUIDs pointing to a configuration file on your computer.

In other words, the settings of a profile aren't stored on the keyboard.

If you connect your razer keyboard to another Windows (virtual) machine, all settings are gone (maybe unless you sync using the cloud).

But there are some basic things that are stored on the keyboard like remapping of keys and the RGB settings.

pkreuzt, 2024-07-30:
> I've done some testings:
> 1. my memory failed me, Fn + 1/2/3/4 does not change profiles by default, it was my own mappings, done so long ago, I forgot about it. Without Synapse these mappings do something with RGB, but not switching profiles, that had mislead me into thinking its working as expected;
> 2. when keyboard works in hardware mode it just uses first onboard profile (red one);
> 3. actuation points ARE saved into onboard profile, they are definitely processed in hardware, even per-key settings are saved. and that works fine.

It seems to store even separate per-key activation point, which seems excessive. There must be some internal structure for this data.

meeuw, 2024-07-30:
You can easily try to set a different actuation point in Synapse, reboot to Linux and see if it's still active.

If it is it shouldn't be too hard to implement in openrazer.

pkreuzt, 2024-07-30:
It is. I used a VM with USB passthrough to set activation point, then powered off VM and rebooted host. Setting seems persistent.

cidious, 2026-09-28:
Sorry for bumping, just made the same user-space driver as meeuw did. It works with full-sized Razer Huntsman V2 Analog, but can be used with other Razer Analog keyboards by using the calibrating tool included in the package. It can coexist with Openrazer with RGB managing, it handles layout switches with optional LED indication. Configurable press / release thresholds. PRs are welcome.
https://github.com/cidious/razer-analog-linux