openrazer/openrazer #1581: Support for Razer Blade 15 Base (mid-2021)

qhartman, 2021-08-15:
### Device Info

* Name: Blade 15 Base Model (Mid 2021)
* Product Number: RZ09-0410BE22
* Model Number: RZ09-0410

### Device Information
```
Bus 003 Device 003: ID 1532:027a Razer USA, Ltd Razer Blade
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x027a 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Blade
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)

```

```
total 0
drwxr-xr-x 2 root root  340 Aug 15 11:54 .
drwxr-xr-x 4 root root 1.1K Aug 15 11:54 ..
lrwxrwxrwx 1 root root   11 Aug 15 09:04 usb-046d_0821_D9E2A9E0-event-if02 -> ../event269
lrwxrwxrwx 1 root root    9 Aug 13 12:30 usb-Azurewave_USB_Camera_0x0001-event-if00 -> ../event4
lrwxrwxrwx 1 root root   10 Aug 15 09:04 usb-Logitech_Gaming_Mouse_G502_1671375A3536-event-if01 -> ../event17
lrwxrwxrwx 1 root root   10 Aug 15 09:04 usb-Logitech_Gaming_Mouse_G502_1671375A3536-event-mouse -> ../event15
lrwxrwxrwx 1 root root   10 Aug 15 09:04 usb-Logitech_Gaming_Mouse_G502_1671375A3536-if01-event-kbd -> ../event16
lrwxrwxrwx 1 root root    9 Aug 15 09:04 usb-Logitech_Gaming_Mouse_G502_1671375A3536-mouse -> ../mouse3
lrwxrwxrwx 1 root root   10 Aug 13 12:30 usb-Razer_Razer_Blade-event-if01 -> ../event10
lrwxrwxrwx 1 root root    9 Aug 13 12:30 usb-Razer_Razer_Blade-event-kbd -> ../event8
lrwxrwxrwx 1 root root    9 Aug 13 12:30 usb-Razer_Razer_Blade-if01-event-kbd -> ../event9
lrwxrwxrwx 1 root root   10 Aug 13 12:30 usb-Razer_Razer_Blade-if02-event-mouse -> ../event13
lrwxrwxrwx 1 root root    9 Aug 13 12:30 usb-Razer_Razer_Blade-if02-mouse -> ../mouse2
lrwxrwxrwx 1 root root   10 Aug 15 09:04 usb-Synaptics_USB-A_to_3.5mm_Jack_Audio_Adapter_0000000000000000-event-if03 -> ../event14
lrwxrwxrwx 1 root root   10 Aug 15 09:04 usb-ZSA_Technology_Labs_Inc_ErgoDox_EZ_Glow-event-if01 -> ../event20
lrwxrwxrwx 1 root root   10 Aug 15 09:04 usb-ZSA_Technology_Labs_Inc_ErgoDox_EZ_Glow-event-kbd -> ../event19
lrwxrwxrwx 1 root root   10 Aug 15 09:04 usb-ZSA_Technology_Labs_Inc_ErgoDox_EZ_Glow-if01-event-kbd -> ../event22

```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

Will attach soon!

### Synapse Screenshots
[image]
[image]
[image]
[image]

qhartman, 2021-08-15:
Hm, I'm not seeing what looks like a usbmon interface in wireshark, and when I just selected "any" and tried filtering for it, I didn't get any traffic. Suggestions?

[image]

Following the guide at https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

lah7, 2021-08-15:
I believe you need to probe the module and then re-open Wireshark:

    sudo modprobe usbmon

z3ntu, 2023-01-16:
A pull request for this device has been opened: #2004

PR #2004 (2023-02-11): Add support for Razer Blade 15 Base model
This keyboard is not addressable, so the features that could be implemented were limited. All the features in this have been tested and are working on my machine.

[image]

Fixes #1581
closed 2023-02-11 by z3ntu