openrazer/openrazer #1583: ERROR: Your 'python3' during Update

hanny00, 2021-08-16:
ERROR: Your 'python3' does not point to /usr/bin/python3 but to /bin/python3, likely a custom environment like anaconda.
Please build this package in a clean chroot (e.g. with https://wiki.archlinux.org/title/DeveloperWiki:Building_in_a_clean_chroot) or **point your PATH variable to prefer /usr/bin/ temporarily.**

Mh, that is too much for someone like me, what does that mean? and how can i set it during my update?

pamac build ?

z3ntu, 2021-08-16:
`export PATH=/usr/bin:$PATH` and then build normally

hanny00, 2021-08-17:
Puh, sorry but where ?

# Maintainer: Luca Weiss <luca (at) z3ntu (dot) xyz>
# Contributor: Gabriele Musco <emaildigabry@gmail.com>

pkgbase=openrazer
pkgname=('python-openrazer' 'openrazer-daemon' 'openrazer-driver-dkms' 'openrazer-meta')
pkgver=3.1.0
pkgrel=1
pkgdesc="An entirely open source driver and user-space daemon that allows you to manage your Razer peripherals on GNU/Linux."
arch=('any')
url="https://github.com/openrazer/openrazer"
license=('GPL2')
makedepends=('python-setuptools')
source=("https://github.com/openrazer/openrazer/releases/download/v$pkgver/openrazer-$pkgver.tar.xz")
sha256sums=('dcec67abba120cca9a5144f96c96df39a81c6debc6ab264881579cbd7d6a9cdb')

prepare() {
  # Do a sanity check in the environment of the builder so the build process doesn't place files into a wrong directory.
  # If you think this is incorrect you can always remove this from the PKGBUILD, but then please don't complain if it doesn't work.
  if [ "$(which python3)" != "/usr/bin/python3" ]; then
    echo "ERROR: Your 'python3' does not point to /usr/bin/python3 but to $(which python3), likely a custom environment like anaconda."
    echo "Please build this package in a clean chroot (e.g. with https://wiki.archlinux.org/title/DeveloperWiki:Building_in_a_clean_chroot) or point your PATH variable to prefer /usr/bin/ temporarily."
    return 1
  fi
}

package_python-openrazer() {
  pkgdesc="Python library for accessing the Razer daemon from Python."
  depends=('openrazer-daemon' 'python-numpy')

  cd "$pkgbase-$pkgver"
  make DESTDIR="$pkgdir" python_library_install
}

package_openrazer-daemon() {
  pkgdesc="Userspace daemon that abstracts access to the kernel driver. Provides a DBus service for applications to use."
  depends=('openrazer-driver-dkms' 'gtk3' 'python-dbus' 'python-gobject' 'python-setproctitle' 'python-daemonize' 'python-notify2' 'python-pyudev' 'xautomation')
  install=openrazer-daemon.install

  cd "$pkgbase-$pkgver"
  make DESTDIR="$pkgdir" daemon_install
}

package_openrazer-driver-dkms() {
  pkgdesc="Kernel driver for Razer devices (DKMS-variant)"
  depends=('dkms' 'udev')
  install=openrazer-driver-dkms.install

  cd "$pkgbase-$pkgver"
  make DESTDIR="$pkgdir" setup_dkms udev_install
}

package_openrazer-meta() {
  pkgdesc="Meta package for installing all required openrazer packages."
  depends=('openrazer-driver-dkms' 'openrazer-daemon' 'python-openrazer')
  optdepends=('polychromatic: frontend'
              'razergenie: qt frontend'
              'razercommander: gtk frontend')
}

z3ntu, 2021-08-17:
In your terminal before calling makepkg (or pamac in your case apparently)

hanny00, 2021-08-17:
I do 
pamac checkupdates -a
pamac update

there are 5 openrazer updates including python-openrazer

I have never used makepkg

lah7, 2021-08-17:
If `pamac` doesn't work for you , try a command line AUR helper like `yay`. Some recent comments in the Telegram chat suggest that `pamac` was causing some trouble. `yay` should be in the Manjaro repo.

    sudo pacman -S yay
    yay -S openrazer-meta python-openrazer openrazer-driver-dkms openrazer-daemon

Likewise, you can build AUR packages with just the `PKGBUILD`. The `-s` flag will install any build dependencies required.

    makepkg -s

hanny00, 2021-08-17:
The Error is the same
yay -S openrazer-meta python-openrazer openrazer-driver-dkms openrazer-daemon
i want to know where in the package build i must type 

export PATH=/usr/bin:$PATH /bin/python3 

I do not understand where i must type the export PATH 

I can not update one package without the other 4

lah7, 2021-08-17:
`PATH` is an environment variable. `export` will change that variable and any command afterwards will use the new value for that Terminal session.

    export PATH=/usr/bin:$PATH

So, it needs to be run before running `yay` or `makepkg`. Basically, this is "correcting" the expected location for Python so the package can be built in the right place, as your system seems to have some kind of other Python installation that is overriding.

hanny00, 2021-08-17:
Ah you  mean 
1. export PATH=/usr/bin:$PATH
2. yay -Syu 

I am sorry, now it works

i am sooo stupid

closed 2021-08-17 by lah7