openrazer/openrazer #1585: Static color not restored after suspend for Cynosa v2

ZwS, 2021-08-17:
### Distribution

Ubuntu 21.04, OpenRazer 3.1.0, Polychromatic 0.7.0

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I've setup Cynosa V2 lighting to static color, but every time I waking my computer lighting become disabled. There no issue If I change lighting mode to Reactive, Breath or Wave. 

Here snippet from daemon logs:
```
2021-08-17 01:05:27 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 381, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 142
2021-08-17 01:05:27 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 381, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 142
2021-08-17 01:05:31 | razer.device0                  | INFO     | Suspending RazerCynosaV2
2021-08-17 01:05:31 | razer.device1                  | INFO     | Suspending RazerBasiliskV2
2021-08-17 19:36:18 | razer.device0                  | INFO     | Resuming RazerCynosaV2
2021-08-17 19:36:18 | razer.device1                  | INFO     | Resuming RazerBasiliskV2
```

lah7, 2021-08-18:
This sounds like similar firmware behaviour to a Chroma HDK I have. If the device is powered on while at zero brightness on a static effect, it's like the effect reset itself to "None". So, even changing the brightness (like the daemon does in this case) remains in darkness until an effect is selected again.

By default, the lighting is set to 0 brightness when the screensavers are active. This can be turned off in `~/.config/openrazer/razer.conf` if desired, or as you're using Polychromatic: Tools → OpenRazer → Configure.

ZwS, 2021-08-19:
> By default, the lighting is set to 0 brightness when the screensavers are active. This can be turned off in ~/.config/openrazer/razer.conf if desired, or as you're using Polychromatic: Tools → OpenRazer → Configure

Thank you, worked for me.

I'm not sure if this issue should be closed since this seems more like a workaround.

th3bucch, 2021-11-04:
Same behavior for my Cynosa Chroma on fresh installed Manjaro, OpenRazer 3.1.0, Polychromatic 0.7.2, manjaro 5.14 kernel: lighting off after waking up from suspend.
Worked fine before on the same hardware but with Ubuntu 20.04, liquorix 5.14 kernel and OpenRazer drivers from stable branch ppa.

bbacskay, 2023-11-03:
Same issue on my system as well.
Device: Razer Cynosa V2
(K)Ubuntu 23.04
Kernel: 6.2.0-36-generic
OpenRazer 3.7.0
Desktop Environment: KDE Plasma 5.27.4

I'm using static red background lights. When only just the screensaver/lockscreen activates it works correctly (turns off the background lights) and unlocking the scrren BEFORE the screen turns off the background lights comes back as it was before. But if the screen was turned off too, the keyboard also disconnects and after reconnecting the background lights remains off.

Here is a verbose daemon log.
At 22:24:40 I locked the screen.
After ~5 minutes at 22:30:59 the screen also turned off.
At 22:31:01 I pressed a button on the keyboard to wake up the system and unlocked the screen.

If @lah7 is right and when the brightness is set to 0 on the static effect the effect resets back to none in the firmware, the during reconnect or resume the static effect shuld be set again.

`
2023-11-02 22:24:40 | razer.screensaver              | DEBUG    | Received screensaver active signal
2023-11-02 22:24:40 | razer.device0                  | INFO     | Suspending RazerCynosaV2
2023-11-02 22:24:40 | razer.device0                  | DEBUG    | DBus call set_brightness
2023-11-02 22:30:59 | razer                          | DEBUG    | Device event [unbind]: /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:07:00.3/usb3/3-6/3-6.4/3-6.4.1/3-6.4.1:1.0/0003:1532:025E.0002
2023-11-02 22:30:59 | razer                          | DEBUG    | Device event [remove]: /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:07:00.3/usb3/3-6/3-6.4/3-6.4.1/3-6.4.1:1.0/0003:1532:025E.0002
2023-11-02 22:30:59 | razer                          | DEBUG    | Device event [unbind]: /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:07:00.3/usb3/3-6/3-6.4/3-6.4.1/3-6.4.1:1.1/0003:1532:025E.0003
2023-11-02 22:30:59 | razer                          | DEBUG    | Device event [remove]: /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:07:00.3/usb3/3-6/3-6.4/3-6.4.1/3-6.4.1:1.1/0003:1532:025E.0003
2023-11-02 22:30:59 | razer                          | DEBUG    | Device event [unbind]: /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:07:00.3/usb3/3-6/3-6.4/3-6.4.1/3-6.4.1:1.2/0003:1532:025E.0004
2023-11-02 22:30:59 | razer                          | DEBUG    | Device event [remove]: /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:07:00.3/usb3/3-6/3-6.4/3-6.4.1/3-6.4.1:1.2/0003:1532:025E.0004
2023-11-02 22:30:59 | razer.device0.keymanager       | DEBUG    | Stopping key manager
2023-11-02 22:30:59 | razer.device0.ripplemanager    | DEBUG    | Closing Ripple Manager
2023-11-02 22:30:59 | razer                          | DEBUG    | Writing persistence config
2023-11-02 22:30:59 | razer                          | WARNING  | Removing 0003:1532:025E.0004
2023-11-02 22:30:59 | razer                          | DEBUG    | Emitted Device Remove Signal
2023-11-02 22:30:59 | razer                          | DEBUG    | Device event [unbind]: /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:07:00.3/usb3/3-6/3-6.4/3-6.4.2/3-6.4.2:1.0/0003:046D:C246.0005
2023-11-02 22:30:59 | razer                          | DEBUG    | Device event [remove]: /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:07:00.3/usb3/3-6/3-6.4/3-6.4.2/3-6.4.2:1.0/0003:046D:C246.0005
2023-11-02 22:30:59 | razer                          | DEBUG    | Device event [unbind]: /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:07:00.3/usb3/3-6/3-6.4/3-6.4.2/3-6.4.2:1.1/0003:046D:C246.0006
2023-11-02 22:30:59 | razer                          | DEBUG    | Device event [remove]: /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:07:00.3/usb3/3-6/3-6.4/3-6.4.2/3-6.4.2:1.1/0003:046D:C246.0006
2023-11-02 22:31:01 | razer                          | DEBUG    | Device event [add]: /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:07:00.3/usb3/3-6/3-6.4/3-6.4.2/3-6.4.2:1.0/0003:046D:C246.000A
2023-11-02 22:31:01 | razer                          | DEBUG    | Device event [add]: /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:07:00.3/usb3/3-6/3-6.4/3-6.4.2/3-6.4.2:1.1/0003:046D:C246.000B
2023-11-02 22:31:01 | razer                          | DEBUG    | Device event [add]: /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:07:00.3/usb3/3-6/3-6.4/3-6.4.1/3-6.4.1:1.0/0003:1532:025E.0007
2023-11-02 22:31:01 | razer                          | DEBUG    | Device event [add]: /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:07:00.3/usb3/3-6/3-6.4/3-6.4.1/3-6.4.1:1.1/0003:1532:025E.0008
2023-11-02 22:31:01 | razer                          | DEBUG    | Device event [add]: /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:07:00.3/usb3/3-6/3-6.4/3-6.4.1/3-6.4.1:1.2/0003:1532:025E.0009
2023-11-02 22:31:01 | razer                          | DEBUG    | Device event [bind]: /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:07:00.3/usb3/3-6/3-6.4/3-6.4.2/3-6.4.2:1.1/0003:046D:C246.000B
2023-11-02 22:31:01 | razer                          | DEBUG    | Device event [bind]: /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:07:00.3/usb3/3-6/3-6.4/3-6.4.2/3-6.4.2:1.0/0003:046D:C246.000A
[... cut ...]
2023-11-02 22:31:05 | razer                          | DEBUG    | State recently changed, writing to disk
2023-11-02 22:31:05 | razer                          | DEBUG    | Writing persistence config
2023-11-02 22:31:11 | razer.screensaver              | DEBUG    | Received screensaver inactive signal
2023-11-02 22:31:11 | razer.device0                  | INFO     | Resuming RazerCynosaV2
2023-11-02 22:31:11 | razer.device0                  | DEBUG    | DBus call set_brightness
2023-11-02 22:31:11 | razer.device0                  | INFO     | Putting device back into driver mode.
`

lah7, 2023-11-03:
In theory, the device being reconnected again should restore the original state. Is persistence enabled in `~/.config/openrazer/razer.conf`?

```ini
[Startup]
restore_persistence = True
```

(There is also `~/.config/openrazer/persistence.conf` worth checking)

bbacskay, 2023-11-03:
> In theory, the device being reconnected again should restore the original state. Is persistence enabled in `~/.config/openrazer/razer.conf`?

And you are right again. I turned on the restore persistence option and now it works. I used the Polychromatic to configure the OpenRazer daemon options and the "Restore last used settings on daemon start" wording for the "restore_persistence" misled me and I tought that the settings are restored only when the daemon starts up.

> (There is also `~/.config/openrazer/persistence.conf` worth checking)
The persistence.conf file is created/updated regardless whether or not the restore persistence is enabled. See the "Writing persistence config" entries in my posted log.

closed 2024-11-14 by github-actions[bot]