openrazer/openrazer #1588: i cant update to 3.1.0

nyc-97, 2021-08-22:
im on kali linux and i cant update to 3.1.0 . it keeps telling me that the newest version 2.9.0 is already installed but i know this isnt the newest version. i have no idea how to install it from this git repo. can some please help im new to this i just want my ornata v2 to have support in polychromatic.

z3ntu, 2021-08-22:
Have you followed the instructions on the website? For Kali Linux will have to use the "Debian Testing" packages

nyc-97, 2021-08-22:
Yeah but if i went through the ppa it would only give me 2.9 if I tried using the binary it would give me error but it’s all good this can be closed . This was just a few of the issues I had trying to get kali running on my razer blade. Ended up reverting to windows and I’m virtualizing kali which is fine for my situation. Thanks anyway.

closed 2021-08-22 by lah7