openrazer/openrazer #1589: Support for Reactive Effect on Razer Mouse Dock (Viper in this case)

ineffyble, 2021-08-22:
Razer Synapse has the option to set the Mouse Dock effect to "Reactive", in which case it lights up every time the mouse is clicked.

I don't know if this is a hardware or software implementation, but it would be great to have it in openrazer.

lah7, 2021-08-23:
Duplicate of #328 

Reactive is a hardware effect, but it needs a software mechanism to trigger the flash. This hasn't been implemented in the daemon.

There is an old gist to run a script in the background - https://github.com/openrazer/openrazer/issues/328#issuecomment-361526340 but it'll need some tweaks: `razer.client` to become `openrazer.client` and `firefly` to `mousemat` (or as appropriate)

lah7, 2021-08-23:
Actually, from looking at the device's methods, Reactive isn't a supported hardware effect for your mouse dock.

https://github.com/openrazer/openrazer/blob/89b2e5808bd20b99ccb871957c4b0ffc51f612a6/daemon/openrazer_daemon/hardware/accessory.py#L57-L70

So, unless the driver/this code is tinkered with and there is a reactive "hardware effect" leftover, it'll need to be performed in software. The script mentioned above wouldn't work.

Parts of that script could make up a new software reactive effect, using the brightness / static methods or matrix protocol, and be contributed as [an example](https://github.com/openrazer/openrazer/tree/master/examples). PRs are welcome to integrate it into the daemon, although apart from Ripple, OpenRazer focuses mostly on enablement of hardware effects.

closed 2022-07-19 by github-actions[bot]