openrazer/openrazer #1599: No device was detected

salvoton, 2021-08-29:
The OpenRazer daemon didn't detect a device that is supported.

uname -a output:
`Linux slvtnnb 5.13.12-1-MANJARO #1 SMP PREEMPT Wed Aug 18 08:06:42 UTC 2021 x86_64 GNU/Linux`

sudo lsusb output:
`Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 1532:0083 Razer USA, Ltd RC30-0315, Gaming Mouse [Basilisk X HyperSpeed]
Bus 001 Device 003: ID 048d:ce00 Integrated Technology Express, Inc. ITE Device(8291)
Bus 001 Device 002: ID 0bda:0129 Realtek Semiconductor Corp. RTS5129 Card Reader Controller
Bus 001 Device 006: ID 8087:0aaa Intel Corp. Bluetooth 9460/9560 Jefferson Peak (JfP)
Bus 001 Device 005: ID 04f2:b65a Chicony Electronics Co., Ltd HD Webcam
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
`

z3ntu, 2021-09-05:
What's the output of the command `sudo dkms install openrazer-driver/3.1.0`  ?

salvoton, 2021-09-05:
> What's the output of the command `sudo dkms install openrazer-driver/3.1.0` ?
`Error! echo
Your kernel headers for kernel 5.13.13-1-MANJARO cannot be found at
/usr/lib/modules/5.13.13-1-MANJARO/build or /usr/lib/modules/5.13.13-1-MANJARO/source.
You can use the --kernelsourcedir option to tell DKMS where it's located.
`

salvoton, 2021-09-05:
Hi, thank you for taking the time for me :) 
Output:
`Error! echo Your kernel headers for kernel 5.13.13-1-MANJARO cannot be found at /usr/lib/modules/5.13.13-1-MANJARO/build or /usr/lib/modules/5.13.13-1-MANJARO/source. You can use the --kernelsourcedir option to tell DKMS where it's located. `

z3ntu, 2021-09-05:
Do you have the package `linux513-headers` installed? If not please do so and try again

salvoton, 2021-09-05:
installing this package fixed the problem . thank you so much :)) have a nice day sir :)

closed 2021-09-05 by salvoton