openrazer/openrazer #1603: Razer Effects question: How to apply custom effects on opening game.

Mayravixx, 2021-09-12:
### Distribution  (With Arch kernel 5.14.2)

Manjaro
Kernel version: 5.14.2-arch1-2

### Support

- [ ] Have you tried rebooting? yes
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page? yes

Just wondering if anyone knows at all how I could apply a specific custom effect to a game I play in Wine, specifically Friday Night Funkin'. I remember with Synapse, it often had custom colors depending on game, for example custom colors for certain keys for Euro Truck Simulator 2, and I was wondering if I could get Polychromatic/OpenRazer to apply these effects automatically when I open the executable I want to set for them.

lah7, 2021-09-12:
For Steam/Proton games, you could run a command before and/or after the game launches. Right click the game, set the launch options. Since you're using Polychromatic, there is a CLI interface to set an effect:

    polychromatic-cli --effect "Name of Effect" ; %command% ; polychromatic-cli -e static -c "#00FF00"

If the game doesn't have launch options (like vanilla Wine), you can trivially create a "wrapper" script and use that as the executable for the game/application:

```bash
#!/bin/bash
polychromatic-cli --effect "Name of Effect"
wine /path/to/game
```

---

Otherwise, I'd say this would be a feature better suited for frontends, scripts or third party projects to implement.  This is something I'd like to do in Polychromatic, but it'll be quite some time before I get to working on that feature. I haven't devised any code on how to do this efficiently neither. It would make a nice [a example script](https://github.com/openrazer/openrazer/tree/master/examples) to demonstrate the use case.

You'll just need a way to _watch_ the process list (`/proc/*`) or _poll_ until the process name has a PID, and then run a command that sets the effect.

This is a pretty crude bash script that would wait for a single application - in this case `nano`:
```bash
#!/bin/bash
while ! pidof nano
do
    sleep 1
done
polychromatic-cli -e "Nano"

# More code is needed to wait until the process exits or start again.
```

🔋 ☠️  "Polling" every second like this probably isn't efficient for laptops with batteries.

Also, for users with other (or no) frontends, a custom effect could be written as a Python script and launched that way. See the [examples](https://github.com/openrazer/openrazer/tree/master/examples).

z3ntu, 2021-10-01:
Closing as the question is now answered.

closed 2021-10-01 by z3ntu