openrazer/openrazer #1606: Razer Kraken Ultimate, static backlight does not work correctly.

VadimVadimVadim1337, 2021-09-16:
Details about the problem
Razer Kraken Ultimate
when i select static effect i cant change color
instead it uses the previous effect color
I will attach a link as it looks
https://1drv.ms/v/s!ArK49L6PGgihgoBRE2qDVjtiGyeLbg

But the static effect can be delivered. if you first press the button "none" and then click on the button
"static" it works correctly.

And "dmesg | grep razeraccessory" outputs nothing (I wrote exactly as it is written).

Operating System / Desktop Environment / Version(s)
OS: Linux mint 20.2 uma
DE: Cinnamon 5.0.5
version Polychromatic v0.7.0
openrazer-meta (3.1.0~ubuntu20.04.1)

Affected Devices
Razer Kraken Ultimate

z3ntu, 2021-10-01:
@tariq86 Can you check with your headset if you have the same behavior?

brett-smith, 2022-02-21:
I have one of these, am seeing exactly the same thing.

VadimVadimVadim1337, 2022-05-28:
@z3ntu [z3ntu](https://github.com/z3ntu), More than a year has passed, do you have any idea why this bug occurs?

z3ntu, 2022-06-18:
No, but you can try hacking around in `driver/razerkraken_driver.c` where you see `case USB_DEVICE_ID_RAZER_KRAKEN_ULTIMATE` and see if you figure out how to fix this.

VadimVadimVadim1337, 2022-08-30:
I can't understand this code. Because I am programming in kotlin. There is a method that, when called, assigns a color to my headphones, I can’t figure out what this method is, where it is at all. 

Just implement a crutch

pseudocode:
def static(self, red: int, green: int, blue: int) -> bool
    if (device.name == "Razer Kraken Ultimate") 
        none()  #i don't know python, i don't know how to call this method correctly but the general idea is clear
    --------
    self._lighting_dbus.setStatic(red, green, blue)

This crutch is better than a static button that doesn't work at all. Static only works after none

brett-smith, 2022-11-09:
I have done some experimenting with `razerkraken_driver.c` and the work-around of turning off all effects before setting a static colour is the only thing that seems to work. In the `razer_attr_write_matrix_effect_static` function, the following is added after the `mutex_lock()` ...

```
    // Work-around for the Kraken Ultimate that doesn't seem to set a static colour until previous color cleared
    if(device->usb_pid == USB_DEVICE_ID_RAZER_KRAKEN_ULTIMATE) {
       effect_byte.bits.on_off_static = 0;
       effect_byte.bits.spectrum_cycling = 0;
       razer_kraken_send_control_msg(device->usb_dev, &effect_report, 0);
    }
```

I tried various other things here too, such as different ordering, but none of that seemed to make any difference. 

There is an almost imperceptible "blink" when swapping from one static colour to another. But as stated above, this is better than the current situation.

One other odd thing. I tried checking with what Synapse was doing using a VM and Wireshark, but the messages being sent  appeared to have no relation to what `razer_attr_write_matrix_effect_static` does. I don't have the specifics to hand, but will try to get more information on that.

closed 2023-08-10 by github-actions[bot]