openrazer/openrazer #1609: Support for [BlackWidow v3 TKL; deathadder v2 mini, razer kraken v3 x]

immellman-test, 2021-09-19:
Hi, the application does not see my devices, although it does display the device IDs. What can you do about it?
Trying to do this on KUBUNTU 21.04.

Perhaps there is a solution, English is not my native language, I could have missed it here.

### Device Info
BlackWidow v3 TKL - RZ03-0349, RZ03-03491800-R3M1
deathadder v2 mini, - RZ01-0334, RZ01-03340100
razer kraken v3 x, - RZ04-03750100-R3M1 (i do not have model name for this headphones)

### Device Information

**lsusb -d 1532: -v**

Bus 005 Device 002: ID 1532:0537 Razer USA, Ltd Razer Kraken V3 X
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0537 
  bcdDevice            0.06
  iManufacturer           1 Razer
  iProduct                2 Razer Kraken V3 X
  iSerial                 3 00000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0123
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              0 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x0072
        bInCollection           2
        baInterfaceNr(0)        1
        baInterfaceNr(1)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID            10
        wTerminalType      0x0201 Microphone
        bAssocTerminal          0
        bNrChannels             2
        wChannelConfig     0x0003
          Left Front (L)
[... 526 lines cut ...]
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1

**ls -lah /dev/input/by-id/**

drwxr-xr-x 2 root root 280 сен 19 21:25 .
drwxr-xr-x 4 root root 760 сен 19 21:25 ..
lrwxrwxrwx 1 root root  10 сен 19 21:25 usb-Razer_Razer_BlackWidow_V3_Tenkeyless-event-if01 -> ../event14
lrwxrwxrwx 1 root root  10 сен 19 21:25 usb-Razer_Razer_BlackWidow_V3_Tenkeyless-event-if03 -> ../event16
lrwxrwxrwx 1 root root  10 сен 19 21:25 usb-Razer_Razer_BlackWidow_V3_Tenkeyless-event-kbd -> ../event10
lrwxrwxrwx 1 root root  10 сен 19 21:25 usb-Razer_Razer_BlackWidow_V3_Tenkeyless-if01-event-kbd -> ../event11
lrwxrwxrwx 1 root root  10 сен 19 21:25 usb-Razer_Razer_BlackWidow_V3_Tenkeyless-if02-event-mouse -> ../event15
lrwxrwxrwx 1 root root   9 сен 19 21:25 usb-Razer_Razer_BlackWidow_V3_Tenkeyless-if02-mouse -> ../mouse1
lrwxrwxrwx 1 root root   9 сен 19 21:25 usb-Razer_Razer_DeathAdder_V2_Mini-event-if01 -> ../event4
lrwxrwxrwx 1 root root   9 сен 19 21:25 usb-Razer_Razer_DeathAdder_V2_Mini-event-mouse -> ../event2
lrwxrwxrwx 1 root root   9 сен 19 21:25 usb-Razer_Razer_DeathAdder_V2_Mini-if01-event-kbd -> ../event3
lrwxrwxrwx 1 root root   9 сен 19 21:25 usb-Razer_Razer_DeathAdder_V2_Mini-if02-event-kbd -> ../event7
lrwxrwxrwx 1 root root   9 сен 19 21:25 usb-Razer_Razer_DeathAdder_V2_Mini-mouse -> ../mouse0
lrwxrwxrwx 1 root root   9 сен 19 21:25 usb-Razer_Razer_Kraken_V3_X_00000000-event-if03 -> ../event9

[image]

lah7, 2021-09-20:
Hi, 2 of your 3 devices are supported. It's likely the installation is not fully complete. **Razer Kraken V3 X** is not supported yet (#1533)

Try clicking the **Troubleshoot** button, as it will tell you what's wrong. It's likely you need to run a command (plugdev) or install the kernel headers package. There's the [troubleshooting guide here](https://github.com/openrazer/openrazer/wiki/Troubleshooting) as well.

📷 💬  Take a screenshot of the **Troubleshoot** window if you're still unsure because of translation.

immellman-test, 2021-09-23:
@lah7, 
Hi, sorry for not answering for so long, worked.

Yes, I tried to see what is behind the "Troubleshoot" button, and there I saw it (attached a screen).
[image]

immellman-test, 2021-09-23:
I updated, after which the devices are not detected, but one error (just with the version) is gone.

But devices (mouse and keyboard) are not detected. Weird

[image]

immellman-test, 2021-09-23:
[image]

immellman-test, 2021-09-23:
However, after another reboot, the problem went away. It's not clear what caused this, to be honest.

closed 2021-09-23 by immellman-test