openrazer/openrazer #1610: Support for Razer Basilisk 3

yochananmarqos, 2021-09-20:
### Device Info

* Name: Razer Basilisk v3
* Product Number: <See under device>
* Model Number: RZ01-04000100-R3U1

### Device Information
```
Bus 001 Device 002: ID 1532:0099 Razer USA, Ltd Razer Basilisk V3
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0099 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Basilisk V3
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x006d
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      79
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        3
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      22
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
```

```
total 0
drwxr-xr-x 2 root root 220 Sep 19 19:53 .
drwxr-xr-x 4 root root 680 Sep 19 19:53 ..
lrwxrwxrwx 1 root root  10 Sep 19 19:53 usb-046d_0825_8F89AF40-event-if00 -> ../event25
lrwxrwxrwx 1 root root  10 Sep 19 19:53 usb-0d8c_USB_Sound_Device-event-if03 -> ../event24
lrwxrwxrwx 1 root root  10 Sep 19 19:53 usb-Corsair_CORSAIR_K55_RGB_PRO_Gaming_Keyboard_F5001BC55F530991AF7A8C0609007011-event-kbd -> ../event22
lrwxrwxrwx 1 root root  10 Sep 19 19:53 usb-Razer_Razer_Basilisk_V3-event-if01 -> ../event20
lrwxrwxrwx 1 root root  10 Sep 19 19:53 usb-Razer_Razer_Basilisk_V3-event-mouse -> ../event18
lrwxrwxrwx 1 root root  10 Sep 19 19:53 usb-Razer_Razer_Basilisk_V3-if01-event-kbd -> ../event19
lrwxrwxrwx 1 root root  10 Sep 19 19:53 usb-Razer_Razer_Basilisk_V3-if02-event-kbd -> ../event21
lrwxrwxrwx 1 root root   9 Sep 19 19:53 usb-Razer_Razer_Basilisk_V3-mouse -> ../mouse2
lrwxrwxrwx 1 root root  10 Sep 19 19:53 usb-Sonix_Technology_Co.__Ltd._Chicony_USB2.0_Camera-event-if00 -> ../event23
```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>

TheNaschkatze, 2021-11-09:
Is this feature comming? D:

laxuscullen, 2021-11-10:
> Is this feature comming? D:

Same question :(

Did you find an alternative? @TheNaschkatze

TheNaschkatze, 2021-11-11:
@laxuscullen I plug in the mouse and the keyboard at the same time, so at least they change color at the same time xD

laxuscullen, 2021-11-11:
@TheNaschkatze That made me laugh lmao.
I would love to customize my mouse but sadly it does not seem like it will be getting supported anytime supported considering the time this issue was open :(

TheNaschkatze, 2021-11-11:
@laxuscullen I mean the issue is open, and not that long ago. The mouse release date is about the same as this issues date. What I'm unsure about, is the meaning of the `need pcap` tag. But I have a good feeling about the support :D

lah7, 2021-11-11:
`need pcap` means it needs packet captures in order to understand what protocol the device speaks. Usually a matter of having a Windows VM handy and [following this guide](https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol).

If you like to get your hands dirty, try modifying the source files yourself and look at [commits implementing other Basilisk devices](https://github.com/openrazer/openrazer/search?q=Basilisk&type=commits) as a reference, such as 7ac21a93c74cc2b6511d73c6f7fa4cf5147bbee5.

laxuscullen, 2021-11-16:
any luck with the pcap? @yochananmarqos

bin4rym4ge, 2021-11-27:
I just got this mouse because my last mouse broke. Why is this so difficult?

laxuscullen, 2021-11-27:
> I just got this mouse because my last mouse broke. Why is this so difficult?

it needs pcap, read lah7's reply. I tried to read through the documentation but welp

TheNaschkatze, 2021-11-30:
> > I just got this mouse because my last mouse broke. Why is this so difficult?
> 
> it needs pcap, read lah7's reply. I tried to read through the documentation but welp

I dont have a wndows VM :/

tildearrow, 2021-11-30:
> > > I just got this mouse because my last mouse broke. Why is this so difficult?
> > 
> > 
> > it needs pcap, read lah7's reply. I tried to read through the documentation but welp
> 
> I dont have a wndows VM :/

It's possible to set up a Windows virtual machine and give it some USB passthrough so that the Razer device in question is passed to the virtual machine so the official Razer drivers interact with it and therefore allow you to look what do the drivers exactly do with the device.
The instructions might be a bit difficult though...

`need pcap` means needs packet capture. A packet capture is a file that Wireshark generates which contains packets, and packets are the things that make these devices do things.

bdach, 2021-12-10:
As a heads up, I'm in the middle of attempting to add this support as I own the device. I believe I've got packet captures of at least the basic parts I need, and I've already got some code in place that looks to be at least part way there. However it still doesn't work because apparently I am seeing `-ETIMEDOUT` errnos from `usb_control_msg` calls and I'm not sure whether this is a device quirk and I need to fiddle with it to increase timeouts somehow, or if I'm just not doing whatever else right.

closed 2022-01-15 by z3ntu

TheNaschkatze, 2022-01-17:
is mergeeeeed <3

laxuscullen, 2022-01-17:
Much love to you @bdach ✨

wheelhub155, 2022-02-03:
are there any updates if there will ever be a release for the Razer Basilsk v3? i just purchased it and non of the released software supports it. let alone i can only get up to version 4.5 to work on my mac due to java script error. please , someone let me know:)

laxuscullen, 2022-02-03:
@wheelhub155 use daily instead of stable branch and v3 is supported. Not sure how the error you are facing will be fixed tho

wheelhub155, 2022-02-03:
> 

Can you please explain what you mean by use daily instead of stable branch? Plz and thx.
Also, I am using macOS. version 10.3.2

I'm using keyboard Razer Black Widow v3 and works with no issues.

lah7, 2022-02-04:
@wheelhub155 OpenRazer doesn't support macOS. It sounds like you're mistaking this project for [1kc/razer-macos](https://github.com/1kc/razer-macos), which studies OpenRazer's driver to add each device separately.

Please see their repository. It looks like they are working on it: https://github.com/1kc/razer-macos/issues/557

JoanaRoqueOS, 2022-02-17:
Hi guys. any changes about the supporting of the basilisk v3? I just bought it yesterday and discover the incompatibility with macOS... I have a macbook pro from 2018 with macOS Monterey version 12.2.1.

I installed the app, but the mouse is not recognized... i already did multiple times the 'refresh device list' and nothing... restarted the mac to see if it makes a difference, and nothing.... But then i discover that this 'issue' was open... Any news about when you will be able to got it to work with the app???
I'm really appreciate your hard work guys!

z3ntu, 2022-02-17:
To reiterate, this project **does not** support macOS, projects that do have their issue tracker somewhere else.