openrazer/openrazer #1613: Viper Ultimate not charging when placed on Charging Dock

d1xt1r, 2021-09-24:
Hi,
I have Razer Viper Ultimate + Charging Dock :

`d1xt1r@pop-os:~$ lsusb | grep 1532
Bus 001 Device 008: ID 1532:007e Razer USA, Ltd RC30-030502 Mouse Dock
Bus 001 Device 007: ID 1532:007b Razer USA, Ltd RC30-0305 Gaming Mouse Dongle [Viper Ultimate (Wireless)]
`

I'm using Pop!_OS 21.04 with the latest available openrazer and latest version of polychromatic. 
Everything is working flawlessly, except mouse battery is not charging while placed on the charging dock. When I plug the cable directly into the mouse it starts charging immediately. The mouse "recognizes" that it is placed on the dock, because the lighting immidiately changes, when the mouse and the dock make contact, but the charning process never starts.

Is there anything I need to do in order to get this working?

Thanks!

spacewonderers, 2021-09-25:
i do not think your problem is involving the software. U should clean your charging dock port and mouse port.

d1xt1r, 2021-09-25:
> i do not think your problem is involving the software. U should clean your charging dock port and mouse port.

Hi, thanks for the answer. In Windows 10 I don't have this issue - it charges and changes the color scheme immediately after you place the mouse on the dock. I did clean the contact points with alcohol but unfortunately it didn't help. I don't think it's a problem with the contact points because when you place the mouse on the dock it immediately changes the color scheme, when you remove it reverts back the color you've chosen as a default.

z3ntu, 2021-10-01:
Not sure but it feels like the lighting effect is controlled by software. Can you confirm (with battery %) that the battery actually doesn't charge?

z3ntu, 2021-10-30:
Then please provide USB packet captures from Windows (https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol) and we can see what Synapse does differently.

Decapitater, 2021-11-01:
Good day gentlemen , 
I would like to present my findings 

Colour of the dock does not change , but the viper shows the battery respective color. 
After placing on dock the polychromatic does not show charging but mouse still charges. 
 & to further clarify i have talked with one of the  maintainer of polychromatic. He told me that the feature is yet to be added.

d1xt1r, 2021-11-01:
> Then please provide USB packet captures from Windows (https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol) and we can see what Synapse does differently.

Hi,

No need, as @Decapitater pointed the issue is not is Openrazer but that Polychromatic doesn't change the battery status when you place the mouse on the dock. I just tried and indeed the battery was few % more when I restarted the Polychromatic app. I'm sorry for the misunderstanding - I had to check that in the first place rather than relying on the battery icon.
Thank you @Decapitater.

Decapitater, 2021-11-01:
My pleasure @d1xt1r

lah7, 2021-11-01:
From what I understand, there is a software effect that changes the colour based on the %, which indeed custom software effects are outside the scope of OpenRazer as they can be achieved by its API.  [A script like the one commented here](https://github.com/polychromatic/polychromatic/issues/369#issuecomment-955575360) could be modified to work for now, and feature request at https://github.com/polychromatic/polychromatic/issues/371, but...

There's still two problems:

- The charging status is incorrect in "docked" mode.
- There is no event emitted when the mouse enters/leaves the dock.

It seems that docked mice continue to use its "wireless" device ID and doesn't switch to the "wired" PID. Packet captures will still be very useful to see what could be missing.

> Polychromatic doesn't change the battery status when you place the mouse on the dock. 

Polychromatic doesn't refresh that information automatically - _yet_ - you'll need to press <kbd>F5</kbd> or <kbd>CTRL+F5</kbd>. It's only as good as what OpenRazer reports:

    python3 -c "import openrazer.client; a = openrazer.client.DeviceManager().devices[0]; print(a.name, a.is_charging, a.battery_level)"

☝🏼 That'll print the device name, if it's "charging" and the battery level. Change `[0]` to `[1]`, etc if you have multiple devices. It sounds like it's always `False` for charging when docked, which will need some work in OpenRazer so it reports the correct status.

Decapitater, 2021-11-01:
the mentioned script is useless as the mouse never shows up in wired mode while docked

Decapitater, 2021-11-01:
however is it usefull to use it for wireless mode but how will it detect the docked state

lah7, 2021-11-01:
Exactly, [please USB capture packets](https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol) if you have access to Windows / Synapse and @z3ntu will be able to analyse these to see if there's something new / missing.

Depending on the outcome, some ideas of improvements: 💡 

- Correctly report `is_charging` in wireless mode → fixes original issue.
- Potentially add `is_docked` to the API, if such condition exists.
- Emit events when `is_charging` changes, and potentially `is_docked` changes.

Decapitater, 2021-11-01:
well i can provide them but they cannot be from the same device i am using as its only linux is other windows device ok ?
(tbh i am more than willing to help soo is there anyway i can) 
providing them in next comment

Decapitater, 2021-11-02:
@lah7 the environment is ready , what which effect ?

lah7, 2021-11-02:
@Decapitater Please capture packets for:

- The mouse when it enters the dock
- The mouse when it leaves the dock
- Maybe try capturing the dock packets when the dock/undock occurs, since we know the hardware changes the lighting on its own, but it's possible it's just a software (custom) effect that overrides it.

Decapitater, 2021-11-03:
[MOUSE(IN&OUT).zip](https://github.com/openrazer/openrazer/files/7467743/MOUSE.IN.OUT.zip)

Decapitater, 2021-11-03:
wow pcapng is not natively allowed @lah7 anything else 
getting this took me a load of time
as the environment was lagging 
waiting for a response

Decapitater, 2021-11-03:
[Dock(in&out).zip](https://github.com/openrazer/openrazer/files/7467793/Dock.in.out.zip)

Decapitater, 2021-11-03:
sry it took some time

Decapitater, 2021-11-04:
@z3ntu something usefull

lah7, 2021-11-07:
I had a quick look a few days ago, but I'm just speculating here: There is an [undocumented](https://github.com/openrazer/openrazer/wiki/Unknown-commands), unknown command `0xbb` which is either `0` or `1`, emitted alongside `GET BATTERY` and `IS CHARGING` packets.

The hardware never changed `is_charging` to `1`, so in terms of OpenRazer and frontends, it's normal that the battery doesn't appear to be charging. I wonder if `0xbb` could be `is_docked` and that's how Synapse knows?

Not sure myself how to hack this into code for further tests, so help appreciated on the next steps!

HanM23, 2021-12-05:
Hi to all,

I would like to know if they are some news about the issue @Decapitater explained ?
I have exactly the same behavior with my Viper Ultimate :

- Dock and mouse are synced, for instance, with spectrum effect
- Then, when i put the mouse on the dock, it charges well, but the Dock keeps spectrum effect and the mouse logo is green (when it's almost fully charged)
- When i remove the mouse from the dock, there are not synced anymore with spectrum effect, i have to do "Apply to all".

I also have to say that when i plug the mouse on the dock, the device "Razer Viper Ultimate (Wired)" does not appear.
Yet, when i plug the mouse with the micro usb cable, it does.

--> I am not able to say if all of this is related to OpenRazer or Polychromatic, i read some issues here and on Polychromatic side, some say it's OpenRazer and some say it's an issue on OpenRazer side.

Please, help me :)
Regards

ps; i do not have Windows, only Linux

Decapitater, 2021-12-05:
@HanM23 you summoned me brother , 
Look at https://github.com/polychromatic/polychromatic/issues/371
All your questions are answered below 

1-
synced then mouse docked then no sync
2-
Does not follow color scheme when docked
It follows the rgb light coloring of default razer
3- Apply to all of required to sync them 
4- wired doesn't appear however you can see it in more option

lah7, 2021-12-05:
From what I gather, it's the technical behaviour of this mouse:

* **The mouse does its own thing when plugged in to a charger.**
  * This causes the "desync" as software does not know the hardware changed effect/colours by itself.
  * Both OpenRazer/Polychromatic rely on hardware effects.
  * Running a custom software effect on a loop might help override it.
* **The mouse does not consider the dock to be "wired"**
  * It still uses wireless mode.
  * The hardware itself says "I'm not charging", even though it is when docked (confirmed via packet capture and @Decapitater earlier)
  * There might be an [undocumented "is docked" command](https://github.com/openrazer/openrazer/issues/1613#issuecomment-962627472).

dogo, 2024-04-16:
Was there any progress? I'm still having the same issue of not charging when it's on the dock, how can I assist?

lah7, 2024-06-15:
If anyone's up for tinkering, I am curious if `0xbb` states whether the mouse is docked [based on my observation](https://github.com/openrazer/openrazer/issues/1613#issuecomment-962627472) when looking at those pcaps. It is a stab in the dark. **Use at own risk.**

I pushed a branch at https://github.com/lah7/openrazer/tree/is_docked_0xbb if anyone would like to try this modified driver/daemon, https://github.com/lah7/openrazer/commit/5437123e12a6e95114f489e7569d340d69583658. If you're on Ubuntu, you can use `./scripts/build_debs.sh` to build the package, install then reboot.

Please let me know the output before and after placing the mouse on your charging dock:

     python3 -c "import openrazer.client; a = openrazer.client.DeviceManager().devices[0]; print(a.name, a.is_docked)"

I don't have the hardware to test - if made an error and the daemon's busted, there should be a new `is_docked` sysfs file under `/sys/bus/hid/drivers/razermouse/` that can be `cat`'d for a `0` or `1`... _or maybe an error!_