openrazer/openrazer #1618: Daemon not detecting user in group

JustKato, 2021-09-26:
### Distribution

Pop!_OS 
With KDE instead of Gnome installed

### Support
- [X] Have you tried rebooting?
- [X] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Running `openrazer-daemon -Fv` will result in the following response:
```
Starting daemon.
2021-09-26 14:49:11 | razer                          | CRITICAL | User is not a member of the plugdev group
2021-09-26 14:49:11 | razer                          | CRITICAL | Please run the command 'sudo gpasswd -a $USER plugdev' and then reboot!
Stopping daemon.
```
Checking my group permissions
```
 кπ ~$ id kato
uid=1000(kato) gid=1000(kato) groups=1000(kato),4(adm),27(sudo),46(plugdev),119(lpadmin),131(docker),133(libvirt)
```

Would love some advice! 
Thank you!

z3ntu, 2021-09-26:
What's the output of the `groups` command?

JustKato, 2021-09-29:
@z3ntu Thanks for the reply.
[image]
It does indeed have plugdev, any idea for me to bypass this?

z3ntu, 2021-10-01:
The daemon has this for checking:

```py
grp.getgrnam('plugdev').gr_gid in os.getgroups()
```

Can you run this and send the output?

```
$ python3 -c 'import grp; import os; print(f"plugdev: {grp.getgrnam(\'plugdev\')}"); print(f"groups: {os.getgroups()}")'
```

closed 2022-06-27 by JustKato