openrazer/openrazer #1620: Support for Razer Huntsman V2

seth-reee, 2021-09-26:
### Device Info

* Name: Razer Huntsman v2
* Product Number: RZ03-03930400-R3U1
* Model Number: RZ03-0393

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```
Bus 001 Device 005: ID 1532:026c Razer USA, Ltd Razer Huntsman V2
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x026c 
  bcdDevice            1.01
  iManufacturer           1 Razer
  iProduct                2 Razer Huntsman V2
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0086
    bNumInterfaces          5
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     177
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0016  1x 22 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      79
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        3
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      22
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        4
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     348
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x85  EP 5 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0002  1x 2 bytes
        bInterval               1

```
Please post the output of
    ls -lah /dev/input/by-id/
```

lrwxrwxrwx 1 root root  10 Sep 26 14:09 usb-Razer_Razer_Huntsman_V2-event-if01 -> ../event12
lrwxrwxrwx 1 root root  10 Sep 26 14:09 usb-Razer_Razer_Huntsman_V2-event-if04 -> ../event14
lrwxrwxrwx 1 root root  10 Sep 26 14:09 usb-Razer_Razer_Huntsman_V2-event-kbd -> ../event10
lrwxrwxrwx 1 root root  10 Sep 26 14:09 usb-Razer_Razer_Huntsman_V2-if01-event-kbd -> ../event11
lrwxrwxrwx 1 root root  10 Sep 26 14:09 usb-Razer_Razer_Huntsman_V2-if02-event-mouse -> ../event13
lrwxrwxrwx 1 root root   9 Sep 26 14:09 usb-Razer_Razer_Huntsman_V2-if02-mouse -> ../mouse2

### Packet Captures

<Please see here as we probably need packet captures>
>

Can do if needed .

### Synapse Screenshots

Can do if needed.

WrathofthePast, 2021-11-01:
> I honestly don't think I have the skills for it, sorry. (Also I lack the time for it, due to my job.)
> 
> But I opened #1650 to maybe get someone with time/skill for jobs like that.

@Trollwut Thank you for trying, but I've had that same discussion before on #1458, so that was an expected outcome. I could theoretically do it, but 1. I use MacOS, not Linux. 2. I have a lot bigger fish to fry right now. Maybe if I'm bored I'll do it just help Linux users out, but that certainly won't be before Christmas.

WrathofthePast, 2021-11-07:
@seth-reee, can you provide the pcap, so support can be added?

WrathofthePast, 2021-11-11:
> @WrathofthePast You can assume and try out, either it will work or it doesn't. But I can't try it out for you without the device.

@z3ntu, the assumption was correct. Thank you. [Source](https://github.com/1kc/razer-macos/issues/482#issuecomment-964496430)

seth-reee, 2021-11-11:
@WrathofthePast 

I'll see what I can do, but I have been absolutely swamped with end of year tickets at work.

AndresHG, 2022-02-07:
Any plans to merge this? I have seen there is currently a PR opened.

Arsimael, 2022-04-01:
Hi. Is there any way I can use this here to build a version of openrazer by myself with this support?

PR #1674 (2022-04-07): Adds Razer Huntsman V2 Support
Adds Razer Huntsman V2 Support

Fixes #1620

Note:  Looks like the Huntsman V2: Analog could also benefit from these sorts of changes/additions

Not Working/Implemented:
- [x] Macros ? (No dedicated M keys... I can either leave the macro stuff in or remove)

Works:
 - [x] Breath Single
 - [x] Breath Dual
 - [x] Breath Random
 - [x] Starlight Single
 - [x] Starlight Dual
 - [x] Starlight Random
 - [x] Ripple
 - [x] Ripple Random
 - [x] Reactive
 - [x] Wave
 - [x] None
 - [x] Spectrum
 - [x] Static
 - [x] Brightness

**EDIT: Moved additional functionality and fixes [... cut]
closed 2022-04-07 by z3ntu