openrazer/openrazer #1627: Upgrade from PPA - DKMS build fails on Ubuntu 20.04

sgofferj, 2021-10-04:
```
DKMS make.log for openrazer-driver-3.1.0 for kernel 5.4.0-88-generic (x86_64)
Mon 04 Oct 2021 08:15:55 PM EEST
-e 
:: Compiling OpenRazer kernel modules
========================================
make -C /lib/modules/5.4.0-88-generic/build M=/var/lib/dkms/openrazer-driver/3.1.0/build/driver modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-88-generic'
  CC [M]  /var/lib/dkms/openrazer-driver/3.1.0/build/driver/razerkbd_driver.o
/var/lib/dkms/openrazer-driver/3.1.0/build/driver/razerkbd_driver.c:153:19: error: ‘KEY_MACRO1’ undeclared here (not in a function); did you mean ‘KEY_MACRO’?
  153 |     { KEY_DELETE, KEY_MACRO1 },
      |                   ^~~~~~~~~~
      |                   KEY_MACRO
/var/lib/dkms/openrazer-driver/3.1.0/build/driver/razerkbd_driver.c:161:19: error: ‘KEY_MACRO2’ undeclared here (not in a function); did you mean ‘KEY_MACRO’?
  161 |     { KEY_PAGEUP, KEY_MACRO2 },
      |                   ^~~~~~~~~~
      |                   KEY_MACRO
/var/lib/dkms/openrazer-driver/3.1.0/build/driver/razerkbd_driver.c:165:21: error: ‘KEY_MACRO3’ undeclared here (not in a function); did you mean ‘KEY_MACRO’?
  165 |     { KEY_PAGEDOWN, KEY_MACRO3 },
      |                     ^~~~~~~~~~
      |                     KEY_MACRO
/var/lib/dkms/openrazer-driver/3.1.0/build/driver/razerkbd_driver.c:173:19: error: ‘KEY_MACRO4’ undeclared here (not in a function); did you mean ‘KEY_MACRO’?
  173 |     { KEY_INSERT, KEY_MACRO4 },
      |                   ^~~~~~~~~~
      |                   KEY_MACRO
make[2]: *** [scripts/Makefile.build:270: /var/lib/dkms/openrazer-driver/3.1.0/build/driver/razerkbd_driver.o] Error 1
make[1]: *** [Makefile:1762: /var/lib/dkms/openrazer-driver/3.1.0/build/driver] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-88-generic'
make: *** [Makefile:37: driver] Error 2
```

w2xim3, 2021-10-04:
I have the same issue.

lah7, 2021-10-04:
Looks like the 5.4.0 LTS kernel originally shipping in Ubuntu 20.04 is no longer compatible with OpenRazer 3.1.0. Fresh installs from 20.04.1 onwards use the [HWE stack](https://ubuntu.com/kernel/lifecycle), which is a newer kernel.

You can use the newer kernel by switching the packages like so:

    sudo apt install --install-recommends linux-generic-hwe-20.04
    sudo apt autoremove linux-generic

If for some reason, DKMS did not install against the newer kernel:

    sudo dkms install openrazer-driver/3.1.0

Then reboot.

I just updated Ubuntu 20.04 and can confirm there's no issues with the HWE kernel 5.11.0-37.

w2xim3, 2021-10-04:
I was stuck doing the procedure, however it worked fine after starting with the following commands.

sudo apt-get remove openrazer-daemon/now openrazer-meta openrazer-driver-dkms/now
sudo dkms remove openrazer-driver/3.1.0 --all

Thank you for the support @lah7

z3ntu, 2021-10-04:
Thanks for reporting! This is fixed with the next daily build.

commit bff485b6 (2021-10-04): razerkbd: fix build with kernels older than 5.5

Add the defines added in newer kernels.

Fixes #1627
closed 2021-10-04 by z3ntu