openrazer/openrazer #1652: Daemon won't start

Dakkaface, 2021-10-31:
### Distribution

Linux Mint 20 Cinnamon

### Support

- [X] Have you tried rebooting?
- [X] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I'm using a Basilisk v2, and the ID is correct, everything is installed, I've got green checks across the board on Polychromatic's troubleshooter, except for one - the daemon will not start, and it's been that way for a few weeks now.

Running `openrazer-daemon -Fv` gets me this output:

```
Traceback (most recent call last):
   File "/usr/lib/python3/dist-packages/daemonize.py", line 245, in start
     self.action(*privileged_action_result)
   File "/usr/bin/openrazer-daemon", line 140, in run_daemon
     daemon = RazerDaemon(verbose=args.verbose,
   File "/usr/lib/python3/dist-packages/openrazer_daemon/daemon.py", line 83, in __init__
     self.read_persistence(persistence_file)
   File "/usr/lib/python3/dist-packages/openrazer_daemon/daemon.py", line 298, in read_persistence
     self._persistence.read(persistence_file)
   File "/usr/lib/python3.8/configparser.py", line 697, in read
     self._read(fp, filename)
   File "/usr/lib/python3.8/configparser.py", line 1017, in _read
     for lineno, line in enumerate(fp, start=1):
   File "/usr/lib/python3.8/codecs.py", line 322, in decode
     (result, consumed) = self._buffer_decode(data, self.errors, final)
 UnicodeDecodeError: 'utf-8' codec can't decode byte 0xf6 in position 5: invalid start byte
```

z3ntu, 2021-10-31:
Please provide the output of `cat /sys/bus/hid/drivers/razermouse/0*/device_serial`

Dakkaface, 2021-10-31:
Output is `322052H13103925`

z3ntu, 2021-10-31:
Oh, I didn't read the stacktrace properly, the file `~/.config/openrazer/persistence.conf` is at fault.. Could you upload that here so I can take a look what went wrong; and then you can delete it and the daemon should work again.

Dakkaface, 2021-10-31:
[persistence.conf.gz](https://github.com/openrazer/openrazer/files/7449639/persistence.conf.gz)

Dakkaface, 2021-10-31:
Deleting persistence.config did the trick, thank you.

closed 2021-10-31 by Dakkaface

z3ntu, 2021-10-31:
I'm going to assume it's a failure with your system so maybe look out for more corrupted files on your system in the future, potentially your disk or ram might be a bit bad. Or just random luck that it broke on yours. If it happens again please tell us and we might make a workaround so the daemon doesn't crash at least.

commit 43423b7b (2022-03-26): daemon: Fix crash when persistence.conf is corrupt

In some circumstances - presumably on system shutdown - the file can get
corrupted. Deleting it and starting from scratch is still better than
crashing.

Fixes #1652
Fixes #1733
Fixes #1777
commit 74b5f642 (2022-03-26): daemon: Fix crash when persistence.conf is corrupt

In some circumstances - presumably on system shutdown - the file can get
corrupted. Deleting it and starting from scratch is still better than
crashing.

Fixes #1652
Fixes #1733
Fixes #1777