openrazer/openrazer #1654: ModuleNotFoundError: No module named 'pyrazer'

roadriverrail, 2021-11-01:
### Distribution

Ubuntu 18.04, system-wide Python switched to 3.8

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

After installing `openrazer-meta` from the PPA per the instructions, all attempts to run razercfg are met with:

```
rhett@downhome:~$ razercfg
Traceback (most recent call last):
  File "/usr/local/bin/razercfg", line 22, in <module>
    from pyrazer import *
ModuleNotFoundError: No module named 'pyrazer'
```

The library is installed:

```
rhett@downhome:~$ find /usr/local/lib -name "pyrazer"
/usr/local/lib/python3.4/dist-packages/pyrazer
```

I've tried things like copying the directory to `...python-3.8/dist-packages` and it didn't seem to help at all.

z3ntu, 2021-11-01:
This project has nothing to do with razercfg outside of trying to solve the same problem. If you want to use razercfg you can completely ignore openrazer, the opposite is also the case.

roadriverrail, 2021-11-01:
My mistake, and my apologies!  I am guessing I have something else installed on my system that's got me mixed up.  I've clearly left some old software laying around on my machine.

closed 2021-11-01 by roadriverrail