openrazer/openrazer #1656: Add Fedora 35 repo

tscherf, 2021-11-03:
Fedora 35 has been GA'ed yesterday. Can you guys please add a new [repo](https://download.opensuse.org/repositories/hardware:/razer/).

andreabellitto, 2021-11-04:
Current version of razergenie and razer-daemon seems to don't be compatible with Fedora 35:

```
Problema: package razergenie-0.9.0-3.1.x86_64 requires razer-daemon, but none of the providers can be installed
  - conflicting requests
  - nothing provides python(abi) = 3.9 needed by openrazer-daemon-3.1.0-1.1.noarch
  - package razergenie-0.9.0-3.1.aarch64 does not have a compatible architecture
  - nothing provides ld-linux-aarch64.so.1()(64bit) needed by razergenie-0.9.0-3.1.aarch64
  - nothing provides ld-linux-aarch64.so.1(GLIBC_2.17)(64bit) needed by razergenie-0.9.0-3.1.aarch64
  - package razergenie-0.9.0-3.1.armv7hl does not have a compatible architecture
  - nothing provides ld-linux-armhf.so.3 needed by razergenie-0.9.0-3.1.armv7hl
  - nothing provides ld-linux-armhf.so.3(GLIBC_2.4) needed by razergenie-0.9.0-3.1.armv7hl
  - nothing provides libgcc_s.so.1(GCC_3.5) needed by razergenie-0.9.0-3.1.armv7hl
  - nothing provides libstdc++.so.6(CXXABI_ARM_1.3.3) needed by razergenie-0.9.0-3.1.armv7hl
  - package razergenie-0.9.0-3.1.ppc64le does not have a compatible architecture
(try to add '--skip-broken' to skip uninstallable packages)
```

tscherf, 2021-11-04:
@andreabellitto  - yes, F35 comes with Python-3.10 while F34 ships Python-3.9.

lah7, 2021-11-04:
Sorry for the delay. We were waiting for [OpenSUSE Build Service](https://build.opensuse.org/project/show/hardware:razer) to show the option to enable the Fedora 35 repositories. It's now available and can confirm it works.

    sudo dnf config-manager --add-repo https://download.opensuse.org/repositories/hardware:razer/Fedora_35/hardware:razer.repo
    sudo dnf install openrazer-meta

Any Python applications sharing that repo (like Polychromatic) may need reinstalling due to the changes in Python version.

closed 2021-11-04 by lah7