openrazer/openrazer #1658: Support for Razer Blade 15 Advanced Late 2019

napsterred, 2021-11-07:
### Device Info

* HELP I am a Linux enthusiast , I relay wanna keep using my blade with ubuntu with the keyboard functions. I wanna get attention to this matter please and thank  you guys, if there is a way I can simply edit device id in my software and make it work please instruct me , if I can provide any other test or information to make it work please instruct me , I am even will with a dev 2-3 bills to get the rgb working for me, please help me out with this I went through all of razer warranty and call center to figure out they couldn't help me. If I can provide any more information to get my device working and added to the supported devices list please instruct me. this was an expensive purchase and I and would love Linux on it, that's for all u guys work and if its impossible please let me know that its IMPOSSIBLE to make it work, until then I keep trying and filing and nothing works. 

* Name: Razer Blade 15 Advanced Late 2019 optical switches
* Product Number: RZ09-03137E02
* Model Number: RZ09-0313

### Device Information
```
Please post the output of
    lsusb -d 1532: -v

Bus 001 Device 005: ID 1532:024b Razer USA, Ltd Gaming Laptop [Blade 15 Advanced (Late 2019)]
Couldn't open device, some information will be missing
Device Descriptor:
bLength 18
bDescriptorType 1
bcdUSB 2.00
bDeviceClass 0
bDeviceSubClass 0
bDeviceProtocol 0
bMaxPacketSize0 64
idVendor 0x1532 Razer USA, Ltd
idProduct 0x024b Gaming Laptop [Blade 15 Advanced (Late 2019)]
bcdDevice 2.00
iManufacturer 1 Razer
iProduct 2 Razer Blade
iSerial 0
bNumConfigurations 1
Configuration Descriptor:
bLength 9
bDescriptorType 2
wTotalLength 0x0054
bNumInterfaces 3
bConfigurationValue 1
iConfiguration 0
bmAttributes 0xa0
(Bus Powered)
Remote Wakeup
MaxPower 500mA
Interface Descriptor:
bLength 9
bDescriptorType 4
bInterfaceNumber 0
bAlternateSetting 0
bNumEndpoints 1
bInterfaceClass 3 Human Interface Device
bInterfaceSubClass 1 Boot Interface Subclass
bInterfaceProtocol 1 Keyboard
iInterface 0
HID Device Descriptor:
bLength 9
bDescriptorType 33
bcdHID 1.11
bCountryCode 0 Not supported
bNumDescriptors 1
bDescriptorType 34 Report
wDescriptorLength 61
Report Descriptors:
** UNAVAILABLE **
Endpoint Descriptor:
bLength 7
bDescriptorType 5
bEndpointAddress 0x81 EP 1 IN
bmAttributes 3
Transfer Type Interrupt
Synch Type None
Usage Type Data
wMaxPacketSize 0x0008 1x 8 bytes
bInterval 1
Interface Descriptor:
bLength 9
bDescriptorType 4
bInterfaceNumber 1
bAlternateSetting 0
bNumEndpoints 1
bInterfaceClass 3 Human Interface Device
bInterfaceSubClass 0
bInterfaceProtocol 1 Keyboard
iInterface 0
HID Device Descriptor:
bLength 9
bDescriptorType 33
bcdHID 1.11
bCountryCode 0 Not supported
bNumDescriptors 1
bDescriptorType 34 Report
wDescriptorLength 159
Report Descriptors:
** UNAVAILABLE **
Endpoint Descriptor:
bLength 7
bDescriptorType 5
bEndpointAddress 0x82 EP 2 IN
bmAttributes 3
Transfer Type Interrupt
Synch Type None
Usage Type Data
wMaxPacketSize 0x0010 1x 16 bytes
bInterval 1
Interface Descriptor:
bLength 9
bDescriptorType 4
bInterfaceNumber 2
bAlternateSetting 0
bNumEndpoints 1
bInterfaceClass 3 Human Interface Device
bInterfaceSubClass 0
bInterfaceProtocol 2 Mouse
iInterface 0
HID Device Descriptor:
bLength 9
bDescriptorType 33
bcdHID 1.11
bCountryCode 0 Not supported
bNumDescriptors 1
bDescriptorType 34 Report
wDescriptorLength 94
Report Descriptors:
** UNAVAILABLE **
Endpoint Descriptor:
bLength 7
bDescriptorType 5
bEndpointAddress 0x83 EP 3 IN
bmAttributes 3
Transfer Type Interrupt
Synch Type None
Usage Type Data
wMaxPacketSize 0x0008 1x 8 bytes
bInterval

```

```
Please post the output of
    ls -lah /dev/input/by-id/
Please post the output of
ls -lah /dev/input/by-id/
napster@RB15:~$ ls -lah /dev/input/by-id/
total 0
drwxr-xr-x 2 root root 180 Oct 14 08:48 .
drwxr-xr-x 4 root root 740 Oct 14 08:48 ..
lrwxrwxrwx 1 root root 9 Oct 14 08:48 usb-Azurewave_Integrated_Camera_200901010001-event-if00 -> ../event5
lrwxrwxrwx 1 root root 9 Oct 14 08:48 usb-Azurewave_Integrated_Camera_200901010001-event-if02 -> ../event6
lrwxrwxrwx 1 root root 10 Oct 14 08:48 usb-Razer_Razer_Blade-event-if01 -> ../event10
lrwxrwxrwx 1 root root 9 Oct 14 08:48 usb-Razer_Razer_Blade-event-kbd -> ../event7
lrwxrwxrwx 1 root root 9 Oct 14 08:48 usb-Razer_Razer_Blade-if01-event-kbd -> ../event8
lrwxrwxrwx 1 root root 10 Oct 14 08:48 usb-Razer_Razer_Blade-if02-event-mouse -> ../event12
lrwxrwxrwx 1 root root 9 Oct 14 08:48 usb-Razer_Razer_Blade-if02-mouse -> ../mouse2

```

### Packet Captures

[staticmode-ff1234-4411ff-bladeadv-1532-024b (1).txt](https://github.com/openrazer/openrazer/files/7491586/staticmode-ff1234-4411ff-bladeadv-1532-024b.1.txt)

static mode
please rename file to .pcapng

### Synapse Screenshots

[image]

nightsky30, 2021-11-13:
Another duplicate of #1041

napsterred, 2021-11-19:
> Another duplicate of #1041

Hey could ya please instruct me on how to get it to work even if i have to do it manually just for my computer

z3ntu, 2021-12-03:
Duplicate of #1041

closed 2021-12-03 by z3ntu