openrazer/openrazer #1661: Support for Razer BlackWidow V3 Mini HyperSpeed

djve60, 2021-11-09:
### Device Info

* Name: Razer BlackWidow V3 Mini HyperSpeed
* Product Number: RZ03-03891900
* Model Number: RZ03-0389
### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```
% sudo lsusb -d 1532: -v

Bus 001 Device 003: ID 1532:0271 Razer USA, Ltd Razer BlackWidow V3 Mini
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0271 
  bcdDevice            1.00
  iManufacturer           1 Razer
  iProduct                2 Razer BlackWidow V3 Mini
  iSerial                 3 000000000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x009f
    bNumInterfaces          6
    bConfigurationValue     1
    iConfiguration          4 Default configuration
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
[... 347 lines cut ...]
lrwxrwxrwx 1 root root  10 Nov  8 16:45 usb-Chicony_Electronics_Co._Ltd._Integrated_Camera_6726-event-if02 -> ../event22
lrwxrwxrwx 1 root root   9 Nov  8 16:45 usb-Razer_ProClickM_000000000000-event-if01 -> ../event8
lrwxrwxrwx 1 root root  10 Nov  8 16:45 usb-Razer_ProClickM_000000000000-event-if03 -> ../event12
lrwxrwxrwx 1 root root   9 Nov  8 16:45 usb-Razer_ProClickM_000000000000-event-mouse -> ../event5
lrwxrwxrwx 1 root root   9 Nov  8 16:45 usb-Razer_ProClickM_000000000000-if01-event-kbd -> ../event6
lrwxrwxrwx 1 root root   9 Nov  8 16:45 usb-Razer_ProClickM_000000000000-if01-event-mouse -> ../event7
lrwxrwxrwx 1 root root   9 Nov  8 16:45 usb-Razer_ProClickM_000000000000-if01-mouse -> ../mouse1
lrwxrwxrwx 1 root root  10 Nov  8 16:45 usb-Razer_ProClickM_000000000000-if02-event-kbd -> ../event11
lrwxrwxrwx 1 root root   9 Nov  8 16:45 usb-Razer_ProClickM_000000000000-mouse -> ../mouse0
lrwxrwxrwx 1 root root  10 Nov  8 16:45 usb-Razer_Razer_BlackWidow_V3_Mini_000000000000-event-if01 -> ../event16
lrwxrwxrwx 1 root root  10 Nov  8 16:45 usb-Razer_Razer_BlackWidow_V3_Mini_000000000000-event-if04 -> ../event20
lrwxrwxrwx 1 root root  10 Nov  8 16:45 usb-Razer_Razer_BlackWidow_V3_Mini_000000000000-event-kbd -> ../event13
lrwxrwxrwx 1 root root  10 Nov  8 16:45 usb-Razer_Razer_BlackWidow_V3_Mini_000000000000-if01-event-kbd -> ../event14
lrwxrwxrwx 1 root root  10 Nov  8 16:45 usb-Razer_Razer_BlackWidow_V3_Mini_000000000000-if01-event-mouse -> ../event15
lrwxrwxrwx 1 root root   9 Nov  8 16:45 usb-Razer_Razer_BlackWidow_V3_Mini_000000000000-if01-mouse -> ../mouse2
lrwxrwxrwx 1 root root  10 Nov  8 16:45 usb-Razer_Razer_BlackWidow_V3_Mini_000000000000-if02-event-mouse -> ../event19
lrwxrwxrwx 1 root root   9 Nov  8 16:45 usb-Razer_Razer_BlackWidow_V3_Mini_000000000000-if02-mouse -> ../mouse3

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>

djve60, 2021-11-09:
Can not supply packets at this time.

The output shows two new devices, a keyboard and mouse. I'm going to hope that it's just a matter of adding the device ID for the keyboard and mouse as they work perfectly well directly out of the box, under Ubuntu 20.04.

lah7, 2021-11-09:
Your device (`1532:0271`) is already supported, but it's not in the stable release yet. (#1617 - 7c353f9c8f1439bce5396218331c412eae0c2563)

Since you're on Ubuntu, update to the "daily" package to try it out:
```
sudo add-apt-repository ppa:openrazer/daily
sudo apt update
sudo apt dist-upgrade
```

closed 2021-11-09 by lah7