openrazer/openrazer #1662: System notification on low and critical battery level

Trollwut, 2021-11-09:
Hey mates!

OpenRazer has the support of periodically notifying the user about the battery status. This happens with no regards on how much the battery is charged.

Is it possible to be only notified when the battery level drops below a certein percentage?

For example show one notification, if the battery percentage falls below 10% (low). And another one, if it falls below 5% (critical).

lah7, 2021-11-09:
There isn't a user-configurable option for this, but you can tweak the daemon's code to accomplish this.

At the moment, the battery notifier appears every X seconds as set in `~/.config/openrazer/razer.conf`, and the messaging changes when it's under 10%.

https://github.com/openrazer/openrazer/blob/999a0d77d6cf8898166d0bb6ea8c5526d6738842/daemon/openrazer_daemon/misc/battery_notifier.py#L91-L93

To patch your local install, open the file (with sudo/root) - which depending on your distro could be either:

> /usr/lib/python3/dist-packages/openrazer_daemon/misc/battery_notifier.py
> /usr/lib/python3.9/site-packages/openrazer_daemon/misc/battery_notifier.py

And add a condition to the end to only show if the level is, say, 25%:

```diff
-                if self._notify2:
+                if self._notify2 and battery_level < 25.0:
                    self._notification.update(summary="{0} Battery at {1:.1f}%".format(self._device_name, battery_level))
                    self._notification.show()
```

The daemon will need to be restarted for it to take effect.

    openrazer-daemon -r

Trollwut, 2021-11-09:
Thanks for your solution! I did this and now I wait for my mouse level to drop.

Maybe this would be a feature request to have it user-configurable?

lah7, 2021-11-09:
It seemed a pretty simple thing to add, so I did and propose changes in #1666. While I was in there, I cleaned up the presentation of the notification too. 🔔

Trollwut, 2021-11-10:
It still sends continuous notifications, like every second. Yes, the user just needs to set it to 5 minutes and is done. It would be even better for the user if one notification is being send upon reaching the "low" threshold, and another one one the "critical".

But hey, your easy implementation is exactly why I for myself am looking for, so thank you for your work!! :) 

Also good that you catched that thing with the icons, appreciate that :D

lah7, 2021-11-10:
No problem! I guess this battery notifier [hasn't changed](https://github.com/openrazer/openrazer/blame/master/daemon/openrazer_daemon/misc/battery_notifier.py) that much since it's original "proof of concept" debut years ago, but it would make sense to me to only have events when it's low or critical, rather than a continuous reminder. That could be some further improvement for another day (unless users with 🔋  prefer like reminders of current percentage? 🤷🏼)

Regarding the _every second_, have you set `mouse_battery_notifier_freq` in `~/.config/openrazer/razer.conf` to the right number? It's measured in seconds, so 5 minutes = `300`.

Trollwut, 2021-11-10:
> No problem! I guess this battery notifier [hasn't changed](https://github.com/openrazer/openrazer/blame/master/daemon/openrazer_daemon/misc/battery_notifier.py) that much since it's original "proof of concept" debut years ago, but it would make sense to me to only have events when it's low or critical, rather than a continuous reminder. That could be some further improvement for another day (unless users with battery prefer like reminders of current percentage? 🤷🏼)

For a constant reminder, an implementation into a DE might be a better idea IMHO. (https://github.com/openrazer/openrazer/issues/1663)

> Regarding the _every second_, have you set `mouse_battery_notifier_freq` in `~/.config/openrazer/razer.conf` to the right number? It's measured in seconds, so 5 minutes = `300`.

Of course I'll do that, if your changes are in the software :)

Just wanted to point out that a normal user might not have the insight and is wondering why he gets a constant notification instead of a threshold warning.

lah7, 2021-11-10:
I agree. Maybe it's worth disabling the battery notification by default? OpenRazer by itself is _supposed_ to just be the "backend" piece of a larger puzzle with no user facing parts, except the config file, and this notification.

That's where the [variety of frontends/scripts](https://github.com/openrazer/openrazer#applications) come in. My app, [Polychromatic](https://github.com/polychromatic/polychromatic), does have a settings window to configure OpenRazer's configuration file, and some work in progress will eventually [make it possible to show the battery %](https://github.com/polychromatic/polychromatic/issues/106#issuecomment-874957189) to appear in its tray applet too (currently, it can only be seen in the main GUI)

HanM23, 2021-12-16:
> Hey mates!
>
> OpenRazer has the support of periodically notifying the user about the battery status. This happens with no regards on how much the battery is charged.
>
> Is it possible to be only [...]

Hi, 

Could you please tell which is your distro ? Because i tried manjaro (xfce and kde) and xubuntu and i have no notification at all.
I have a Viper Ultimate, and it's working properly, but i never get any notification, i really would like to know why.
Maybe if i use the same distro as you, i could see a difference.

[1698](https://github.com/openrazer/openrazer/issues/1698)

Thanks,
Regards

Trollwut, 2021-12-17:
Sure mate!

Manjaro, up-to-date, currently with KDE

lah7, 2021-12-17:
In order for #1666 to progress, [we're wondering](https://github.com/openrazer/openrazer/pull/1666#pullrequestreview-823265177) what is the highest % reading for your mice when it's fully charged?

Does it read 100%, or does it hover around 90%-99%?

HanM23, 2021-12-17:
Hi,
I charged my mouse until 100% today.

[image]

When 100% is reached, and my mouse is on the dock, the green logo becomes static, before (even at 99%) it still breathes green.

closed 2022-01-15 by z3ntu

LazyOtto, 2022-08-09:
This is merely a more extensive mod along the same lines mentioned above.
Effects: Sends messages very slowly/infrequently when battery charge is high. Ramps up frequency of messaging as the charge drops into low/critical ranges.  Also sends messages as the device charges back up.

```
           if self._notify2:
                if self.percent == 0:
                    self.percent = 100

                # 3.0 as lower bound filters out 'noise' caused when changing settings from RazerGenie
                if 3.0 < battery_percent <= self.percent:
                    self._notification.update(summary=title, message=message, icon=icon)
                    self._notification.show()

                    # graduates frequency of messages/notifications bothering the user
                    if battery_percent >= 80:
                        self.percent = 70
                    elif battery_percent >= 70:
                        self.percent = ((battery_percent - 9) // 10) * 10
                    elif battery_percent >= 60:
                        self.percent = ((battery_percent - 9) // 10) * 10
                    elif battery_percent >= 40:
                        self.percent = battery_percent - 5
                    elif battery_percent >= 30:
                        self.percent = battery_percent - 3
                    elif battery_percent >= 21:
                        self.percent = battery_percent - 2
                    elif battery_percent >= 15:
                        self.percent = battery_percent - 1

            if 3.0 < battery_level < self.lastLevel:
                # logging for discharging
                self.lastLevel = battery_level
                self._logger.info("{0} battery and next message levels are | {1:.2f} | {2}".format(self._device_name, battery_level, self.percent))

            elif 3.0 < battery_level > self.lastLevel:
                # ratchet the trigger level back up if charging
                if battery_percent >= 80:
                    self.percent = 70
                elif battery_percent >= 70:
                    self.percent = ((battery_percent - 9) // 10) * 10
                elif battery_percent >= 60:
                    self.percent = ((battery_percent - 9) // 10) * 10
                elif battery_percent >= 40:
                    self.percent = battery_percent - 5
                elif battery_percent >= 30:
                    self.percent = battery_percent - 3
                elif battery_percent >= 21:
                    self.percent = battery_percent - 2
                elif battery_percent >= 15:
                    self.percent = battery_percent - 1

                # logging for charging
                self.lastLevel = battery_level
                self._logger.info("{0} battery and next message levels are | {1:.2f} | {2}".format(self._device_name, battery_level, self.percent))
```

PR #1666 (2022-01-15): Improve battery notification presentation & add option to configure % threshold
* Adds `battery_notifier_percent` to set when notifications should start appearing. New default: 33%.
* Use standard `battery-*` icons and adapt based on percentage. `notification-battery-low` seemed non-standard and couldn't be found in Adwaida, GNOME or Breeze.
* Clean up code for easier reading.
* Change application title to "OpenRazer", for notification systems like Breeze which support showing the application name.

Closes #1662.

### Before
[image]

### After
[image]
commit 0b9182f4 (2022-01-15): daemon: Add config option to set battery % condition

Closes #1662.

This sets a new default of 33% before notifications start appearing,
which would previously have been 100%.