openrazer/openrazer #1663: Show battery status besides other battery devices in DEs

Trollwut, 2021-11-09:
Hey mates!

For example, KDE has a menu, where all devices with batteries are listed: https://i.imgur.com/ZTNbV2W.png

Is it possible to also show the battery status of OpenRazer devices?

lah7, 2021-11-09:
It's not possible at the moment. I know that many environments have a battery indicator, usually for a laptop battery, but I have seen Bluetooth peripheral batteries appear in the same menu too, so it's likely possible. Some work will be needed in the daemon to support this desktop integration, preferably agnostic to any desktop environment too. 

From [a quick look at the KDE's code](https://github.com/KDE/plasma-workspace/blob/c97dddf20df5702eb429b37a8c10b2c2d8199d4e/dataengines/powermanagement/powermanagementengine.cpp#L281), it suggests that it might be D-Bus session bus `org.freedesktop.PowerManagement` (and `org.kde.Solid.PowerManagement` is KDE's specifically)

Contributions from any one willing to take a stab at this is much appreciated.

z3ntu, 2022-01-15:
Similar to / duplicate of #87

closed 2022-07-19 by github-actions[bot]