openrazer/openrazer #1664: Wheel tilt edges aren't being detected on the Mamba Elite

rafifos, 2021-11-09:
### Distribution

Pop!_OS 21.10

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Hi, I was trying to bind these scroll wheel tilt events on my Mamba Elite to a specific action, but the supposed `5` and `6` aren't being detected by `xev` (and any other tool that i've used). I´d take a look at the code and saw that both OpenRazer supports it and the Mamba Elite is supposed to have it working.

z3ntu, 2021-12-03:
Does wheel tilt currently scroll up and down? If so, it's likely that we keep the device in "device mode" where some functionality is handled by the device itself instead of having to reimplement functionality. Downside is we cannot get these events in the OS.

rafifos, 2021-12-22:
Sorry for the delay, @z3ntu.
No, it doesn't scroll up/down (actually, nothing happens) when tilting the scroll wheel.

z3ntu, 2022-01-15:
Does the same happen without openrazer installed?

rafifos, 2022-01-19:
@z3ntu Yes, there's not output from `xev` when openrazer isn´t installed.

closed 2022-07-19 by github-actions[bot]