openrazer/openrazer #1665: Support for Kyro Pro

fabiocarneiro, 2021-11-09:
### Device Info

* Name: Razer Kyro Pro
* Product Number: RZ19-03640100
* Model Number: RZ19-0364

### Device Information
```
Bus 004 Device 006: ID 1532:0e05 Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.20
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0         9
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0e05 
  bcdDevice            8.21
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x05b4
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              896mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass         14 Video
      bFunctionSubClass       3 Video Interface Collection
      bFunctionProtocol       0 
      iFunction               0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass        14 Video
      bInterfaceSubClass      1 Video Control
      bInterfaceProtocol      0 
      iInterface              0 
      VideoControl Interface Descriptor:
        bLength                13
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdUVC               1.00
        wTotalLength       0x0069
        dwClockFrequency       30.000000MHz
        bInCollection           1
        baInterfaceNr( 0)       1
      VideoControl Interface Descriptor:
        bLength                 9
        bDescriptorType        36
        bDescriptorSubtype      3 (OUTPUT_TERMINAL)
        bTerminalID             4
        wTerminalType      0x0101 USB Streaming
        bAssocTerminal          0
[... 838 lines cut ...]
        wMaxPacketSize     0x0084  1x 132 bytes
        bInterval               4
        bRefresh                0
        bSynchAddress           0
        bMaxBurst               0
        AudioStreaming Endpoint Descriptor:
          bLength                 7
          bDescriptorType        37
          bDescriptorSubtype      1 (EP_GENERAL)
          bmAttributes         0x01
            Sampling Frequency
          bLockDelayUnits         0 Undefined
          wLockDelay         0x0000

```

```
lrwxrwxrwx 1 root root  10 Nov  9 22:29 usb-Razer_Inc_Razer_Kiyo_Pro-event-if00 -> ../event23
```

### Packet Captures
No VirtualBox :(

### Synapse Screenshots
I don't have windows :(

j-maynard, 2021-11-12:
Is there anything really to capture.  Ultimately its a USB Web Cam, theres no lighting to control or anything else really to control in software which isn't already provided by the v4l driver.  Though it raises a point about should there be stubs in the driver for devices such as thing and other razer devices which don't have lighting or driver control?  I notice that polychomatic shows all razer devices even ones which aren't supported by OpenRazer.  Unsupported devices show up as just their USB identifier 1532:<device_id>.

lah7, 2021-11-13:
If there's no lighting or special "proprietary" software functionality, there's probably nothing for OpenRazer to support. The project is mainly focused on lighting and some device-specific functionality (like keyboard M1-M5 macro keys)

Polychromatic does list "unrecognised" devices to give users a helping hand, but that app could show webcam options (links to programs) or just omit it altogether.

fabiocarneiro, 2021-11-13:
What motivated me to request support was the lack of control of Autofocus. I'm sure synapse is able to modify this behavior but i have never seen how it works.

Additionally the camera provides 3 different fields of view (narrow/medium/wide) and HDR activation/deactivation, FPS control.

lah7, 2021-11-13:
Sounds like there could be hardware toggles, but equally the settings could just be specific for the Windows driver. For support to be possible, we will need packet captures (pcaps) to see what Synapse is sending to the device. Then it could be possible to expose such options to APIs for frontends and scripts to support.

It'll require a Windows VM, Razer Synapse (no account required), Wireshark and a bit of time: https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

davefedele, 2021-11-13:
> 
> 
> What motivated me to request support was the lack of control of Autofocus. I'm sure synapse is able to modify this behavior but i have never seen how it works.
> 
> Additionally the camera provides 3 different fields of view (narrow/medium/wide) and HDR activation/deactivation, FPS control.
> 
> @lah7 Yes maybe this is not a case for Polychromatic, since there are no lighting features. It is however a case for OpenRazer which can be controlled by RazerGenie

I'm also hoping to access the hdr and field of view toggles. Here are the images from synapse. 

[image]

[image]

[image]

ObiWahn, 2021-11-20:
I would love to have the option to change the brightness and VOF!

davefedele, 2021-11-20:
> I would love to have the option to change the brightness and VOF!
> 
> Edit: `v4l-utils` and `v4l2-ctl -d /dev/video0 --list-ctrls` cover probably most or my needs.

Thanks for this suggestion @ObiWahn . Poked around a little and found this =>[ https://www.jcheng.org/post/webcam/]( https://www.jcheng.org/post/webcam/)
It's not showing the pro, but provides some example usage and images to go along.

cdgriffith, 2021-11-22:
I was working on pcap for the other ticket https://github.com/openrazer/openrazer/issues/1545 so figured I'd post here as well:
[kiyo_pro_pcap.zip](https://github.com/openrazer/openrazer/files/7582112/kiyo_pro_pcap.zip)
 
 HDR was in on state then turned off and back on for first one. 
 FOV was in narrow to start, switcted to medium, then wide, then back to narrow for the second.

cdgriffith, 2021-11-22:
Another issue happening with the Kiyo Pro is that it will randomly crash Ubuntu 20.04 all together. All USB devices will stop working and will have to reboot to get anything back. All I see in the `journalctl` logs are: 

```
Nov 22 09:08:05.504869  kernel: usb 2-2.4: new SuperSpeed Gen 1 USB device number 3 using xhci_hcd
Nov 22 09:08:05.504964  kernel: usb 2-2.4: New USB device found, idVendor=1532, idProduct=0e05, bcdDevice= 8.21
Nov 22 09:08:05.505058  kernel: usb 2-2.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Nov 22 09:08:05.505152  kernel: usb 2-2.4: Product: Razer Kiyo Pro
Nov 22 09:08:05.505239  kernel: usb 2-2.4: Manufacturer: Razer Inc
...
Nov 22 09:08:06.168216  kernel: usb 2-2.4: current rate 16000 is different from the runtime rate 48000
...
Nov 22 09:08:21.426498  kernel: usb 2-2.4: 3:2: cannot set freq 16000 to ep 0x82
Nov 22 09:08:26.558497  kernel: usb 2-2.4: 3:3: cannot set freq 24000 to ep 0x82
Nov 22 09:08:26.582485  kernel: usb 2-2.4: 3:4: cannot set freq 32000 to ep 0x82
Nov 22 09:08:26.666454  kernel: usb 2-2.4: 5:2: cannot get min/max values for control 2 (id 5)
Nov 22 09:08:26.750425  kernel: usbcore: registered new interface driver snd-usb-audio
Nov 22 09:08:26.830460  kernel: usb 2-2.4: 5:2: cannot get min/max values for control 2 (id 5)
Nov 22 09:08:26.898414  kernel: usb 2-2.4: 5:2: cannot get min/max values for control 2 (id 5)
Nov 22 09:08:27.038459  kernel: usb 2-2.4: 5:2: cannot get min/max values for control 2 (id 5)
Nov 22 09:08:27.110477  kernel: usb 2-2.4: 5:2: cannot get min/max values for control 2 (id 5)
Nov 22 09:08:27.194476  kernel: usb 2-2.4: 3:0: usb_set_interface failed (-71)
Nov 22 09:08:27.198460  kernel: usb 2-2.4: 3:1: usb_set_interface failed (-71)
Nov 22 09:08:27.206473  kernel: usb 2-2.4: 3:0: usb_set_interface failed (-71)
Nov 22 09:08:27.214477  kernel: usb 2-2.4: 3:0: usb_set_interface failed (-71)
Nov 22 09:08:27.222471  kernel: usb 2-2.4: 3:1: usb_set_interface failed (-71)
Nov 22 09:08:27.226464  kernel: usb 2-2.4: 3:0: usb_set_interface failed (-71)
Nov 22 09:08:27.234463  kernel: usb 2-2.4: 3:0: usb_set_interface failed (-71)
Nov 22 09:08:27.242469  kernel: usb 2-2.4: 3:1: usb_set_interface failed (-71)
Nov 22 09:08:27.250451  kernel: usb 2-2.4: 3:0: usb_set_interface failed (-71)
(repeat last line 59,165 times)
```

fabiocarneiro, 2021-11-22:
@cdgriffith Wow, that is intense.

I never had that of all USB devices stopping but I had all my sound stuff stopping to work (empty list on the settings). In my case, this is fixed by removing the camera and plugging it back on. I suspect it is mostly related to the microphone.

j-maynard, 2021-11-23:
I can confirm the above bug... I am not entirely sure whats going on but it is something specific to the Kiyo pro.

j-maynard, 2021-11-23:
@cdgriffith is there a pulse audio error before this occurs by any chance?  A quick search of the internet points to possible issues with PulseAudio causing problems.  This maybe related?

cdgriffith, 2021-11-23:
@j-maynard The only line with anything about audio is the one in that section actually: 

```
Nov 22 09:08:26.750425 kernel: usbcore: registered new interface driver snd-usb-audio
```

Everything else is in it's own time block and doesn't seem related. Is there a way to disable that driver being loaded for that device? Just disable it's audio from starting all-together?

cdgriffith, 2021-12-03:
Anyone good at reading usb pcap posted above? 

I can try writing up a simple script to test it out if I know what needs to be set. I tried with what it looked like to me without any success https://github.com/openrazer/openrazer/issues/1545#issuecomment-955598906

z3ntu, 2021-12-03:
@cdgriffith Assuming the packet capture was done correctly, I don't see any normal Razer packets in there, but given there are v4l2 controls it's likely Razer implemented it via USB UVC or whatever this is using internally. Nothing to support from OpenRazer side really. I'll close the issue with that reasoning but feel free to continue chatting about this device.

closed 2021-12-03 by z3ntu

fabiocarneiro, 2022-01-10:
@z3ntu what would be the recommended approach given it would still be nice to be able to control the device on razegenie, polycromatic, etc...? Should the support be added on that side via this methods you mentioned?

z3ntu, 2022-01-15:
I don't really see a point in adding support on the openrazer frontends (polychromatic/RazerGenie) as it's not Razer-specific functionality. E.g. if you put the device into OBS then you can adjust the settings there.

cdgriffith, 2022-01-15:
> I don't really see a point in adding support on the openrazer frontends (polychromatic/RazerGenie) as it's not Razer-specific functionality. E.g. if you put the device into OBS then you can adjust the settings there.

You can't turn on / off HDR or change the filed of view through anything other than the Razor Synapse app.

madhavajay, 2022-02-20:
Theres a paid app called Webcam Settings for MacOS which actually works really great to toggle things like autofocus, zoom, tilt and all the color settings you can want plus the ability to save the settings and force apply them.
[image]

But it does seem like MacOS itself sometimes can't start the camera and requires a little bit of unplugging and plugging in so maybe the default driver in MacOS Monterey has some issues?

soyersoyer, 2022-03-15:
I created a script to setup the kiyo pro custom controls in linux:
[kiyoproctrls](https://github.com/soyersoyer/kiyoproctrls)

prx0, 2022-05-17:
@soyersoyer Hello, have you succeed to get a stream record from a kiyo pro using v4l2 on Linux ? I'm unable to get an output from VLC or OBS.

soyersoyer, 2022-05-17:
Yes, but it works only with YUYV format and USB3. The other formats are buggy. They don't care about UVC.