openrazer/openrazer #1669: Support for Seiren Emote

Jpe230, 2021-11-13:
### Device Info

* Name: Razer Seiren Emote
* Product Number: RZ19-03060100-R3M1
* Model Number: RZ19-0306

### Device Information
## lsusb
```
Bus 001 Device 004: ID 1532:0524 Razer USA, Ltd Razer Seiren Emote
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        16
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0524 
  bcdDevice            1.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0121
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              0 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x006a
        bInCollection           2
        baInterfaceNr(0)        1
        baInterfaceNr(1)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0101 USB Streaming
        bAssocTerminal          0
        bNrChannels             2
        wChannelConfig     0x0003
          Left Front (L)
          Right Front (R)
        iChannelNames           0 
        iTerminal               0 
      AudioControl Interface Descriptor:
        bLength                12
[... 296 lines cut ...]
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      22
         Report Descriptors: 
           ** UNAVAILABLE **
```

## ls -lah /dev/input/by-id/
```
total 0
drwxr-xr-x 2 root root 100 nov 12 20:58 .
drwxr-xr-x 4 root root 340 nov 12 20:58 ..
lrwxrwxrwx 1 root root   9 nov 12 20:58 usb-Razer_Inc_Razer_Seiren_Emote_242003L02800055-event-if03 -> ../event7
```

### Packet Captures

[capture.zip](https://github.com/openrazer/openrazer/files/7531367/capture.zip)

### Synapse Screenshots
[image]
[image]

### Streamer Companion Screenshots
[image]

z3ntu, 2021-12-03:
Thanks for the pcaps. The device uses 0x0c transaction ID in the driver and has `MATRIX_DIMS = [4, 16]` in the daemon

If you could work on this, you can follow recent device support commits for razeraccessory driver https://github.com/openrazer/openrazer/commits/master/driver/razeraccessory_driver.c and replicate that.

Mocha3k, 2022-08-10:
Year late but also would love this support!

trebizscett, 2024-11-02:
Any progress on this? It's been almost 3 years now and the lack of support for the main feature of this thing under Linux still bugs me.

z3ntu, 2024-12-07:
@trebizscett Support needs to be added by someone who owns the device, https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md has some hints how that works.
This is very much a community-powered project :)