openrazer/openrazer #1671: selecting certain effects does not work

miramar-labs, 2021-11-14:
### Distribution

ubuntu 20.04 fresh install

### Support

- [ ] Have you tried rebooting? y 
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page? y

fresh install of Ubuntu 20.04/openrazer/polychromatic on RazerBlade 15 base (early2020)
on selecting certain keyboard effects (eg starlight)  an error message pops up:

OpenRazer encountered an error processing this request. Try restarting the backend/application. If this message keeps appearing, please report this as a bug on OpenRazer's issue tracker.

stack trace:

Traceback (most recent call last):
  File ’/usr/lib/python3/dist-packages/polychromatic/backends/openrazer.py’, line 967, in set_device_state
    rzone.starlight_random(speed)
  File ’/usr/lib/python3/dist-packages/openrazer/client/fx.py’, line 586, in starlight_random
    self._lighting_dbus.setStarlightRandom(time)
  File ’/usr/lib/python3/dist-packages/dbus/proxies.py’, line 141, in __call__
    return self._connection.call_blocking(self._named_service,
  File ’/usr/lib/python3/dist-packages/dbus/connection.py’, line 652, in call_blocking
    reply_message = self.send_message_with_reply_and_block(
dbus.exceptions.DBusException: org.freedesktop.DBus.Python.PermissionError: Traceback (most recent call last):
  File ’/usr/lib/python3/dist-packages/dbus/service.py’, line 711, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File ’/usr/lib/python3/dist-packages/openrazer_daemon/dbus_services/dbus_methods/chroma_keyboard.py’, line 582, in set_starlight_random_effect
    with open(driver_path, ’wb’) as driver_file:
PermissionError: [Errno 13] Permission denied: ’/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:1532:0255.0004/matrix_effect_starlight’

### Operating System / Desktop Environment / Version(s)
Ubuntu 20.04 fresh install

### Affected Devices

lah7, 2021-11-14:
A reboot will probably fix this. It's to do with the OpenRazer driver not having the correct permissions (`plugdev`). It's not a bug with Polychromatic.

Running this command will show you that some are `root root`, instead of `root plugdev`:

    ls -l /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:1532:0255.0004/

Example:

```
--w--w---- 1 root plugdev 4096 Nov 14 17:53 matrix_custom_frame
--w--w---- 1 root plugdev 4096 Nov 14 17:53 matrix_effect_breath
--w--w---- 1 root plugdev 4096 Nov 14 17:53 matrix_effect_custom
```

`udev` normally assigns the `plugdev` permission when a device is plugged in, but since you can't physically remove/reinsert your laptop's keyboard like a USB device, a reboot is your best bet.

lah7, 2021-11-14:
If a reboot doesn't work, please could you output:

    ls -l /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:1532:0255.0004/

This command should fix it temporarily until reboot:

    sudo chown root:plugdev /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:1532:0255.0004/*

A similar issue (https://github.com/openrazer/openrazer/issues/1513) has appeared previously, which was solved by manually copying [99-razer.rules](https://github.com/openrazer/openrazer/blob/master/install_files/udev/99-razer.rules) to `/usr/lib/udev/rules.d/99-razer.rules` since it was missing.

closed 2022-07-19 by github-actions[bot]