openrazer/openrazer #1675: Support for Black Widow V3 Pro Bluetooth/USB/2.4 GHz Keyboard

KlausStoertebeker, 2021-11-18:
I'd like to have support for the "**Black Widow V3 Pro**" keyboard which can be accessed via

- Bluetooth
- USB
- 2.4 GHz-Dongle (provided with the keyboard)

The model number of the keyboard is: **RZ03-0353**

Product page at Razer: [https://www.razer.com/de-de/gaming-keyboards/razer-blackwidow-v3-pro/RZ03-03530100-R3M1](url)

If I can assist with additional information, tell me which information are needed and how to obtain them.

Regards

Ansgar

GeoDoX, 2021-11-18:
Duplicate of #1279

KlausStoertebeker, 2021-11-18:
Sorry, and maybe I'm wrong. but the openrazer version I installed today doesn't recognize my wireless keyboard...

GeoDoX, 2021-11-18:
2.4 GHz Wireless isn’t yet supported, I’m working on it in my spare time
and will submit a PR when ready. Wired works 100% in all my cases of
testing.
-- 
Travis McKee

GeoDoX, 2021-11-18:
> Sorry, and maybe I'm wrong. but the openrazer version I installed today doesn't recognize my wireless keyboard...

Did you use the build instructions in the PR to install followed by a full reboot?

KlausStoertebeker, 2021-11-19:
Hi GeoDoX,
I'm running Linux Mint 20.1 and entered the following commands to install openrazer - they all completes without error:
`sudo add-apt-repository ppa:openrazer/stable`
`sudo apt update`
`sudo apt install openrazer-meta`
`sudo gpasswd -a $USER plugdev`
After that I powered off my laptop, followed by a power on immediately...
Or do I have to build the packages by myself?

GeoDoX, 2021-11-19:
@KlausStoertebeker you have to build the packages yourself using my repository and the branch that adds support for the specific support you’d like to use

Please note at the time of writing this not everything is completed for wireless support and Bluetooth is not even started yet due to issues noted in one of the issues/prs. However, wired is fully working in my tests and other users who’ve commented on the PR

z3ntu, 2021-12-03:
Duplicate of #1279

Btw #1622 was merged just now, so the daily ppa should support the wired version in the next 24 hours.

closed 2021-12-03 by z3ntu