openrazer/openrazer #1676: Cannot shutdown correctly after installation

Novationo, 2021-11-19:
### Distribution

Pop_OS! 21.10 Beta
uname -a output 
`Linux jamming 5.15.2-76051502-generic #202111121041~1636734399~21.10~1c02b32 SMP Fri Nov 12 16:56:49 U x86_64 x86_64 x86_64 GNU/Linux`
Razer Laptop I have: RZ09-3009E97 / 1532:0246

After installing openrazer when attempting to shutdown normal through the gui / terminal i get this output, then will not shutdown.
(I'm pretty sure this is a kernel 5.15 problem since i have not had this problem in the past.)
Problem also occurs with razer-laptop-control

Output on screen after trying to shutdown
`[ 5468.766665] wlo1: deauthenticating from 78:23:ae:86:fb:39 by local choice (Reason: 3=DEAUTH_LEAVING)
[ 5468.766681] fbcon: Taking over console
[ 5468.779791] Console: switch to colour frame buffer device 240x67`

Haven't found anything on this same issue, its not a big problem for me but i wanted to bring attention to it in case someone knows how to fix it :)

z3ntu, 2021-11-20:
Can you try uninstalling openrazer and see if the problem persists?

Novationo, 2021-11-21:
> Can you try uninstalling openrazer and see if the problem persists?

Oops my bad, I thought i wrote in the issue when uninstalled the problem doesn't persist. Guess I should fix that

z3ntu, 2021-12-03:
Can you check if `sudo rmmod razerkbd` works properly? Note that this might make your keyboard stop working temporarily until a reboot, so maybe have an external keyboard handy.

closed 2022-07-19 by github-actions[bot]