openrazer/openrazer #1678: Add support for Fedora 35 please

xmstspider, 2021-11-22:
### Distribution

Fedora 35

OpenRazer packages are built against [Fedora 33](https://download.opensuse.org/repositories/hardware:/razer/Fedora_33/noarch/openrazer-meta-3.1.0-1.1.noarch.rpm) , which ships with Python 3.9. Fedora 35 migrated to Python 3.10, which means that the openrazer-daemon package cannot be installed.

``` 
Problem 1: problem with installed package razergenie-0.9.0-3.1.x86_64
  - package razergenie-0.9.0-3.1.x86_64 requires razer-daemon, but none of the providers can be installed
  - package openrazer-daemon-3.1.0-1.1.noarch requires python(abi) = 3.9, but none of the providers can be installed
```

z3ntu, 2021-11-22:
The repo contains Fedora 35 packages but your repo list likely points to the old release.

xmstspider, 2021-11-22:
> The repo contains Fedora 35 packages but your repo list likely points to the old release.

Aaah, I see. Ok, I see it [here](https://download.opensuse.org/repositories/hardware:/razer), but this [link](https://software.opensuse.org/download.html?project=hardware%3Arazer&package=openrazer-meta) which I open from [here](https://openrazer.github.io/#download) still point to Fedora 33.

xmstspider, 2021-11-22:
Ok, reopening, the openrazer-daemon is packed with Requires: python(abi)=3.9, Fedora 35 install Python 3.10.

z3ntu, 2021-11-29:
Cannot see where it should be that.
```
$ wget https://download.opensuse.org/repositories/hardware:/razer/Fedora_35/noarch/openrazer-daemon-3.1.0-1.1.noarch.rpm
$ rpm -qpR openrazer-daemon-3.1.0-1.1.noarch.rpm
<snip>
python(abi) = 3.10
<snip>
```

xmstspider, 2021-11-29:
I don't get it - I downloaded the Fedora 35 packages on the 22nd Nov and it pointed to python 3.9. Now I downloaded them again from the same directory and they point to python 3.10.

Something is spooky with my system.

Anyway, we can close the issue :-)

closed 2021-11-29 by xmstspider