openrazer/openrazer #1680: Razer Viper 8k problems

AlexanderCurl, 2021-11-24:
### Distribution
Arch Linux

### Support

- [X] Have you tried rebooting?
- [X] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I decided to open a new ticket instead of commenting the merge request.
I did further testing, as I got the device just yesterday.

I successfully compiled the source from AUR, it recognizes the device.

As I mentioned I've tested the poll rate with evhz and it's reporting only 1000Hz, hence it's set to 8k in polychromatic.
I compiled polychromatic-git also from AUR, same thing.

```
Razer Razer Viper 8KHz: Latest   975Hz, Average   992Hz
Razer Razer Viper 8KHz: Latest  1024Hz, Average   993Hz
Razer Razer Viper 8KHz: Latest  1002Hz, Average   993Hz
Razer Razer Viper 8KHz: Latest   999Hz, Average   993Hz
Razer Razer Viper 8KHz: Latest   982Hz, Average   993Hz
Razer Razer Viper 8KHz: Latest  1019Hz, Average   993Hz
Razer Razer Viper 8KHz: Latest  1000Hz, Average   993Hz
Razer Razer Viper 8KHz: Latest   500Hz, Average   985Hz
Razer Razer Viper 8KHz: Latest   977Hz, Average   985Hz
Razer Razer Viper 8KHz: Latest  1022Hz, Average   985Hz
Razer Razer Viper 8KHz: Latest   495Hz, Average   977Hz
Razer Razer Viper 8KHz: Latest  1012Hz, Average   977Hz
Razer Razer Viper 8KHz: Latest   334Hz, Average   966Hz
Razer Razer Viper 8KHz: Latest   905Hz, Average   965Hz
Razer Razer Viper 8KHz: Latest   342Hz, Average   955Hz
Razer Razer Viper 8KHz: Latest   143Hz, Average   942Hz

Average for Razer Razer Viper 8KHz:   942Hz```

The mouse behaves funky, I've tested in CS:GO, feels like some mouse acceleration is turned on, but I disabled any, both in-game, and both my desktop environment setting also.
I'm using XFCE, went into the mouse settings, turned acceleration down to 1.
As I move the mouse from one end to the other on my mousepad slow, and same fast movement, turns more somehow if fast paced movement detected, but it should move always the same speed. Maybe the polling rate not consistent? I don't know how to check..
It feels weird in-game, but on the desktop works just fine. If you guys need more information, please point me what logs or outputs you need to look further.

AlexanderCurl, 2021-11-24:
Did further testing, I was able to fix the polling rate, now `evhz` reports 8000Hz.
All I did was Sync DPI to Hardware, possibly that fixed it.
```
Razer Razer Viper 8KHz: Latest  8000Hz, Average  8000Hz
Razer Razer Viper 8KHz: Latest  8000Hz, Average  8000Hz
Razer Razer Viper 8KHz: Latest  8000Hz, Average  8000Hz
Razer Razer Viper 8KHz: Latest  8000Hz, Average  8000Hz
Razer Razer Viper 8KHz: Latest  8000Hz, Average  8000Hz
Razer Razer Viper 8KHz: Latest  8000Hz, Average  8000Hz
Razer Razer Viper 8KHz: Latest  7874Hz, Average  7999Hz
Razer Razer Viper 8KHz: Latest  8130Hz, Average  8001Hz
Razer Razer Viper 8KHz: Latest  4016Hz, Average  7939Hz
Razer Razer Viper 8KHz: Latest  7936Hz, Average  7938Hz
Razer Razer Viper 8KHz: Latest  7874Hz, Average  7935Hz
Razer Razer Viper 8KHz: Latest  8130Hz, Average  7939Hz
^C
Average for Razer Razer Viper 8KHz:  7939Hz
```

lah7, 2021-11-24:
> As I move the mouse from one end to the other on my mousepad slow, and same fast movement, turns more somehow if fast paced movement detected, but it should move always the same speed. 

Does XFCE have a setting to choose between "Flat" vs "Adaptive" acceleration?

KDE for example has this:

[image]

I'm not sure for XFCE, but there might be something or a tool/script you can use to set it at a software level:

https://forum.manjaro.org/t/xfce-default-mouse-setting-is-killing-me-day-by-da/9559

---

> Okay I tested with evhz on Arch linux, it's reporting only 1000Hz maximum, hence it's set to 8000Hz in polychromatic.
> (from #1633)

I'm not familiar with this mouse, but if you spin your cursor in a circle and use the keyboard's arrow keys to flick the poll rate - does the cursor momentarily pause when the poll rate changes? 

[image]

That should help confirm if the hardware is _doing something_ when it switches all the way up to 8000 Hz - but this pause could also just be a quirk on my Razer Taipan (125-1000 Hz). It seems weird that "Sync DPI to hardware" (`set_dpi_stages`) would make it work the first time.

Poll rate controls how fast the mouse sends data to the computer - so if it's on a congested USB bus, or the system is bottlenecked, maybe too high polling is causing a problem when the system is under CPU load (in-game). Try a different port, no hubs or unplug some USB devices. It could be software (Xorg) too, I see errors like this all the time:

```
$ cat /var/log/Xorg.0.log

[   856.291] (EE) event26 - Razer Razer BlackWidow Chroma: client bug: event processing lagging behind by 47ms, your system is too slow
[   948.493] (EE) event26 - Razer Razer BlackWidow Chroma: client bug: event processing lagging behind by 20ms, your system is too slow
```

It sounds more likely to be _acceleration behaviour_ in XFCE though.

AlexanderCurl, 2021-11-24:
> Does XFCE have a setting to choose between "Flat" vs "Adaptive" acceleration?

It doesn't have that option.
I've checked what you mentioned, while I was opened `evhz` next to the polychromatic app.
Changed the DPI's and in `evhz` showed me the correct Polling rates now, when I changed in polychromatic.
But I haven't noticed any momentarily pause when I was changed, it moved as it should.
And I cannot toggle between DPI settings, because the hardware switch is located in the bottom on the mouse.

I've checked Xorg.log as you mentioned, but haven't seen any related errors for the mouse, but I've seen what you shown above, as I have Chroma keyboard. :D

This mouse just plugged in in the old mouse place, that was working fine before. (Deathadder Elite)
No hubs, nothing, direct connection to my back in my PC.

I was screwing all mouse settings in XFCE, but no luck so far.

AlexanderCurl, 2021-11-24:
I've checked the link what you sent, tried with the xorg.conf.
Added this config:
```
Section "InputClass"
    Identifier "Razer Viper 8KHz"
    MatchIsPointer "yes"
    Option "AccelerationProfile" "-1"
    Option "AccelerationScheme" "none"
    Option "AccelSpeed" "-1"
EndSection
```

No effect sadly.. Still doesn't work as it should. No idea what's going on in this stage.

AlexanderCurl, 2021-11-24:
Here is the `xinput --list-props` output if somewhat useful:
```
Device 'Razer Razer Viper 8KHz':
	Device Enabled (154):	1
	Coordinate Transformation Matrix (156):	1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
	libinput Natural Scrolling Enabled (353):	0
	libinput Natural Scrolling Enabled Default (354):	0
	libinput Scroll Methods Available (357):	0, 0, 1
	libinput Scroll Method Enabled (358):	0, 0, 0
	libinput Scroll Method Enabled Default (359):	0, 0, 0
	libinput Button Scrolling Button (360):	2
	libinput Button Scrolling Button Default (361):	2
	libinput Button Scrolling Button Lock Enabled (362):	0
	libinput Button Scrolling Button Lock Enabled Default (363):	0
	libinput Middle Emulation Enabled (364):	0
	libinput Middle Emulation Enabled Default (365):	0
	libinput Accel Speed (366):	-1.000000
	libinput Accel Speed Default (367):	0.000000
	libinput Accel Profiles Available (368):	1, 1
	libinput Accel Profile Enabled (369):	0, 1
	libinput Accel Profile Enabled Default (370):	1, 0
	libinput Left Handed Enabled (371):	0
	libinput Left Handed Enabled Default (372):	0
	libinput Send Events Modes Available (276):	1, 0
	libinput Send Events Mode Enabled (277):	0, 0
	libinput Send Events Mode Enabled Default (278):	0, 0
	Device Node (279):	"/dev/input/event5"
	Device Product ID (280):	5426, 145
	libinput Drag Lock Buttons (355):	<no items>
	libinput Horizontal Scroll Enabled (356):	1
	libinput Scrolling Pixel Distance (373):	15
	libinput Scrolling Pixel Distance Default (374):	15
 ```

AlexanderCurl, 2021-11-24:
Okay, done some googleing, and it turns out, it is indeed a XFCE problem it seems.
I've read over stackexchenge, I should give a shot to `lxinput`.
In XFCE mouse settings I disabled everything, and hence I disabled everything lxinput reported 2x acceleration still.
Disabled there, and SEEMS like and feels like it solved the problem.

AlexanderCurl, 2021-11-26:
I gonna re-open the case, it's still doing that.
I've done further testing in-game, with various poll rates, and with 1000 Hz my AIM is consistent, no slowdowns, and weird behaviour.
If you guys needs specific logs or something feel free to ask, I will try my best to provide any information I could.

z3ntu, 2021-12-03:
Just so know, we likely won't be able to do anything as openrazer is just there to control the settings on the device, not much else really. Whatever is causing your problems, I doubt it's caused by openrazer.

AlexanderCurl, 2021-12-04:
Fine, if it's some other libraries error, maybe libinput, or something similar. Just I have no clue where to start to debug. On Windows works fine.

lah7, 2021-12-05:
If you have a spare HDD/SSD or partition, you could try installing a different desktop environment (like KDE) on a fresh distro installation or clone/backup your existing Arch install and see how another environment handles the mouse behaviour. 

Two desktop environments shouldn't _really_ co-exist together, even "swapping" can sometimes cause problems/conflicts, hence a separate disk/partition so you don't wreck your main system.

I also agree there's not much else OpenRazer can do, since the poll rate is correctly being set by the hardware.

robertswiecki, 2022-06-12:
Not sure if related, but this mouse have tendency to switch itself to the USB1 mode, in which maximum polling is 1000Hz. It happened to me with a KVM switch. What helped, was restting port with ```uhubctl```

You can check it with ```lsusb -tv```, if you see 12Mbps, it's USB1

closed 2023-03-13 by github-actions[bot]