openrazer/openrazer #1681: Support for Razer Huntsman V2 Tenkeyless

ThomChroma, 2021-11-26:
The TKL version of the Huntsman V2 fixed in #issue-1054291882

### Device Info

* Name: Razer Huntsman V2 Tenkeyless
* Product Number: RZ03-03940600-R3F1
* Model Number: RZ03-0394

### Device Information
```
Bus 001 Device 004: ID 1532:026b Razer USA, Ltd Razer Huntsman V2 Tenkeyless
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x026b
  bcdDevice            1.01
  iManufacturer           1 Razer
  iProduct                2 Razer Huntsman V2 Tenkeyless
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0086
    bNumInterfaces          5
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      1 Keyboard
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     177
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0016  1x 22 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      79
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        3
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      22
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        4
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      1 Keyboard
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     348
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x85  EP 5 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0002  1x 2 bytes
        bInterval               1
```

```
lrwxrwxrwx 1 root root  10 24 nov.  14:16 usb-Razer_Razer_Huntsman_V2_Tenkeyless-event-if01 -> ../event17
lrwxrwxrwx 1 root root  10 24 nov.  14:16 usb-Razer_Razer_Huntsman_V2_Tenkeyless-event-if04 -> ../event19
lrwxrwxrwx 1 root root   9 24 nov.  14:16 usb-Razer_Razer_Huntsman_V2_Tenkeyless-event-kbd -> ../event6
lrwxrwxrwx 1 root root   9 24 nov.  14:16 usb-Razer_Razer_Huntsman_V2_Tenkeyless-if01-event-kbd -> ../event7
lrwxrwxrwx 1 root root  10 24 nov.  14:16 usb-Razer_Razer_Huntsman_V2_Tenkeyless-if02-event-mouse -> ../event18
lrwxrwxrwx 1 root root   9 24 nov.  14:16 usb-Razer_Razer_Huntsman_V2_Tenkeyless-if02-mouse -> ../mouse3
```

### Packet Captures
…

### Synapse Screenshots
[image]
[image]

CSSlegend, 2021-12-09:
Yes plz add support, I have a Razer huntsman v2 tkl too and I would love to be able to use per key rgb and other features. I started using the program for my viper ultimate and I love it thus far

appuchias, 2022-04-07:
Hi,
I have this keyboard and am willing to help.
I have read and completed the prerequisites from https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol
I have a VM with Windows 10 and Synapse installed as well as wireshark configured.

The question now is, is my help needed? And if so, what can I do to help?

z3ntu, 2022-04-07:
@appuchias There's already some support in PR #1754 but if you can capture some pcaps it would still be nice to have them as reference, definitely not necessary though.

appuchias, 2022-04-07:
Oh, I hadn't seen the PR status, it's good to see.
I'll try to capture them this weekend. How should I upload them?

z3ntu, 2022-04-07:
Upload them to this issue as attachment (in a zip), Google Drive or whatever

0pLuS0, 2022-06-20:
I asked for the support too in another post.

I just got one of these keyboards, really nice too...

THANKS

deafpolygon, 2022-11-09:
Currently using this kb, I'm considering a switch to Linux and would like this to be supported someday too!

z3ntu, 2022-11-09:
PR #1754 has been merged and device support will be in the next release. Closing.

closed 2022-11-09 by z3ntu