openrazer/openrazer #1684: Numlock not working (Huntsman Mini)

Decapitater, 2021-11-27:
### Distribution
Arch 

Openrazer v3.10
### Support
@lah7 
- [yes (i am not a retard lol) ] Have you tried rebooting?
- [ yes ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Numlock was working with Fn+o/(scr Ik)  before using the daemon in arch , it used to work in my previous installation in Debian based Distros 

tldr: Numlock used to work before using openrazer  and used to work even with openrazer in debian based distros

Extra: Is there a way to change colour of highlighted buttons when using fn in 60% keyboard

Decapitater, 2021-12-03:
@lah7 any help luke skywalker ?

z3ntu, 2021-12-03:
Can you please explain what you mean with numlock? The device have a numpad nor do I see numbers labeled as secondary function on the main keys.

Decapitater, 2021-12-04:
Hi @z3ntu  
How is health 
(insert apology for using wrong term)
I wanted to know why was i unable use scrlock 
Which stops keyboard from typing until unlocked
the default without drivers was fn+O as o is also side printed as scrlock

z3ntu, 2021-12-04:
What do you expect to happen when scroll lock is active and what actually happens?

Decapitater, 2021-12-04:
@z3ntu  it does not work , the assigned way to use it is fn+o , after which o glows (if on Debian or on arch without drivers) 

(Cute cat tho )

z3ntu, 2021-12-04:
Please elaborate, does it do anything other than making the O key glow? I've never seen the device so you have to explain everything, I have no clue what that does in this device.

Decapitater, 2021-12-05:
Alright 
Fn key acts as an acess to the buttons unavailable on the keyboard 

For ex Fn+1 acts as F1,
Whenever some key such as caps lock is pressed it changes its color to Indicate it's being used 

When fn+u is used it starts the game mode disabling windows key , it highlights the u key 

Fn+O is labelled as scrlock it's supposed to stop you from writing further. It used to glow to indicate it's function, now it doesn't work as it doesn't glow and function. 

[image]

z3ntu, 2021-12-05:
Okay thanks for explaining that.

Can you check with `xev` or `libinput debug-events --show-keycodes` if the keyboard really doesn't emit the event anymore? 

Also generally I don't think that scroll lock should "stop you from writing further", read more e.g. here https://www.technipages.com/what-does-scroll-lock-do

Decapitater, 2021-12-05:
@z3ntu in terminal it does or used to

z3ntu, 2021-12-05:
> Can you check with `xev` or `libinput debug-events --show-keycodes` if the keyboard really doesn't emit the event anymore?

Please do this, otherwise I cannot help you.

Decapitater, 2021-12-05:
@z3ntu  alright give me some time

closed 2022-07-19 by github-actions[bot]