openrazer/openrazer #1686: Razer DeathAdder Elite V2 DPI switch buttons do not registered properly

nvaert1986, 2021-11-27:
Hello all,

I've recently switched from a Razer DeathAdder Elite to a Razer DeathAdder V2, but my DPI buttons on the mouse do not seem to get function properly. I'm using the Polychromatic GUI to set the DPI stages and when I change the DPI in the GUI (by clicking on the button) the DPI gets adjusted accordingly, but when I press the DPI up or DPI down button on the mouse, it does not and it switches to one of the factory preset DPI settings.

Is anybody familiar with this problem? Is this a known bug? Is there any additional information you guys need from me?

lah7, 2021-11-27:
Hi. It sounds like expected behaviour, if you mean that these "DPI Stages" do not sync to the hardware's up/down buttons:

[image]

The DPI buttons currently only operate in the default hardware mode. OpenRazer doesn't handle DPI stages in software yet. There are a few mice that are capable of setting DPI stages at a hardware level, and would display a "Sync DPI to hardware" button if supported. This GUI allows users to specify their own stages to click on.

I can see how this can be misleading, so I'll rework this "Edit DPI" button for that project. 

For here, in OpenRazer, contributions are welcome by anyone willing to add support for software DPI - currently the DPI button events are being _swallowed_, so no app knows if they're being pressed. #201 is related to this.

nvaert1986, 2021-11-29:
> Hi. It sounds like expected behaviour, if you mean that these "DPI Stages" do not sync to the hardware's up/down buttons:
>
> [image]
>
> The DPI buttons currently only operate in the default hardware [...]

Then I think you're right and that it's expected behavior then, though just to clarify: The correct DPI stages do get set when pressing the software button on the screen in the Polychromatic GUI, but they do not get set pressing the physical hardware buttons on the mouse. When pressing the physical DPI buttons on the mouse the factory default pre-configured DPI stages get set.

closed 2022-07-19 by github-actions[bot]