openrazer/openrazer #1688: Where can I get archived Fedora 32 packages?

rpgdev, 2021-11-28:
Current repos hold versions for Fedora 33 and up which is understandable but can someone please tell me where can I get archived Fedora 32 packages? For reasons out of my control I can go through the upgrade process now.

z3ntu, 2021-11-28:
I don't think OBS archives disabled repositories, but you can build for Fedora 32 yourself from source https://github.com/openrazer/openrazer/wiki/Building-a-package

Or alternatively even build it for Fedora 32 in your own OBS project depending on your needs

closed 2021-11-28 by z3ntu

rpgdev, 2021-11-28:
@z3ntu OBS archives? Can you please point what are those? I tried replacing the 33 for a 32 in the opensuse repo settings but it didn't work https://software.opensuse.org/download.html?project=hardware%3Arazer&package=openrazer-meta

rpgdev, 2021-11-28:
Never mind, I was able to build my own package using the link you posted and I was able to install it. I am getting issue #181 but I was able to work around it changing the python path. Thanks!