openrazer/openrazer #1689: Openrazer crashing

PlamenStilyianov, 2021-11-28:
openrazer-daemon.service - Daemon to manage razer devices in userspace
   Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service; disabled; vendor preset: disabled)
   Active: failed (Result: exit-code) since Sun 2021-11-28 13:14:16 GMT; 24s ago
     Docs: man:openrazer-daemon(8)
  Process: 3206 ExecStart=/usr/bin/openrazer-daemon -F (code=exited, status=1/FAILURE)
 Main PID: 3206 (code=exited, status=1/FAILURE)

z3ntu, 2021-11-28:
Can you post the output of `openrazer-daemon -Fv` ?

PlamenStilyianov, 2021-11-28:
razergenie does not open the above is log from the GUI, command prompt does not write any output out
orajava@suse152:~> razergenie 
Translation loaded: false

z3ntu, 2021-11-28:
I didn't ask about anything from RazerGenie, can you please run the command `openrazer-daemon -Fv` in your terminal and copy-paste the output here?

PlamenStilyianov, 2021-11-28:
orajava@suse152:~> openrazer-daemon -Fv
Starting daemon.
2021-11-28 13:31:30 | razer                          | CRITICAL | User is not a member of the plugdev group
2021-11-28 13:31:30 | razer                          | CRITICAL | Please run the command 'sudo gpasswd -a $USER plugdev' and then reboot!
Stopping daemon.

jeffreyvangorkum, 2021-11-28:
It literally tells you what the issue is and what you have to do ;)

In short, your user is no member of the group "plugdev". To fix this run the command 'sudo gpasswd -a $USER plugdev'

closed 2022-07-19 by github-actions[bot]