openrazer/openrazer #1690: Scrolling gets triggered twice

Ricky12Awesome, 2021-11-28:
Distro: Garuda (Arch Based) - 5.15.5-zen1 kernal

Openrazer version: 3.1.0
Mouse: Razer Basilisk v2

Scrolling gets triggered twice, just to make sure it wasn't something else, I'd uninstalled the driver and scrolling worked fine, then I installed it again and now it doesn't

### `xev -event button` with driver
```
ButtonPress event, serial 25, synthetic NO, window 0x5200001,
    root 0x1f0, subw 0x0, time 935869, (588,312), root:(4483,1078),
    state 0x0, button 4, same_screen YES

ButtonRelease event, serial 25, synthetic NO, window 0x5200001,
    root 0x1f0, subw 0x0, time 935869, (588,312), root:(4483,1078),
    state 0x800, button 4, same_screen YES

ButtonPress event, serial 25, synthetic NO, window 0x5200001,
    root 0x1f0, subw 0x0, time 935870, (588,312), root:(4483,1078),
    state 0x0, button 4, same_screen YES

ButtonRelease event, serial 25, synthetic NO, window 0x5200001,
    root 0x1f0, subw 0x0, time 935870, (588,312), root:(4483,1078),
    state 0x800, button 4, same_screen YES
```

### `xev -event button` without driver
```
ButtonPress event, serial 25, synthetic NO, window 0x5200001,
    root 0x1f0, subw 0x0, time 1224351, (266,321), root:(4161,1087),
    state 0x0, button 4, same_screen YES

ButtonRelease event, serial 25, synthetic NO, window 0x5200001,
    root 0x1f0, subw 0x0, time 1224352, (266,321), root:(4161,1087),
    state 0x800, button 4, same_screen YES
```

Ricky12Awesome, 2021-12-01:
I refunded the mouse and got the G604 which works great, so I wont be able to help in the future for this issue.

Teberon, 2021-12-01:
I have the same issue. If you need any details let me know.

z3ntu, 2021-12-03:
So you mean when you turn the scrollwheel one notch, then it does it but twice?

Teberon, 2021-12-03:
yes, exactly. I noticed this first while playing Satisfactory where turning a Splitter was impossible as it was rotating 180 degrees instead of 90 degrees. I have uninstalled openrazer and it does again the 90 degree turn. Also my old Steelseries did this correct when replacing it

z3ntu, 2021-12-03:
CC @mahmoudhossam @jgrim Do you also see this with your mouse? 

There is some custom code in the driver for this device, I would guess that's somehow causing this behavior.

mahmoudhossam, 2021-12-03:
I'm not experiencing this issue with my mouse, running openrazer 3.1.0 on Manjaro.

jgrim, 2021-12-06:
I've not seen this either; however, I'm not daily driving Ubuntu like I was before.

andrew-mcmahon, 2021-12-07:
I'm able to replicate the issue in the first post with my Basilisk v2 as well.
I've noticed recently that my scrolling has been a bit erratic.
If I remove the module (rmmod razermouse) I get two events per scroll/click.
I'm using Debian Testing with their openrazer package.
I don't believe the mouse is physically defective.

Polychromatic shows the mouse firmware as 1.4 which is how it came out the box. 
The Razor support site only shows [1.03](https://mysupport.razer.com/app/answers/detail/a_id/4554/~/razer-basilisk-v2-firmware-updater-%7C-rz03-03400) available.
But 1.04 does exist:
https://dl.razerzone.com/drivers/Basilisk%20V2/win/BasiliskV2_Mouse_FirmwareUpdater_v1.04.00_r1.exe

Perhaps the latest libinput is at fault? I've seen a topic [mentioning another mouse](https://gitlab.freedesktop.org/libinput/libinput/-/issues/681) & [Link-#2](https://www.reddit.com/r/archlinux/comments/puiysn/libinput_119_and_high_resolution_scrolling/).
```
ii  libinput-bin                          1.19.2-1                           amd64        input device management and event handling library - udev quirks
ii  libinput-tools                        1.19.2-1                           amd64        input device management and event handling library - command line tools
ii  libinput10:amd64                      1.19.2-1                           amd64        input device management and event handling library - shared library
ii  xserver-xorg-input-libinput           1.2.0-1                            amd64        X.Org X server -- libinput input driver

```
I'll have to try purging Debian's openrazer package and building master and 3.0.1 from source. 
Maybe even downgrade libinput. 
I don't recall it doing this before.

[image]
Attached the output of **libinput record basilisk_v2.txt**
[libinput_record_basilisk_v2.txt](https://github.com/openrazer/openrazer/files/7669132/libinput_record_basilisk_v2.txt)

andrew-mcmahon, 2021-12-07:
I believe **libinput 1.19** is at fault unfortunately - must be the new high resolution scrolling stuff.
You clever chaps will know better than a layman like me. Perhaps try a debian testing live image?
https://cdimage.debian.org/cdimage/weekly-live-builds/amd64/iso-hybrid/

I did purge debian's openrazer package (3.1.0) and built openrazer 3.0.1 from source as well as master - no difference there so it's not openrazer at fault.

The next step was to downgrade libinput from above to the versions in stable (1.16.4-3) i.e:
https://packages.debian.org/bullseye/xserver-xorg-input-libinput
https://packages.debian.org/bullseye/libinput-bin
https://packages.debian.org/bullseye/libinput10
```
ii  libinput-bin                          1.16.4-3                           amd64        input device management and event handling library - udev quirks
ii  libinput-tools                        1.19.2-1                           amd64        input device management and event handling library - command line tools
ii  libinput10:amd64                      1.16.4-3                           amd64        input device management and event handling library - shared library
ii  xserver-xorg-input-libinput           0.30.0-1                           amd64        X.Org X server -- libinput input driver
```
With the razermouse module loaded I'm getting two events per scroll (ButtonPress/ButtonRelease) as it should be.
[image]

azndd, 2021-12-14:
I can confirm, i had the same issue with my Razer Basilisk Ultimate Wireless.
After downgrading libinput to 1.16.4 and xf86-input-libinput to 1.1.0, everything works as excepted again.
Hopefully an update for libinput will come out soon, that fixes this. Using old versions of software doesn't feel like the best solution.

z3ntu, 2022-01-15:
Anything new here? Still a problem with libinput 1.19.3?

azndd, 2022-01-15:
@z3ntu Ive just tested it. Everything works normally again with version 1.19.3. Issue could be closed now, i guess.

z3ntu, 2022-01-15:
Thanks for confirming!

closed 2022-01-15 by z3ntu