openrazer/openrazer #1698: Battery notification does not pop-up - Viper Ultimate

HanM23, 2021-12-05:
### Distribution

Manjaro 21.2.0 - XFCE

### Support

- [ ] Have you tried rebooting? Yes
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page? Yes

Hi,

The battery notification does not pop-up. I set it to 60 secondes withing Polychromatic (in OpenRazer configuration window). I have this message in razer.log which is related :

`razer.device0.batterynotifier  | WARNING  | Failed to init notification daemon, err: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.Notifications was not provided by any .service files`
--> Notifications work for all other applications in my system.

Then, i disable xfce4-notifyd so it does not start at boot, i installed the regular notification-daemon and i did the following from archwiki:
[https://wiki.archlinux.org/title/Desktop_notifications#Standalone](url)

> after installing the notification-daemon package, add the following configuration to D-Bus services directory (/usr/share/dbus-1/services or $XDG_DATA_HOME/dbus-1/services

```
org.freedesktop.Notifications.service

[D-BUS Service]
Name=org.freedesktop.Notifications
Exec=/usr/lib/notification-daemon-1.0/notification-daemon
```

--> i tested it with success with

```
#!/bin/bash
notify-send 'Hello world!' 'This is an example notification.' --icon=dialog-information
```

Now, the only thing i have in razer.log related to battery (in debug mode) is 

`razer.device0                  | DEBUG    | Adding razer.device.power.getBattery method to DBus`

Notifications does not pop-up either
I do not have message from razer.device0.batterymanager or razer.device0.batterynotifier, i do not see if the initialization is OK or not.

Do you have any idea about what's wrong ?

Thanks a lot for your help
Regards

z3ntu, 2021-12-05:
Can you please run `qdbus --session org.freedesktop.Notifications` in your terminal and post the output here? (qdbus is part of qt5-tools)

HanM23, 2021-12-05:
Hi,

Here is the ouput :

```
$ qdbus --session org.freedesktop.Notifications
/
/org
/org/freedesktop
/org/freedesktop/Notifications
```
Does it help ?
I already had qt5-tools on my system.

Regards

z3ntu, 2021-12-05:
Maybe it's a race condition with the notification daemon, can you restart the daemon with `systemctl --user restart openrazer-daemon.service` while you're already logged in? And then check the log again if the same error appears again.

HanM23, 2021-12-05:
Indeed, i do not have the following warning 

`razer.device0.batterynotifier | WARNING | Failed to init notification daemon, err: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.Notifications was not provided by any .service files`

anymore when i restart the daemon, which is great. 
Yet, the notification does not pop-up, so it does not fix my issue.

In razer.log, i have this :

> systemd[1523]: Starting Daemon to manage razer devices in userspace...
> openrazer-daemon[2462]: Starting daemon.
> systemd[1523]: Started Daemon to manage razer devices in userspace.
> openrazer-daemon[2462]: 2021-12-05 15:34:58 | razer                          | INFO     | Initialising Daemon (v3.1.0). Pid: 2462
> openrazer-daemon[2462]: 2021-12-05 15:34:58 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
> openrazer-daemon[2462]: 2021-12-05 15:34:58 | razer                          | INFO     | Found device.0: 0003:1532:007B.0001
> openrazer-daemon[2462]: 2021-12-05 15:34:58 | razer.device0                  | INFO     | Initialising device.0 RazerViperUltimateWireless
> openrazer-daemon[2462]: 2021-12-05 15:34:58 | razer                          | INFO     | Found device.1: 0003:1532:007E.0007
> openrazer-daemon[2462]: 2021-12-05 15:34:58 | razer.device1                  | INFO     | Initialising device.1 RazerMouseDock
> openrazer-daemon[2462]: 2021-12-05 15:34:58 | razer                          | INFO     | Serving DBus

Is it normal that there is nothing related to batterynotifier or something like that ?

Thanks

HanM23, 2021-12-15:
Hi,

I add some new informations for this battery notification issue.
Since last week, I made some tests, i installed a fresh Manjaro with the latest stable ISO, and i have no notification at all for the battery.
Then, on another machine (not a vm) , the machine i am using right now, i have just installed the latest Xubuntu, with same result, no battery notification. On every machine i use, i always to the same thing, install the openrazer meta package, then, polychromatic package, add the right user in the right group, install the headers etc...

I have no error message, everything is green when i click on "Troubleshoot" within the OpenRazer window -> Preferences -> Backend (through Polychromatic).

I do not know what to do, i guess this is related to XFCE and their notification system. I mean, i receive no notification from openrazer in xfce4-notification GUI, nothing in the logs either....

Anybody has a idea ?
Thanks

lah7, 2021-12-16:
> EDIT : I also tried Manjaro KDE, same result, no notification. So, it's not related to XFCE. Maybe it's something specific with my mouse, but as it's working properly, i am lost now :(

Let's clarify the notification situation, do these test commands work? XFCE and/or KDE?

    notify-send -i battery "This is a test" "notify-send works!!" -a appname

The Python way - OpenRazer uses the `python-notify2` package:

    python3 -c 'import notify2; notify2.init("appname"); a = notify2.Notification("This is a test", "notify2 works!", "battery"); a.show()'

Some time ago when I was doing some testing for #1662, I noticed a bug in KDE's notification system where if the notification timed out on its own (not closed by the user), it **never** appeared again unless it came from a new `openrazer-daemon` process. If the notification was closed, then it would appear after the specified interval. 

---

Next, what are your settings set to in `~/.config/openrazer/razer.conf`?

OpenRazer 3.2.0 changed the setting for this notification 6900641232c13440d3ef4766dc017133bb37aad3, but it will read the old key if present.

```diff
-mouse_battery_notifier = True
-mouse_battery_notifier_freq = 300
+battery_notifier = True
+battery_notifier_freq = 300
```

(`freq` is in seconds - reference: https://github.com/openrazer/openrazer/blob/master/daemon/resources/razer.conf)

HanM23, 2021-12-17:
Hi Luke,

I made the tests on my both machines, one with Manjaro XFCE and the other with KDE.

All the tests work, with notify-send and with python. I receive the notification each time, there is no problem there.
On my both machines, i have the latest version of OpenRazer, 3.2.0 :
```

[General]
verbose_logging = False

[Startup]
sync_effects_enabled = True
devices_off_on_screensaver = True
battery_notifier = True
battery_notifier_freq = 60
restore_persistence = True
```
I suspect two possibilities, the first one if that, it does not go though the right loops in the file battery_notifier.py
The second one is that there is something specific with the mouse model, a viper ultimate, as we already know there is something specific with the dock and charging state (even if it's a different issue here).

I will try with the latest EndeavousOS version this week-end and i will let you know. But With Manjaro (XFCE and KDE), then with Xubuntu, i have no notification at the moment.

I have no knowledge with Python, i cannot investigate with battery_notifier.py unfortunately :(

lah7, 2021-12-17:
I think I've found the bug - this device (and a few others) were not setting the `battery_notifier_freq` configuration, so the notifier likely doesn't initialize. I noticed this when I was using the "fake" driver to simulate the Viper Ultimate. The daemon never output a `batterynotifier` line like this:

```
2021-12-17 15:44:36 | razer.device0.batterynotifier  | DEBUG    | Razer Mamba Battery at 100.0%
```

Please try https://github.com/openrazer/openrazer/pull/1710 to see if it fixes it. You can patch your (up-to-date) installation like so:
```bash
curl https://patch-diff.githubusercontent.com/raw/openrazer/openrazer/pull/1710.diff > 1710.diff
sudo patch /usr/lib/python3.10/site-packages/openrazer_daemon/hardware/mouse.py < 1710.diff
rm 1710.diff
```

Then restart the daemon and fingers crossed. 🤞🏼

HanM23, 2021-12-17:
Hi Luke,

I get the following error when 

```
patching file /usr/lib/python3.10/site-packages/openrazer_daemon/hardware/mouse.py
Hunk #1 FAILED at 1744.
Hunk #2 FAILED at 2266.
Hunk #3 FAILED at 2446.
Hunk #4 FAILED at 2477.
Hunk #5 FAILED at 2514.
5 out of 5 hunks FAILED -- saving rejects to file /usr/lib/python3.10/site-packages/openrazer_daemon/hardware/mouse.py.rej
patch: **** Can't reopen file /usr/lib/python3.10/site-packages/openrazer_daemon/hardware/mouse.py : No such file or directory
patch: **** Can't reopen file /usr/lib/python3.10/site-packages/openrazer_daemon/hardware/mouse.py : No such file or directory
patch: **** Can't reopen file /usr/lib/python3.10/site-packages/openrazer_daemon/hardware/mouse.py : No such file or directory
patch: **** Can't reopen file /usr/lib/python3.10/site-packages/openrazer_daemon/hardware/mouse.py : No such file or directory
```
....

I have hundreds of "No such file or directory", then hundreds of 

```
patch: **** Can't create file /usr/lib/python3.10/site-packages/openrazer_daemon/hardware/mouse.py.orig : Too many open files
patch: **** Can't create file /usr/lib/python3.10/site-packages/openrazer_daemon/hardware/mouse.py.orig : Too many open files
patch: **** Can't create file /usr/lib/python3.10/site-packages/openrazer_daemon/hardware/mouse.py.orig : Too many open files
```

The last line is :

```
patch: **** Can't create file /usr/lib/python3.10/site-packages/openrazer_daemon/hardware/mouse.py.origSegmentation fault

```

Now, i have to file into `/usr/lib/python3.10/site-packages/openrazer_daemon/hardware/` mouse.py.reject and an empty mouse.py.orig

I have the feeling something went wrong 😅

What i am supposed to do ? Rename mouse.py.reject into mouse.py ?

lah7, 2021-12-18:
Ah, it's likely because Manjaro (based on Arch) is still on Python 3.9 whereas Arch is now at 3.10. Try again, but change `python3.10` to `python3.9`. 

It's safe to delete leftovers from the failed attempt:

    sudo rm /usr/lib/python3.10/site-packages/openrazer_daemon/hardware/mouse.py.rej
    sudo rm /usr/lib/python3.10/site-packages/openrazer_daemon/hardware/mouse.py.orig

💡  Something to bear in mind is that once the Python 3.10 update lands for Manjaro, you'll need to reinstall your AUR Python packages (if you need help, see https://github.com/polychromatic/polychromatic/issues/392#issuecomment-995647310) and patch again.

HanM23, 2021-12-18:
Hi Luke,

Notifications work now !!! Yeah ! It does pop-up every xxx minutes, whether i close it manually or not.
Thanks a lot for your help!

PR #1710 (2022-01-15): daemon: Fix missing battery notifications for some devices
Fixes #1698.

Some mice did not specify the `frequency` variable. As a result, the notification would not initialize for these devices:

- Razer Viper Ultimate (Wireless)
- Razer Basilisk Ultimate
- Razer DeathAdder V2 Pro Wireless
- Razer Basilisk X HyperSpeed
closed 2022-01-15 by z3ntu