openrazer/openrazer #1699: Devices do not turn off when Xscreensaver is activated or when screen is locked - Viper Ultimate w/ Dock

HanM23, 2021-12-05:
### Distribution

Manjaro 21.2.0 - XFCE

### Support

- [ ] Have you tried rebooting? yes
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page? yes

My mouse does not turn off when Xscreensaver is activated. It does not turn off when i lock my screen either. Yet, the checkbox is checked in OpenRazer configuration window (accessible in Polychromatic)
I only have this message in razer.log

`razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor`

In fact, the mouse turns off (which i suppose because the logo becomes black) 4 minutes after my screensaver is displayed, i do no know why. But even in this case, the mouse dock does not turn off. What can i do to investigate ?

Thanks a lot for your help
Regards

z3ntu, 2021-12-05:
Please run `dbus-monitor --session | grep ActiveChanged`, lock the screen and unlock it again and post the output here.

HanM23, 2021-12-05:
Hi @z3ntu,

Thanks for your support, nothing happens, there is no output at all when i run your command, either when i lock the screen or when the screensaver is activated. 

What can we conclude ?

Thanks

HanM23, 2021-12-06:
Hi,

I installed xfce4-screensaver, and it does work perfectly, devices (mouse + dock) turn off right when the screensaver is activated.
I conclude there is no issue with my system.

So, i guess, there is yet no support for Xscreensaver, am i right ?
If so, could you please add it ? I am refering to this enhancement request for xfce4-screeensaver, maybe it could be possible to add support for Xscreensaver too ? [#1284 ](https://github.com/openrazer/openrazer/pull/1284)

Thanks a lot,
Regards,

z3ntu, 2021-12-07:
If xscreensaver doesn't emit `ActiveChanged` on dbus (which apparently it doesn't) then there's no trivial way to support it.

HanM23, 2021-12-07:
> If xscreensaver doesn't emit `ActiveChanged` on dbus (which apparently it doesn't) then there's no trivial way to support it.

You are right @z3ntu, i am afraid it does not have any dbus interfance. 
Should i close the issue or should i let it open ?

Regards,

z3ntu, 2021-12-07:
This is out of scope then for openrazer. And there also doesn't seem to be anyone before you who missed xscreensaver compatibility.

closed 2021-12-07 by z3ntu